iNavFlight/inav #447: recipNORM

robert-b, 2016-08-09:
`        // Just scale by 1G length - That's our vector adjustment. Rather than 
        // using one-over-exact length (which needs a costly square root), we already 
        // know the vector is enough "roughly unit length" and since it is only weighted
        // in by a certain amount anyway later, having that exact is meaningless. (c) MasterZap
        recipNorm = 1.0f / GRAVITY_CMSS;
`
recipNorm should be static const - else just it's just a cpu wasting thing.

the flag useCOG is also used in a wrong manor.
e

digitalentity, 2016-08-09:
I agree about recipNorm, although compiler is usually smart enough to optimize that out.
What's wrong with useCOG?

robert-b, 2016-08-09:
const correctnes is up to the programmer. it is a hint for the compiler.
`    if (useMag || useCOG) {
         ...
        }
        else if (useCOG) {
        .....
        }
        else {
            ex = 0;
            ey = 0;
            ez = 0;
        }
`
still problem seeing the wrong usage?

digitalentity, 2016-08-09:
I don't see anything wrong here. If you feel like optimizing it - please do so in a form of pull request.

robert-b, 2016-08-09:
and the course over ground is 'only' valid when there are enough satelites.

robert-b, 2016-08-09:
just delete the first useCOG and add a sanity check.

digitalentity, 2016-08-09:
@robert-b please re-read the code carefully. Nothing is wrong with `useCOG` usage.

closed 2016-08-09 by digitalentity

robert-b, 2016-08-09:
`if (useMag || useCOG) { ... } else if (useCOG) { ..... } else { ex = 0; ey = 0; ez = 0; } `
ok, the else if is dead code. if useCOG is true then only the first if will be executed.
and this is just wrong if useMAG is false.

digitalentity, 2016-08-09:
Please read the code carefully:
```
if (useMag || useCOG) {
    if (useMag) {
        ...
    }
    else if (useCOG) {
        ...
    }
    else {
        ...
    }

    ...
}

```

robert-b, 2016-08-09:
correct.
try to read the misra standard.
the code can be written simpler.

digitalentity, 2016-08-09:
We are mostly following Linux Kernel coding guidelines here. If you feel that code can be simpler (yet also efficient) - submit a PR

robert-b, 2016-08-09:
there is a simple problem with linux kernel programming/guidelines.
the code is not 'good enough' to go though iec 61508 and obtaining sil seal.
it must be converted ... which is a pain.

with regards to the const attribute - as in java you can give the compiler a few hints.
then he's able to produce even better code.

const correctness is a problem today.

commit 8d16193c (2016-08-09): Minor IMU optimisation. Closes #447