iNavFlight/inav #452: CC3D taret broken

digitalentity, 2016-08-11:
Gyro and acc are detected but read zero values on CC3D.
`git bisect` narrowed this down to b7bf34c4c4d21e0279e2fb972660f4b8eabc7ac8 (#435)

@martinbudden can you have a look at this ASAP?

martinbudden, 2016-08-11:
OK, I'll look at this today.

digitalentity, 2016-08-11:
Thanks! I tried digging into the code, but alas, I don't have much time today to spen on INAV.

martinbudden, 2016-08-11:
@digitalentity , well I've tried to reproduce this. My build seems fine. I haven't flown the quad, but gyro and acc outputs show up fine in iNav configurator. I use the CC3D_OPL build.

As part of this I've done a betaflight catchup - I'll merge that in anyway, but my build worked before and after the catchup.

digitalentity, 2016-08-11:
@martinbudden this is odd. Apparently my CC3D doesn't work. acc and gyro are detected correctly. I'll re-test after you merge the betaflight catchup. If it will continue to fail, I'll start digging into the problem myself

digitalentity, 2016-08-11:
@martinbudden interesting.
GCC-4.9.3 - not working
GCC-5.4 - working fine

Might be either a GCC regression or possible memory corruption somewhere in the source.

digitalentity, 2016-08-11:
BTW, what version of GCC are you using?

martinbudden, 2016-08-11:
I was thinking it might be a compiler issue.

GCC, I'm on version 4.2.1! Ha ha!

I know, I need to upgrade. I need to upgrade my OS as well, I'm on OSX 10.9.5, latest is 10.11.6 with 10.12 in public beta.

Somehow I never seem to have time to upgrade my tools.

digitalentity, 2016-08-11:
It might still be unrelated to compiler - different compilers provide memory layout. If we have memory corruption - different memory gets corrupted with different effects.

Interestingly - after I put `debug[0] = gyroADCRaw[0]` into gyroRead - everything started to work :astonished:

martinbudden, 2016-08-11:
Actually you're right. It's more likely to be memory corruption than a compiler bug. So this problem needs to be better understood.

The gyro code driver area that remains different from betaflight, in particular gyro sync and filtering are handled differently - another thing to look into.

digitalentity, 2016-08-11:
Also code that uses memset might be guilty, or filtering implementation.

digitalentity, 2016-08-11:
Memory corruption is extremely hard to catch - you can't recompile code - memory will be re-arranged and the issue won't reveal itself. I tried playing with features and issue is still present.

martinbudden, 2016-08-12:
Another possibility is that there is a stack overflow problem.

What is the stack size, and where is it set up?

digitalentity, 2016-08-12:
Quite possible. Current stack is 1K and it set by _Min_Stack_Size in stm32_flash.ld

digitalentity, 2016-08-12:
No it's not stack - increased to 4K and CC3D still gets zero gyro and zero acceleration readings on the same code and same GCC version.

martinbudden, 2016-08-12:
Oh well, it was worth a try. 1K stack does seem pretty small though.

As you say, this is hard to debug. One thing you might like to do is look and the `.map` files. Look at the location of the gyro data in the map files, and what is next to it in memory. If this changes between compiler versions you have a possible culprit.

stronnag, 2016-08-12:
Seems OK with gcc 6.1 (for what it's worth)

digitalentity, 2016-08-13:
Something is terribly wrong with the code. While trying to figure out what's wrong with the gyro I ended up seing this:
[image]

Please note the -1430deg roll which shouldn't be possible at all as it's the output of `atan2_approx`.

digitalentity, 2016-08-13:
I strongly believe this is not related to latest IO commits - I recall having issues with not moving servos on SPRacingF3 @ d225f8e334cf9897534acfc48096225089314b8b (#355), just changing indexes in mixer from `uint8_t` to `int` (45fc7a68a3945c05d92ab28a00a2efd31e3dd22c) seemingly fixed the issue.

At that time only a few things about new IO was merged in.

digitalentity, 2016-08-13:
Interestingly, on CC3D usually both `gyroADC` and `accADC` are zero despite the fact that they are located in different places in memory and updated from different parts of the code. `gyroUpdate` is executed, but `gyroADC` stays zero, same for `accADC`. 

LTO hides static variables making debugging very hard - I'd be very interested to see what values are read from gyro into `gyroADCRaw` but I can't find the localtion in memory.

jflyper, 2016-08-17:
> GCC-4.9.3 - not working

Confirmed.

digitalentity, 2016-08-18:
Looks line #471 fixed the issue. No surprise, since we do `#define TARGET_MOTOR_COUNT 4` and thus have had
```
#define MAX_MOTORS  4
#define MAX_SERVOS  8
#define MAX_PWM_OUTPUT_PORTS 8
```
while initializing 10 outputs (overwriting some bytes right after the `pwmOutputPorts`).

closed 2016-08-18 by digitalentity