iNavFlight/inav #454: Flying Wing Refinement V1.1

moggiex, 2016-08-11:
Howdy,

I've labeled this as being "refinement" as I'm sure thats all it needs, probably settings on my Naze32 board with iNav

Yesterday I took the Teksumo up with this setup:
- Naze32 Rev 6 Board
- Crius Neo-6 GPS
- D4R-II Receiver with telemetry 
- Dump of the settings here [https://www.dropbox.com/s/6gthvwdm7p6s5nh/settings-dump.txt?dl=0](url)

Photo here: [https://www.dropbox.com/s/th401x0jk9xfcd1/2016-08-10%2017.36.08.jpg?dl=0](url)

What I experienced was the following:

**1. The GPS RTH was anything but home!**
After changing the nav_rth_altitude value to 5000 from 15000 it made its way at a sensible height, so that works well. But it didn't really come home.

It was a good 50-60 metres when enabling away and then went off away from me, I left it to do it's thing but I had to step in and take it back before it got any further away (it went off out into the distance!)

**2. Loiter mode**
This might be a setting thing, but when loiter mode, it doesn't really loiter. 

Again it went off on it's own little mission and didn't show any signs of coming back. It tries to circle but the radius on it is epic, say 800metres or more. Again I had to step in and rescue it before it got too small.

After reading through the settings again, I've increased _nav_fw_bank_angle_ (from 20 to 35) because it looked anemic in the air, also I've lowered _nav_fw_loiter_radius_ to 4000 to see if that will tighten up the circle in the air (I presume!) it should be making.

So my questions for refinement are:

**1. Does the GPS look OK where it is or is it too close to the servo?**

Note: It does get a GPS lock very quickly.

**2. Have I missed any settings out? Do you have any suggestions for a flying wing?**

**3. How can I give the servo's more movement in Angle, rate or horizon mode?**

Note: Yes I know this may cause the wing to oscillate, but at the moment it's extremely anemic to fly with any of these modes selected.

**4. And finally for altitude/height hold**
I'd like for the wing to keep it's height & direction, unless tweaked by stick inputs. What combination of flight modes do you suggest?

Note: I was thinking Heading Lock, but do I also enable MAG at the same time? What about Horizon too?

I know thats a lot of questions, I'm curious & stuck at the same time. Hence the plea for help :)

Matt

oleost, 2016-08-11:
1. Please close this and use forum for support. Also follow the wiki as there a key points as Sensor calibration you missed out on..

RTH/Loiter Issue: Your accelerometer is not calibrated.

GPS can ble close to servo no issue. As long as you are getting lock and enough servos.

Give it more P values i PID controller to get rate value. Start with regular P value.

moggiex, 2016-08-11:
That's odd, I calibrated the sensors & also did them again on the flight line to be sure. What setting told you that?

And also do you have a link to the forums please Oleost :) I'll happily post there!

Matt

oleost, 2016-08-11:
http://www.rcgroups.com/forums/showthread.php?t=2495732

oleost, 2016-08-11:
https://github.com/iNavFlight/inav/wiki/4.-Sensor-calibration

Use CLI to verify that accgain_x, accgain_y and accgain_z parameters are NOT 4096. If they are, algorithm failed to converge, calibration failed and needs to be repeated.

moggiex, 2016-08-11:
Oh yea, found them! Thanks, I'll redo them right now and keep an eye on those settings. I appreciate the help!

closed 2016-08-11 by moggiex

moggiex, 2016-08-11:
@oleost you mentioned that the board wasn't calibrated. because of these settings:

_set acczero_x = 0
set acczero_y = 0
set acczero_z = 0
set accgain_x = 4096
set accgain_y = 4096
set accgain_z = 4096_

I've just tried 2 other Naze32 boards and all of them won't calibrate with version 1.1 with FrSKY telemetry enabled. So thats 3 boards, different board makers, same hex file, none of them set accgain_x etc.. values

Matt

oleost, 2016-08-11:
Are you sure you are doing it right?

https://youtu.be/HxWSn_noNgg

Watch with sound then you here when I click.

11. aug. 2016 6.33 p.m.

moggiex, 2016-08-11:
Jackpot! 

Thank you for that video! I presume I'm not the first one to have that issue then :)

Also I spotted I had the buzzer far to close to the board, that was screwing with the magnetometer. 

Matt :+1: