iNavFlight/inav #457: INAV 1.2RC1 LED page

khomouda, 2016-08-11:
In the last commits, there is a problem accessing the LED strip page in configurators and increasing packet errors, which will need disconnecting and reconnecting without accessing the LED strip page
board is SPRACINGF3

chemmex, 2016-08-13:
Did you try opening Configuration and Failsafe tabs as well? Do they work in 1.2 RC1?

khomouda, 2016-08-13:
yes I tried and they are all working. I am using always the last commit, the LED page is working in Naze target

gaelj, 2016-08-13:
Latest configurator as well ?

khomouda, 2016-08-14:
yes

bazyl78, 2016-08-17:
I can't configure led strip on Naze on both configurators. Option for ordering simply doesn't work for Naze. Led comand is not available on CLI too.

gaelj, 2016-08-17:
I will check this evening after work.

gaelj, 2016-08-17:
`\src\main\target\NAZE.target.h` line 189
`//#define LED_STRIP`

This came from 0489eb8b0840719277450a8d15bb6f1d5e6f62be.
It was done to save on memory. Re-enabling it and compiling brings this error:
`/arm-none-eabi/bin/ld.exe: region `FLASH' overflowed by 4600 bytes`

closed 2016-08-17 by gaelj

bazyl78, 2016-08-20:
Anyway .... how many other features are disabled for Naze?

oleost, 2016-08-20:
https://github.com/iNavFlight/inav/wiki/1.-Getting-started-with-iNav

bazyl78, 2016-08-23:
If someone needs it it is possible to compile without telemetryband serial_rrx (ppm only) an then led strip works :)

khomouda, 2016-08-23:
This is what I am doing but still the page is not working. I have to
restore led commands from backup

bazyl78, 2016-08-23:
For me worked thankfully.

luxoid1, 2016-09-12:
Can you please help me how to compile the Naze version of iNav without telemetryband serial_rrx?
I could use the "make TARGET=NAZE" command without a problem and the hex file was created also but I don't know how to exclude the telemetryband from the build. 
I would like to enable LED strip function on Flip32+ board but on both GUI's it is showing 0 LED remaining to configure.

DzikuVx, 2016-09-12:
src/main/target/NAZE/target.h

There, you can enable LED by uncommenting `#define LED_STRIP`
Serial RX can be disabled in `src/main/target/common.h`

luxoid1, 2016-09-12:
Thank you very much for you fast reply and help! I will try it now.

luxoid1, 2016-09-12:
Compile returned with an error about region "FLASH" overflowed by 264 bytes. Can you please recommend anything else to disable to make it fit?

khomouda, 2016-09-12:
Yes if you are exceeding the flash limit, you have to keep removing
features. I had to remove even all compasses and barometers which are not
in the Naze

luxoid1, 2016-09-12:
I have removed each accelerometer, gyroscope, baro and magnetometers except those what are existing on the flip32+. The LED Strip and the rest are working now perfectly!
Thanks for your help!

khomouda, 2016-09-12:
Welcome