iNavFlight/inav #459: No Naza V2 detection

ienders13, 2016-08-13:
Set up same as 1.1 but not detecting Compass or Mag

Naze R6

oleost, 2016-08-13:
https://github.com/iNavFlight/inav/wiki/1.-Getting-started-with-iNav#notes--common-issues

Notes / Common issues

iNav has removed all other telemetry options when using Naze32 targets but LTM and FrSky in official builds, if you need let's say SmartPort telemetry you need to compile your own. See link. Eg. F3 board, which have more memory, have Smartport and HOTT included.

iNav on F1 targets(example Naze32) only supports UBLOX GPS per default due to size limitation. If other are needed you need to modifiy and compile your own version.

@digitalentity Maybe add this to release notes for iNav 1.2 as well.

ienders13, 2016-08-13:
Well it works on version 1.1 your link appears to apply to early versions. Not 1.2 specific, am I wrong?

oleost, 2016-08-13:
It's new from 1.2 and onwards. 

1.1 had naza gps included. But it got removed on 1.2 in favor of FrSky telemetry.

ienders13, 2016-08-13:
Ahh, ok, I've got an sprf3 delux I can use. Thanks. Deffo needs adding to the release notes.

closed 2016-08-13 by ienders13