iNavFlight/inav #467: PH rcCommands strange readings

xdigyx, 2016-08-15:
Today by chance I noticed that in PH mode after I moved pitch/roll stick and release it then my copter bounces like it would be too high I gain. I did check the same stick movement in angle mode and no bounces here.
Made a log and viewed it to see PID reaction, but noticed that after main stick movement there are another stick oscilationg movements which I did not do. These can be seen as rcCommands roll/pitch depending on what movement had been done.

Do you have it too?
FW ver 1.1.

bk79, 2016-08-15:
Bounces like breaks and get back?

xdigyx, 2016-08-15:
Say more like oscillations.
[image]

digitalentity, 2016-08-18:
This is normal. PH works by overriding stick input.

closed 2016-08-18 by digitalentity

DzikuVx, 2016-08-18:
@xdigyx could you publish BBlog of that exact case. I have an theory here I want to confirm