iNavFlight/inav #469: Extend blackbox flight modes

DzikuVx, 2016-08-16:
Catch up with Betaflight how flight modes are saved in BB logs. See 6343da6f091dde3e5eec2cad1ac054840166a489

* [ ] cleanup unused box modes (GOVERNOR, LEDMAX, more?)
* [ ] change `rcModeActivationMask` to uint64_t
* [ ] in blackbox , send `rcModeActivationMask` instead of `flightModeFlags`
    * [ ] remember to send 64 bits
* [ ] merge iNavFlight/blackbox-log-viewer#2

digitalentity, 2016-08-18:
Keep in mind that MSP_STATUS also returns current active box modes (similar to `rcModeActivationMask`) and it's 32-bit integer. Should be deprecate MSP_STATUS or break compatibility with CF/BF?

stronnag, 2016-08-18:
MSP_STATUS is used by ground stations (and presumably OSDs). Please don't break it .... at least without lots of warning.

DzikuVx, 2018-04-04:
I belive we already have this covered in latest versions

closed 2018-04-04 by DzikuVx