iNavFlight/inav #470: All rate values are zero after boot (also TPA breakpoint value)

iforce2d, 2016-08-17:
I flashed official 1.2 onto my flip32 that has been running 1.1 very well for a few months. Now every time after startup, each of the rate values are zero, rendering it un-flyable. I have to connect to the configurator (or EZ-GUI) and save some rate values, which causes the FC to remember the values I had set earlier. Unfortunately, this means I cannot actually set any values that I want :/

Here is a video that explains this better
https://youtu.be/Q4iHVJMIJeI

Another video, this time use EZ-GUI and trying to save some new values.
https://youtu.be/G5IsQpNbOuA

The correct values that appear when I click Save were set by me when I was using a pre-release build (built by me) of 1.2 about a week before the official release.

DzikuVx, 2016-08-17:
Are you using INAV configurator or CF or BF configurator?
Also, rates has been rescaled. What are the values you are trying to enter in a first place. 
Have you read this: https://github.com/iNavFlight/inav/wiki/Rate-conversion-for-INAV-1.2 ?

iforce2d, 2016-08-17:
Those questions are answered in the videos...
I am using inav configurator.
The rates I enter are not being accepted regardless of what I enter. Around 0.07 - 0.09 is what I was using before this happened.
I had not read that page, but a) my settings for pre-release 1.2 were ok, b) a value of zero is invalid no matter what the scaling.

digitalentity, 2016-08-17:
@iforce2d please update the configurator and see if the bug is fixed.

DzikuVx, 2016-08-17:
If not, I will try to debug this case today in the evening. I have Flip32

digitalentity, 2016-08-17:
I've tested on PixRacer (that's what's hooked up to my PC at the moment - no issues). It's either related to old configurator or specific to NAZE target.

@eziosoft, you might want to look at current INAV MSP code (with regard to RX configuration and PID tuning/rates/TPA) and sync EzGUI to it.

iforce2d, 2016-08-17:
But the symptom occurs without involving any configurator at all. I connect the battery and try to fly, and I have no pitch/roll/yaw control (only throttle). I will try with the updated configurator but I don't see how that could help, when the problem occurs without any configurator connection.

DzikuVx, 2016-08-17:
@iforce2d @digitalentity @martinbudden I just tested both 1.2-RC1 and latest master (dd6f6cd7eb5a8cb8d0f531935ed3196ea82f22e5) using Flip32 10DOF with latest master on configurator (9661c584c93ec0b252312766bb56bf4460fc52ab)

On 1.2-RC1: 
* rates and TPA are not saved to board, CLI values are not changed. No idea if it is flyable, it's only an board. Should be flyable though since in my case CLI values were at defaults: 200dps for rates and disabled TPA
* Configuration and Failsafe tabs are not accessible

On latest master dd6f6cd7eb5a8cb8d0f531935ed3196ea82f22e5 situation is slightly different. Tabs are accessible, but Rates and TPA are written only from time to time. Unable to determine when exactly. They were written at least once

DzikuVx, 2016-08-17:
Did some additional tests on SPRacingF3 and latest master on configurator

* 27822a5 -> Rates are saved, Failsafe and Configuration tabs are not accessible
* lates master  dd6f6cd -> Configuration and Failsafe tabs are accessible, rates and TPA are saved

So, it is at least Naze target specific

stronnag, 2016-08-17:
Strange, all works on my flip32 and CC3D with latest master and 1.2RC1. In fact I can't ever recall the configurator not working.

martinbudden, 2016-08-17:
This is all very odd. I'm away from home at the moment and don't have access to hardware, but I've looked through the code and can't see why rates shouldn't be loaded.

digitalentity, 2016-08-18:
Digged out my old Flip32 board, flashed with latest master and rates are gone. Really odd.

digitalentity, 2016-08-18:
This is either NAZE-specific or a symptom of #458.

digitalentity, 2016-08-18:
Looks like it's a case of #458. MSP_RC_TUNING returns values from `currentControlRateProfile` pointer, that is being set by calling `setControlRateProfile()` from `readEEPROM()`. If it points to zero-initialised memory, it's probably memory corruption.

digitalentity, 2016-08-18:
After boot up the value of `currentControlRateProfile` is 0x4000043C which is an address of ARM configuration area. After I switch profiles to Profile 2 and back to Profile 1 the value goes back to correct 0x20002B3E (within a `masterConfig` structure).

digitalentity, 2016-08-18:
Narrowed the problem down to `pwmInit(&pwm_params);` - before the call value is correct, after it it's set to 0x4000043C. According to CMSIS headers it's a CCMR2 of TIM3 timer.

Hard thing is that the linker rearranges memory and LTO hides some symbols so I don't really know what's in the memory before to `currentControlRateProfile`.

Ok, found it! We define TARGET_MOTOR_COUNT as 6, therefore we have `MAX_PWM_OUTPUT_PORTS` also defined as 6 and we have only six indexes in `pwmOutputPorts` array.

However our `multiPPM` array defined **10** motor outputs when using PPM/SerialRX and `pwmOutConfig` doesn't check for array bounds violation. Therefore everything in close vicinity of `pwmOutputPorts` gets overwritten and corrupted. `currentControlRateProfile` was unlucky to be one of those.

Same is true for `motors[]` and `servos[]` arrays - indexes might get out of bounds when initializing the values.

closed 2016-08-18 by digitalentity

iforce2d, 2016-08-18:
Thanks!
Do you ever sleep??? :)

digitalentity, 2016-08-18:
Err.... no! :smile: 
Just kidding! Although I live in GMT+10 time zone I work for a company with HQ in GMT+3 zone which affects the time when I sleep.

DzikuVx, 2016-08-18:
That kind of explains few things... I was wondering how 9h diff (GTM+1 here) works with Konstantins availability hours...

iforce2d, 2016-08-20:
Tested this today, no problems.

PR #471 (2016-08-18): Fix the memory corruption caused by pwmInit()
Fixes #470