iNavFlight/inav #472: Preparation for conversion to parameter groups

martinbudden, 2016-08-18:
This issue is a task list of the things that need to be done before the conversion to parameter groups can begin. These are non-disruptive changes that can be implemented in their own right, but put the code into a position where the conversion to parameter groups can happen in one or two evenings of solid work.

- [ ] 1. Split `config.c` and `config.h` into their constituent parts, as in Cleanflight
- [ ] 1a. Split eeprom code out of `config.c`
- [ ] 1b. Split features out of `config.c`
- [x] Add in `streambuf` code
- [ ] Split MSP into its constituent parts, as in Cleanflight
- [ ] Convert MSP to use streambuf
- [ ] Add in `parameter_group` code
- [ ] Convert to parameter groups - separate task

digitalentity, 2016-08-18:
Last one is going to be a one hell of a challenge, we'll introduce zillions of bugs on the way :laughing:

martinbudden, 2016-08-18:
Yes, the last one will be the big one.

I've held off doing any work on this, since some of the changes in file structure/ directory structure although safe in themselves are a bit disruptive to the code. I thought it might obscure the work on the memory corruption bug.

Do you think it's OK to proceed now?

digitalentity, 2016-08-18:
Let's branch master -> development and start working on PGs there, I want complete feature freeze in master until 1.2.

martinbudden, 2016-08-18:
OK, sounds good.

No rush to create a development branch - I could do with a little break from this anyway. I'll be glad when it's all done, so I can spend more time implementing new stuff.

digitalentity, 2016-10-04:
Looks like this is finished. Closing.

closed 2016-10-04 by digitalentity