iNavFlight/inav #473: Convert to parameter groups

martinbudden, 2016-08-18:
Dependent on issue https://github.com/iNavFlight/inav/issues/472 -Preparation for conversion to parameter groups

digitalentity, 2016-10-04:
Looks like #472 is done. Unblocking.

digitalentity, 2016-11-03:
@martinbudden what's the progress of this?

martinbudden, 2016-11-03:
I haven't done any work on this for a while. All the preparatory work has now been done, so getting it finished should not be too much work. Probably several evenings of solid work.

digitalentity, 2016-11-03:
Take your time, we'll probably release 1.4 without PGs, I don't see this as a big issue.

martinbudden, 2016-11-03:
One of the motivations was making it easier to port code over from Cleanflight. But now iNav and betaflight are ahead of Cleanflight in most areas that is less important.

I'll probably do PGs in betaflight first, since there are some guys actively working on the configurator there, and also there is more flight testing. Once that is done, porting to iNav should be very straightforward (although there will be adding new iNav specific PGs).

DzikuVx, 2017-01-24:
@martinbudden can we assume this is done and ticket can be closed?

martinbudden, 2017-01-24:
Yes, we can close.

closed 2017-01-24 by martinbudden