iNavFlight/inav #474: Port notch filter from Betaflight

DzikuVx, 2016-08-18:
Port notch filter from betaflight 86626acbdf118f783188e8ae42fdfea985d53e52

martinbudden, 2016-09-08:
@DzikuVx , did you have a pull request for this?

DzikuVx, 2016-09-08:
No, it's only a reminder

martinbudden, 2016-09-08:
OK.

digitalentity, 2016-10-04:
Supporting code for notch filter done. Now we need core to actually apply it to gyro data.

digitalentity, 2016-11-15:
Guys, what do you think? Is this worth pushing into 1.4?

DzikuVx, 2016-11-15:
I see no reason to do it

martinbudden, 2016-11-15:
Yes, postpone this to 1.5. Let's get 1.4 out the door.

miradarya, 2016-12-15:
As this currently doesn't seem to be linked to any milestone, I was wondering if there are still plans to implement the notch filter. I personally think that large quads using inav would benefit greatly from notch filtering due to requiring more D but being noisier at the same time.

closed 2017-01-04 by digitalentity