iNavFlight/inav #479: Align permanent IDs of BOX modes with Betaflight

digitalentity, 2016-08-19:
As @martinbudden pointed out:
There were some differences between the `boxes` in `serial_msp.c` e.g.:
**betaflight**
```
    { BOXAIRMODE, "AIR MODE;", 28 },
    { BOX3DDISABLESWITCH, "DISABLE 3D SWITCH;", 29},
    { BOXFPVANGLEMIX, "FPV ANGLE MIX;", 30},
    { CHECKBOX_ITEM_COUNT, NULL, 0xFF }
```
**iNav**
```
    { BOXNAVWP, "NAV WP;", 28 },
    { BOXAIRMODE, "AIR MODE;", 29 },
    { BOXHOMERESET, "HOME RESET;", 30 },
    { BOXGCSNAV, "GCS NAV;", 31 },
    { BOXHEADINGLOCK, "HEADING LOCK;", 32 },
    { BOXSURFACE, "SURFACE;", 33 },
    { BOXFLAPERON, "FLAPERON;", 34 },
    { BOXTURNASSIST, "TURN ASSIST;", 35 },
    { CHECKBOX_ITEM_COUNT, NULL, 0xFF }
```

When PGs will be introduced some of the configuration might survive the firmware change and it would be wise to keep those parts of the configuration compatible.

digitalentity, 2016-11-17:
@martinbudden @DzikuVx @oleost what do you think? Is this still relevant?

oleost, 2016-11-17:
How many IDs are available?
If I understand correctly this is relevant when copying from cli as backup?

And for OSD etc to read what mode the FC are in?

I'm agreeing in an ideal world the features that are the survives a transfer between different flavors of Cleanflight.

It's not an huge issue however from a user perspective.

But would it be possible to just move all iNav / Nav specific to a free area, example above 50? ( Expect althold and poshold as they already exists in Cleanflight )

DzikuVx, 2016-11-17:
Personally I would ignore this issue completely. 
Alignment would make sense if also CF and BF would change their IDs to keep compatibility with INAV on that level. We can try to align with CF/BF every time they add a new BOX, but does that makes sense?

oleost, 2016-11-17:
It's of course necessary for everyone to try and keep in sync and not only for iNav to do it.

@hydra @borisbstyle

martinbudden, 2016-11-17:
I think generally when doing development we should try and keep alignment with  Betaflight and Cleanflight.

As long as we continue to make that effort, I think we can close this particular PR.

digitalentity, 2016-11-17:
I don't think that aligning specific flight mode IDs is a must have thing. I agree we should align the APIs where possible (mostly MSP to make it easy for 3-rd party tools). I'm closing this for now.

closed 2016-11-17 by digitalentity