iNavFlight/inav #486: Detect and ignore accelerometer if measured vibration is high

digitalentity, 2016-08-21:
In some cases above certain throttle value accelerometer starts to read garbage rendering inertial modes unflyable:
[image]
[image]

Not sure where this effect comes from. Possible causes:
1. Electrical noise when above certain power level ESCs draw currect in spikes causing high-frequency noise.
1. Mechanical noise

In both cases I suspect resonance with MEMS accelerometer natural frequency. Datasheet doesn't specify it for accelerometers but for gyros it's in range 27-40kHz for typical gyros. If I'm correct adding mechanical damping that absorbs high-frequency vibrations and electrical filtering (adding a capacitor to power supply) will help.

In any case the resonance effect is unsafe for the craft and we should be able to detect it and act accordingly.

bk79, 2016-08-21:
That"s the sacred Gral for any closed control loop...and very little can be
done...if the measure is noisy before the reading you can correct it...if
the measire come from the observer(sensor)itself nothing can be done and
you need to change it or an external same sensors which is not affected by
the noise. If the problem is resonance it have to be solved only by
hardware...a typical example of resonance is "tacoma bridge"(look for the
video on youtube) detecting to ignore it doesn't solve the issue and in a
very short time during the flight the aircraft dismatle by itself.

digitalentity, 2016-08-21:
I know we can't solve it in software, but we can detect it and prevent issues with Alt/Pos hold by disabling inertial estimation and relying only on sensors that are less accurate but more reliable at the moment (baro/GPS).

This will degrade performance considerably but will keep the copter/pilot/surroundings safe.

bk79, 2016-08-21:
Physical you can also add a diode in series that will also prevent surges
coming from esc...the problem is finding a diode small and of a lot of
amps...

digitalentity, 2016-08-21:
I'm thinking about adding a lowpass filter to the line powering the FC

digitalentity, 2016-08-21:
Hm... I wonder if this is the cause of problems with airplanes and Naze32 full boards. Baro weight is very small - good enough for stable accelerometer but insacely small for noisy one. GPS-only setups use GPS Z coord/vel to correct the inertial estimate and GPS weight is usually set much higher - enough to keep the estimated position/velocity within sane limits.

Airplanes also usually have much higher vibration levels than multirotors...

bk79, 2016-08-21:
Cap will be 0hz lp and adding the diode in the line will make it better

digitalentity, 2016-08-21:
ESCs can easily draw 20-30A each and 30A+ diodes are heavy and bulky and I suspect it's not electrical noise that is driving accelerometer crazy. I suspect it's a mechanical issue. 

Going out to verify in about half an hour - added silicone pads to soft-mount my FC, will fly and do a few punch-outs to see what do I have in logs. No electrical change so experiment will be clean and correct.

oleost, 2016-08-21:
"disabling inertial estimation and relying only on sensors that are less accurate but more reliable at the moment (baro/GPS)." 

Good idea,  but maybe we should test fly and see how it behaves without the inertial estimator?

bk79, 2016-08-21:
Wilco1969 had it also with no props so i doubt it's mechanica or he should
had the reading with much more less throttle since the rotor spins at much
higher speed for the same throttle. If it is that, damping pads should
solve if the fc is not integrated in the structure..i suppose it's
electrical...an 1n4001series with 1000uf cap in parallel is small enough
for the fc filtering...about detecting the issue you already have the
answer..baro reading... if those big changes in altitude are there also
baro or gps or sonar should feel them... But a good idea is integrate acc
reading and derivate baro or whatever pos reading we take in consideration
so to compare them in terms of velocity

> ESCs can easily draw 20-30A each and 30A+ diodes are heavy and bulky and
I suspect it's not electrical noise that is driving accelerometer crazy. I
suspect it's a mechanical issue.
>
> Going out to verify in about half an hour - added silicone pads to
soft-mount my FC, will fly and do a few punch-outs to see what do I have in
logs. No electrical change so experiment will be clean and correct.
<https://github.com/iNavFlight/inav/issues/486#issuecomment-241239121>, or
mute the thread
<https://github.com/

digitalentity, 2016-08-21:
It's a mechanical issue. Added o-rings to my FC mount and hit 4.3g at full throttle without any garbage radings from accelerometer.
[image]

wilco1967, 2016-08-21:
It's definitely mechanical.... If it where electrical, I would expect it also on the other motors...

What caught me by surprise was the better 'feeling' motor, actually being the one causing the resonance issue. No way I'm going to be able to remove the problem with balancing.. it is critical speed issue.
What about the ACC sensor range ? Is it large enough to never ever be overloaded with the highest peaks? Similar problem happened on MultiWii a few years ago.

Don't worry about the copter disintegrate on flight... It's very small, but enough to make the ACC go crazy... I'm just unlucky to hit the wrong frequency.

wilco1967, 2016-08-21:
I think I will order an SPracing Evo board
That is not part of the frame,so can be mounted on some double sided foam tape.... The same copter with all its vibrations has been flying perfectly before using a Naze on foam.
Then F3FC provides a clean build, but a nightmare to remove vibrations....

digitalentity, 2016-08-21:
I think it's resonance of internal structures of MEMS accelerometer - it's a mass on the spring so it must have a natural frequency and you are unlucky to hit it.
https://www.youtube.com/watch?v=eqZgxR6eRjo

bk79, 2016-08-21:
Try putting o-ring below the motors that should isolate them somehow

bk79, 2016-08-21:
It is indeed very odd...mems have a negligible mass so their resonance
frequency should be above ghz.. Propably you get some spurious freq of the
BPF that is already double the rotation speed or triple if you use 3 leafs
prop

digitalentity, 2016-08-21:
Yes, that might help as well. We are probably fighting very high frequencies here - probably khz or even mhz here. Probably we are hitting a harmonic of propeller noise which is transfered through a stiff frame - this also explains why simply putting rubber o-rings is enough to kill the vibration (which is only possible if vibration is small and of very high frequency).

digitalentity, 2016-08-21:
This [presentation ](http://ocw.mit.edu/courses/electrical-engineering-and-computer-science/6-777j-design-and-fabrication-of-microelectromechanical-devices-spring-2007/lecture-notes/07lecture24.pdf) analyse ADXL50 MEMS accelerometer and estimate resonant frequency of only 24kHz which is very possible with small props or damaged bearings.

DzikuVx, 2016-08-21:
Ha, in my case issues like this were usually mechanical. Faulty motor bearing / not good enough balance / loose frame element / crack in one of the booms / shitty motor mount. Motor noise hitting resonance frequency and bam.

@digitalentity this not only has to be noise hitting resonant frequency of MEMS accelerometer. It can be also noise hitting resonant frequency of anything else

wilco1967, 2016-08-21:
I'll try fitting a wooden arm.... Should have a similar effect... But will have to wait until this evening.... (Wife won't let me...😞)

digitalentity, 2016-08-21:
@wilco1967 actually more (or less) stiff arms might help. Wood is also quite good at isolating high freq. Adding some extra stiffless to the frame may also help as well as adding a rubber/silicone pad under the motors. Possibilities are many, thing is that if this happens again firmware will go crazy again.

We need a way to detect it and react in a safe way, not going heywire and skyrocketing or dropping.

bk79, 2016-08-21:
Mmm...probably bearing related bpf in that case can do little if we think
of a rotation speed of 30krpm bpf for two leaf. Prop should be 10khz a tree
leaf should be there but bearings i mean roll bearing can give also 5x or
6x  rotation speed since there are 5 or 6 or 7 balls

wilco1967, 2016-08-21:
I have 2blade 9x5 HQ props on 900 kV motor... Most tests were flown on 3s I guess I hover at approx 6000 rpm, or 100 Hz. Give or take a bid...
Definitely not a mini quad....

bk79, 2016-08-21:
You had the same issue with no props...dp probably some scratched bearing...

wilco1967, 2016-08-21:
On one motor, but that's replaced

wilco1967, 2016-08-21:
The new motor actually sounds worse, but gives no ACC effect without a prop... I guess it's the arms, rather then the motors

wilco1967, 2016-08-21:
Well... It's the combination, hitting a resonance frequency

bk79, 2016-08-21:
Just my experience i build an hexa for a friend when i was using 5045x3 i
could see the tip of the gps box shaking like in a mixer..just not to talk
on how arms were vibrating....everything disappeared as i swtiched for
5030x3..if you can try to decrease the props pitch and increase size or
number of blades i'm sure it will help

wilco1967, 2016-08-21:
I've ordered an SPracing F3 Evo board.... Don't think I'll ever solve the vibrations without being able to isolate the FC from the frame....
 And wooden arms are OK for test, but I already have an old all wooden copter... Don't need another ugly looking one. 
That old wooden tricopter is using MultiWii navigation version, still working perfect, with enormous amount of vibrations....  OK, not as precise as iNav, but very reliable....

bk79, 2016-08-21:
You don't need to fill it with wood if you could just place a rubber o-ring
like the ones used for plumbing it is more than enough and you won't see
any aesthetic change

wilco1967, 2016-08-21:
Problem is, the F3FC board IS the frame..... Nothing to isolate.... 
I will try to put some foam between the arms and frame instead... If it works, fine... Anyway, that new SPracing board will find some use.... If not on the Tricopter, then on a plane....

bk79, 2016-08-21:
I understood that but what i mean is put an o-ring between the motors base
and the frame so to have each motor sitting on a soft pad

wilco1967, 2016-08-21:
Ahh, ok.... I misunderstood....
Bit difficult the way the motor mount is attached to the arm. The bolts
holding the motor are flush against the sides of the arm, without any
play,so would still transfer a lot of vibrations I guess.
I will put some foam tape between the arms and top and bottom frame plate,
and then not overtighten the bolt that goes through the arm....
So the motor and arm can vibrate, without transferring too much high
frequency vibrations to the frame / FC
Or just use the tape to hold everything together if the bolt still
transfers too much vibrations....
And if done neatly, it doesn't look bad....
With wooden arms, all cables will run outside...

For sure it will change the vibrations, but if it will solve it remains to
be seen.... Will see how it goes....Now if I hurry, and the weather
improves, I might get one flight in this evening.

bk79, 2016-08-21:
Put some glue on the bolts if not fully tight they are going get loosen

wilco1967, 2016-08-21:
OK, that looks promissing.....
Almost all vibration effects on the ACC gone....
[image]

now out for a test flight

wilco1967, 2016-08-21:
Finally SUCCESS !!!!!!

Flew (almost) perfectly..... Alt holt works 99% OK now.... few times saw a small hunch of throttle pulsations, but gone almost immediately if it started... It holds atltitude very well now.....
The rest is optimization in PID parameters.....

Blacklog is also very smooth now.... 

Flight is not optimal yet.... I think I lost GPS for a few seconds but it quickly recovered. And there is a slight hunch of toilet bowling, but that is just because the compass isn't well aligned at the moment... Nothing that can't be solved with some settings & tidying up of wiring.

Now have to do some cleaning up to do....a few drops of loctite on the nuts (thanks BK79) and some tuning / optimization. Luckily those damn vibrations are mostly under control. 

I'll try tightening up the bolts to see how far I can tighten them before the problems return.... atm pretty loose (but tight enough that it will not fall apart in mid air ;-)

THANKS A LOT to ALL of you for your help. Very much appreciated....

The next few days, I'll write a little report on the RCExplorer website about the issues I had.... And what can be done about it.... I'm sure others will run into the same issues when they try this board.... It is just so much more sensitive for vibrations than an external mounted FC.
It wasn't noticable on flightmodes  other than Alt-hold, so most people flying Triflight or cleanflight will perhaps never notice it.....
Will try to convince some Tricopter owners to try iNav... The RCExplorer tricopter is not a racing machine... So I think many people will be interrested in proper GPS functions.

Here is the log file, for comparison.
[LOG00216.TXT](https://github.com/iNavFlight/inav/files/428998/LOG00216.TXT)

The SPRacingF3 EVO will have to find another home once it arrives..... I have a plane lying which would make a good test rig.  New toys

Thanks again

digitalentity, 2016-08-21:
@wilco1967 glad that you managed to resolve the problem. 

We are not fighting low frequency (<200-300Hz) vibrations here so any sort of dampening that will attenuate multi-kHz mechanical noise will probably do equally as good. Another example of how important vibration dampening is for higher-level functions.

Actually, the issue affects Triflight and Cleanflight as well - might cause level drift if exhibited very close to hover - accelerometer is used to correct gyro drift in ANGLE/HORIZON modes and if it measures garbage - horizon level will slowly drift.

Imagine attaching a GoPro to your Tri and having it hover at 1650 throttle where the issue starts to manifect itself. You'll probably be trying to fight the issue like "my tri doesn't fly level, after a few minutes of flight it drifts and I have to land!".

wilco1967, 2016-08-22:
Indeed you would expect this to also effect auto level modes, but it didn't
on any of my tests.... autolevel always perfect.

Now I have to admit, I don't fly pure auto level much....never more than 30
seconds or so....I either use Acro mode or GPS modes. GPS modes have always
worked perfectly (at least, for as long as my GPS plays nice.... new one on
the way).

wilco1967, 2016-08-22:
Update:
Seems like my vibration problems are not completely over yet.....
A lot better than before, but still coming back too often....

Using the slightly heavier 4S battery brings the RPM right back in the critical range.... not as bad as before, but still very noticable....
it's the same vibration issue... not something I can tune out.

Same will happen if I weigh it down with FPV camera etc.

Will put some doublesided tape also under the motors, see if that helps...... 

Looks like the SPracing board will go on the Tricopter after all when it arrives..... trying to deal with this isn't fun.

wilco1967, 2016-08-23:
Ok.... Adding some soft double sided tape under the motors as well did it....
Haven't seen any strange behaviour anymore....
Flew perfectly....
Now I hope the problem doesn't return....

krzysztofmatula, 2016-08-24:
Just a stupid question - how did you secure the motor screws? Did you screw them tight? Is the tape still absorbing vibrations then?

Some time ago I've developed more sophisticated motor mount on my bigger quad, which works pretty well actually, but I'd prefer not to replicate it on my 250:

[image]

wilco1967, 2016-08-24:
I used almost the same idea..
Some Scotch double sided clear mounting tape below and on top of the arm, and the motor bolts not too tight....
This stuff is quite rubbery. Will see how long it stays this flexible..

lkaino, 2016-09-07:
@wilco1967 I'm now having similar issues with ACC noise that you described. It would be nice if you wrote a summary of your finding to RCExplorer forums as you mentioned. Would be good information for any others thinking of running iNav on the tricopter board.

wilco1967, 2016-09-07:
Hi Ikaino,

I haven't flow it for almost two weeks now, because on a business trip... but I managed to write a 'small' story on the RCexplorer website....
[http://rcexplorer.se/forums/topic/f3fc-inav-experiences/](url)

digitalentity, 2016-11-15:
I have some things in the works, but not for 1.4

oleost, 2017-02-08:
Is this a bug and is it critical @digitalentity ?

Seems like an enhancement ?

digitalentity, 2017-02-08:
I agree, it's not a critical issue.

digitalentity, 2018-05-12:
Ok, time to look into this. It appears to be triggered by high vibration, causing internal ADC saturation. Changing acc scale to 16G makes problem less prominent (https://github.com/iNavFlight/inav/pull/3203)

Also, DLPF on Invensense chips seems to be implemented in such way that it's processing analog signal from the MEMS itself before the ADC. Reducing DLPF to 42Hz made accelerometer resonance issues completely disappear on the bench. I still have to confirm that in logs in flight.

Lowering DLPF will definitely affect ACRO performance, but it's a possible software workaround for the problem. I'm also working on some code to detect the vibration levels.

wilco1967, 2018-05-13:
Hi Konstantine....

It just happens that my old tricopter developed some vibration issues again after a crash. Guess the prop adapter got slightly bent, so it wouldn't properly hold altitude anymore. sometimes, at certain RPM, it would shoot up....
replaced the prop adapter, and much better already, but still it happened sometimes....

So I just tried your newest update. Seems better, but difficult to tell for sure.... After I replaced the prop adapter, it already improved a lot, but sometimes, it still surged. With latest firmware, I didn't really notice any surge.

attched two blackbox log files.... one with iNav 2.0.0 development from 3 days ago (log21)... one brand new (log 23).... both are already with the replaced prop adapter, so only change is firmware version.

https://drive.google.com/open?id=1yQi3tQmR1GpOYCZRGD2oK50vnHBrdLl6

digitalentity, 2018-05-13:
@wilco1967 what `gyro_hardware_lpf` setting are you flying with?

wilco1967, 2018-05-13:
Whow, that's fast. ;-)

set gyro_hardware_lpf = 256HZ

here's the complete dump....
`# version
# INAV/OMNIBUS 2.0.0 May 13 2018 / 10:32:34 (56ca670b)

# resources

# mixer

mmix reset

mmix 0  1.000  0.000  1.333  0.000
mmix 1  1.000 -1.000 -0.667  0.000
mmix 2  1.000  1.000 -0.667  0.000

# servo mix
smix reset

smix 0 5 2 100 0

# servo
servo 0 1000 2000 1500 100 
servo 1 1000 2000 1500 100 
servo 2 1000 2000 1500 100 
servo 3 1000 2000 1500 100 
servo 4 1000 2000 1500 100 
servo 5 1000 2000 1500 100 
servo 6 1000 2000 1500 100 
servo 7 1000 2000 1500 100 

# feature
feature -RX_PPM
feature -VBAT
feature -TX_PROF_SEL
feature -MOTOR_STOP
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature -FW_LAUNCH
feature -TRACE
feature VBAT
feature GPS
feature TELEMETRY
feature CURRENT_METER
feature BLACKBOX
feature PWM_OUTPUT_ENABLE
feature OSD

# beeper
beeper RUNTIME_CALIBRATION
beeper HW_FAILURE
[... 656 lines cut ...]
set nav_mc_vel_xy_i = 15
set nav_mc_vel_xy_d = 100
set nav_mc_heading_p = 60
set nav_fw_pos_z_p = 50
set nav_fw_pos_z_i = 0
set nav_fw_pos_z_d = 0
set nav_fw_pos_xy_p = 75
set nav_fw_pos_xy_i = 5
set nav_fw_pos_xy_d = 8
set nav_fw_heading_p = 60
set thr_mid = 50
set thr_expo = 0
set tpa_rate = 20
set tpa_breakpoint = 1650
set fw_tpa_time_constant = 0
set rc_expo = 70
set rc_yaw_expo = 20
set roll_rate = 55
set pitch_rate = 48
set yaw_rate = 20
set manual_rc_expo = 70
set manual_rc_yaw_expo = 20
set manual_roll_rate = 100
set manual_pitch_rate = 100
set manual_yaw_rate = 100`

digitalentity, 2018-05-13:
Try going down to 98HZ or even 42HZ. You might have to change tuning to account for larger filtering delay. On my testquad lowering gyro hardware LPF helped a lot.

wilco1967, 2018-05-13:
Ok... I'll try 42hz and report back....

btw, I also adjusted (just now)
set acc_notch_hz = 80
set acc_notch_cutoff = 50
(was 0 / 1)
not visible in configurator, but just noticed them in CLI dump...

wilco1967, 2018-05-13:
Alright.... did another testflight (actually 2, but forgot to insert SD card first time ;-)

What can I say..... it Flies perfect.... 
Not a hint of surging.... holds altitude perfect....

not sure if the 42Hz caused it, or the ACC notch filtering.... had 2 slight peaks in ACC_Z anaylser display and those are gone now. ACC_Z trace is now absolutely smooth...

[LOG00001 42Hz + ACC notch 80_50.TXT](https://github.com/iNavFlight/inav/files/1998352/LOG00001.42Hz.ACC.notch.80_50.TXT)

Not needing to re-tune anything.... For my flying style, I don't think I would notice anyway... 
besides, the software gyro_lpf_hz = 30 should have a similar effect I assume ?

Will try again with ACC notch filter disabled again.... to be continued...

wilco1967, 2018-05-13:
Ok, notch filter disabled doesn't make much difference...... Still the ACC_Z trace is very smooth (just slightly more noisy then with notch enabled).

So the 42Hz seems to have the biggest impact....
[LOG00001 42hz ACC notch disabled.TXT](https://github.com/iNavFlight/inav/files/1998372/LOG00001.42hz.ACC.notch.disabled.TXT)

I'll switch the ACC notch filter back on.... pretty happy with how it flies now...

Thanks a lot !

digitalentity, 2018-05-13:
I did some investigation and it appears that vibration causes unfiltered acceleration values to be clipped to match the sensor output scale (8G/16G). 

However, I believe DLPF inside the sensor is able to correctly read and process the out of range values without saturating and clipping and make filtered values fit in sensor output scale range. If we set DLPF low enough high amplitude/high frequency vibration will be attenuated and smoothed and we'll see correct values as output of the sensor.

Price for this is increased delay in both acc and gyro since DLPF setting affects both. The usual tradeoff between robustness and performance 😄

wilco1967, 2018-05-13:
Just did another test, with the earlier firmware (8G), ACC_LPF at 42Hz and ACC notch filters active again....
[LOG00001 ACC 8G 42Hz ACC notch 80_50.TXT](https://github.com/iNavFlight/inav/files/1998394/LOG00001.ACC.8G.42Hz.ACC.notch.80_50.TXT)

ACC_Z is still smooth, so for me, it doesn't seem to matter if we run at 8G or 16G range.... the 42Hz gyro_LPF setting seems to make the difference.

but my highest ACC_Z peaks (mostly unfiltered) never went up much above 2G anyway.... So I guess for the kind of vibrations I'm seeing, it wouldn't matter.. But I can see it making a difference for copters which show (or not show, but still encounter) peaks above 8G.

I couldn't notice any downside in 16G setting, but my flying is not 'race style', on a rather large tricopter (RXExplorer V4 tricopter).

I'll be reflashing to latest 16G version for now....

Thanks again !

digitalentity, 2018-05-13:
@wilco1967 we don't see actual mess in the logs since our accelerometer always goes through LPF (and maybe notch) before being used or logged. 

Vibration/clipping detection code (#3210) operates on raw accelerometer data. I'll appreciate if you can do several tests with #3210 with 42HZ and 256HZ (and if you are brave enough - trigger the skyrocket bug). We need data to create a baseline for acceptable acceleration.

wilco1967, 2018-05-13:
Ok... will try (compiling now....)
this is using git checkout de_acc_vibre, correct ? (not so good at github.... it's more like monkey see, monkey do for me ;-)

what do you mean with 'skyrocket bug' ?  (sure going to try... if I knew what it was... ;-).

digitalentity, 2018-05-13:
Yep, `de_acc_vibre` branch it is.
Skyrocket bug is the event when `ALTHOLD` shoots up into the sky

wilco1967, 2018-05-13:
Ah, ok..... got it....
but it didn't happen again when filtering at 42.... will try de_acc_vibre branch with 256hz , no notch first.

[9 more replies]

closed 2019-05-24 by DzikuVx