iNavFlight/inav #488: Changed definition of MAVLINK breaks stuff

stronnag, 2016-08-21:
Changing the definition in a gratuitously incompatible way
````
FUNCTION_TELEMETRY_MAVLINK   = (1 << 9), // 512
````
from the prior 256 **without announcement** breaks lots of stuff:

* Working configurations
* The configurator (hangs on startup if 512 set in the CLI)

Is it too hard to make these changes in a backwards compatible fashion ?

digitalentity, 2016-08-21:
@martinbudden what's FUNCTION_PASSTHROUGH ?

martinbudden, 2016-08-21:
@digitalentity , I presume it is to do with serial passthrough, that is the ability to reprogram the ESCs via the configurator. I also presume it was added by betaflight, but why it was not added on the end, I don't know

stronnag, 2016-08-21:
Ah, so we have a real mess if this is embedded in the closed source bllheli application?  And maybe have no choice other than an ABI break in iNav?

digitalentity, 2016-08-21:
I suggest we merge #489 to keep things backward compatible and look at `FUNCTION_PASSTHROUGH` when we start to actually merge it into INAV. I'm going to stick to [YAGNI](https://en.wikipedia.org/wiki/You_aren%27t_gonna_need_it) for now - passthough works fine wighout it so we probably don't need the `FUNCTION_PASSTHROUGH`, at least not yet.

stronnag, 2016-08-21:
Indeed, the patched Cf configurator that embeds the BLHELI protocols for flashing / configuring ESCs works with MAVLINK == 256.

digitalentity, 2016-08-21:
Thus I don't see a reason to keep `FUNCTION_PASSTHROUGH` breaking the established ABI.

digitalentity, 2016-08-21:
Merging #489

martinbudden, 2016-08-21:
Agreed. It's annoying that `FUNCTION_PASSTHROUGH` was not added in a compatible way.

It's all probably going to be moot soon anyway - there is a blheli-configurator under development which would supersede passthrough anyway.

PR #489 (2016-08-21): Change MAVLink function mask back to 256
Fixes #488
closed 2016-08-21 by digitalentity