iNavFlight/inav #491: Post 1.2rc2 loses target max waypoints

stronnag, 2016-08-21:
Appear not to be picking up target defines. Max 15 on all platforms.

digitalentity, 2016-08-21:
Not sure I can confirm the issue. The #define from target.h is the one being used in compilation. What target did you check?

stronnag, 2016-08-21:
Apologies, EVO. Where it is missing. Other than that. EVo is now working well on all nav modes. No evidence of #431 on that target on my tricopter.

digitalentity, 2016-08-21:
@stronnag do you have a spare SPRacingF3 to try replacing on your quad and see if #431 is still there? Same setup, same board family, but with a board from different batch.

stronnag, 2016-08-21:
I have the quad here too. Still affected on the gps-debug branch. I'll reflash master and try that and report all findings in #431 when I get home.

digitalentity, 2016-08-29:
Fixed

closed 2016-08-29 by digitalentity