iNavFlight/inav #498: sensorfusion - crosstalk b

robert-b, 2016-08-23:
https://www.youtube.com/watch?v=sV-VEtVUOt0&feature=youtu.be
https://youtu.be/ZnZAZw-hg-g

those simplifications in the sensorfusion have some impact.
so pls. revert the simplifications.

oleost, 2016-08-23:
https://www.youtube.com/watch?v=HbRV-RvBSP8

Also tried with more yawing etc, no issue att all.

Can you please share your dump?

digitalentity, 2016-08-23:
@robert-b the problem is that you probably exceed the 2000deg/s rotation rate - the maximum value gyroscope is able to detect.

robert-b, 2016-08-23:
ok, 2000 dps is 0.18 seconds for a full rotation.
and in the 2nd i tried not be fast.

and the px4 kalman filter has no problems with such tests.

oleost's video is the wrong approach. turning one axis is no problem.
the error calclulated and corrected in pi controller of the sensorfusion needs 'components' of all exis.
then you see, that the level of trust for the accelerometer is to low.
tuning the value to high and you get problems on the yaw axis.

cli - i just do some desktop flying and use the standard settings given - no value changed.
when i set the looptime to 1000 it's a tad better.

oleost, 2016-08-23:
I already wrote I have done it with turning on all axis as well.  I just didn't bother making a video. 

Could you share your dump?

robert-b, 2016-08-23:
the board is already flashed with another piece of sw.
the cli settings are standard - so just press reset in the gui
and you have what i have with regards to the configuration.

oleost, 2016-08-23:
Then you need to calibrate your accelerometer as I already told you. Information on how to do that is in the wiki.

robert-b, 2016-08-23:
normally i do calibrate all sensors.
for desktop flying this is not really required. the resulting sinusoidal swing on roll and pitch is negligible.
for me the important thing is the working of the pi controller for stable and painless operations.
have a look and you will see how the things are put together.

robert-b, 2016-08-23:
before i forget - have a look at the mahony papers.
there you see, that only the normalised accel vector is taken into account for the calculations.
the additional info from the six point accel calibration goes down the toilet.
also think why the normalised accel is being used.
if you compare the simplification used within the inav and the paper you can't say that the 'inav' way is correct. you will notice this when flying loopings with a few G.
i am only interested in safe and painless flying.

digitalentity, 2016-08-23:
@robert-b Mahony assumes that accelerometer readings are not biased, however in reality they are meaning the your sensor is not measuring exactly 1G in downwards direction when stationary and flat. This is why 6-point calibration is mandatory step. Feel free to reopen the bug if the issue persists after you re-try with properly calibrated sensors.

closed 2016-08-23 by digitalentity

robert-b, 2016-08-24:
set accgain_x = 4107
set accgain_y = 4091
set accgain_z = 4056

would that be ok?
if yes, you need to look in depth at the mahony papers.

digitalentity, 2016-08-25:
I'm not talking about scale, I'm talking about bias.

If you accelerometer is measuring (0, 0.10, 0.85) when completely flat and stationary accelerometer will be ignored most of the time because |A| <> ~1G. It will only be used when moving UP at 0.15G acceleration and will yield a 84deg tilt while the craft stays completely level.

Please make sure you keep your sensors properly calibrated and not exceeding their maximum limits. Please also do your "desktop flights" while the board is armed and record blackbox logs to provide us with real sensor data to investigate.

robert-b, 2016-08-25:
strange sw solution.
any requirements to do so?

robert-b, 2016-08-30:
no requirements to so - just remove the wrong behaviour.
in case the accelerometer goes into saturation you can move to gyro only mode.
but not when the magnitude of the accelrometer is close to one G.

digitalentity, 2016-08-30:
I don't see an error in the code. Please make a PR and we'll discuss the proposed change.

robert-b, 2016-08-31:
#define MAX_ACC_SQ_NEARNESS 60 // works better

        // Just scale by 1G length - That's our vector adjustment. Rather than 
        // using one-over-exact length (which needs a costly square root), we already 
        // know the vector is enough "roughly unit length" and since it is only weighted
        // in by a certain amount anyway later, having that exact is meaningless. (c) MasterZap
        ax *= (1.0f / GRAVITY_CMSS);
        ay *= (1.0f / GRAVITY_CMSS);
        az *= (1.0f / GRAVITY_CMSS);
is a pain.
the sqrtf functions acts as a filter and dampens vibrations.
the 2nd point is calculations are only 'valid' with a normalised accel vector.

the 'meaning less' means not understanding what is up.

digitalentity, 2016-08-31:
```
#define MAX_ACC_SQ_NEARNESS 60 // works better
```

That will accept ranges from 0.63g to 1.26g which means much higher drift in presence of static accelerations which actually happen quite frequently.

As for the 
```
    ax *= (1.0f / GRAVITY_CMSS);
    ay *= (1.0f / GRAVITY_CMSS);
    az *= (1.0f / GRAVITY_CMSS);
```
I don't see how this affects accuracy.
Magnitude of an error vector is 
`|e| = |a|*|estG|*sin(error)`
where error is an angle between estimated and measured gravity. 

Not properly normalising measured acceleration will result in slight mistake in calculation of error vector **magnitude** but won't result in incorrect error vector **direction** (this is a nature of cross product). Therefore this simplification won't cause attitude estimation to be incorrect it will only affect time to converge to measured gravity vector.

robert-b, 2016-09-01:
yes, but vibrations ... the cross produkt pulses due to the change og the magnitude.
tested ...
handycrafter ...

digitalentity, 2016-09-01:
@robert-b 
Hm... good point. Didn't though about vibrations. Doesn't explain drift in your case - you didn't have any vibrations.

robert-b, 2016-09-01:
in case sqrtf causes problems - use a taylor's expansion for 1/sqrt(x*x).
this would make more sense for me.
the 2nd try is not to weight the accelerometer but use the 'full' error until the result of the crossproduct does exceed 1/4G (for one of the vector elements).

digitalentity, 2016-09-01:
I would really like another oppinion here. I don't see much ill effects in keeping things as they are now.

martinbudden, 2016-09-01:
Which is the PR that made the simplifications? I'll have a look at that.

digitalentity, 2016-09-01:
@martinbudden simplification was introduced in https://github.com/iNavFlight/inav/commit/8860c9034baec587cdc3ceaa671b48308fbd16d9

@robert-b I recall you complaining in https://github.com/iNavFlight/inav/issues/447 about suboptimal implementation of the same part of the code and nothing was related to logic correctness but merely a suboptimal implementation.

robert-b, 2016-09-01:
another question:
void imuTransformVectorBodyToEarth(t_fp_vector * v)
{
    float x,y,z;

    /* From body frame to earth frame */
    x = rMat[0][0] * v->V.X + rMat[0][1] * v->V.Y + rMat[0][2] * v->V.Z;
    y = rMat[1][0] * v->V.X + rMat[1][1] * v->V.Y + rMat[1][2] * v->V.Z;
    z = rMat[2][0] * v->V.X + rMat[2][1] * v->V.Y + rMat[2][2] * v->V.Z;

    v->V.X = x;
    v->V.Y = -y;
    v->V.Z = z;
}

void imuTransformVectorEarthToBody(t_fp_vector * v)
{
    float x,y,z;

    v->V.Y = -v->V.Y;

    /* From earth frame to body frame */
    x = rMat[0][0] * v->V.X + rMat[1][0] * v->V.Y + rMat[2][0] * v->V.Z;
    y = rMat[0][1] * v->V.X + rMat[1][1] * v->V.Y + rMat[2][1] * v->V.Z;
    z = rMat[0][2] * v->V.X + rMat[1][2] * v->V.Y + rMat[2][2] * v->V.Z;

    v->V.X = x;
    v->V.Y = y;
    v->V.Z = z;
}
why is there a need to change the sign?
never seen this before.

and there is still one more thing :-)

digitalentity, 2016-09-01:
INAV uses NEU coordinate system, while IMU and sensors use NED

robert-b, 2016-09-01:
you think so.
to make shure:
- ned: north east down
- neu: north east up
so i would expect a sign change on the z-axis and nothing else.