iNavFlight/inav #500: hi i'm new to this  so patience please my naze32 rev6 not holding position it holds alt and level but wanders of to werever

harryaruba, 2016-08-23:


oleost, 2016-08-24:
Hi.

Please start here, and read trough the wiki. https://github.com/iNavFlight/inav/wiki

If you still dont figured it out after reading trough the wiki, you can go to the rcgroups thread for user support. http://www.rcgroups.com/forums/showthread.php?t=2495732

Please close this Issue.

andre92, 2016-08-24:
Does the copter stay perfectly level and drift with the wind, or does it fly off into a direction that is not home?
In the first case, gps may not be activated properly.
In the second case, check that your compass is working properly and is calibrated. GPS needs to know what is the right direction to go, or it will fly in a wrong direction.

closed 2016-09-01 by digitalentity