iNavFlight/inav #501: Acc calibration is confusing

mhv-shared, 2016-08-23:
Not sure if this belongs here or in the configurator repo. Couple things about the accelerometer calibration really confused me.

1. There is no indication of the existence of 6-point calibration, except for the wiki. Until I performed it, I was getting slow pitch/roll drift at rest (which didn't happen with cleanflight). Instructions and an indication of success would be useful (Ardupilot/PX4 does a pretty good job with this).

2. My FC is mounted with a yaw of 45 degrees, which I entered on the configuration tab. After doing so, the 3D rendering on the Setup page appeared correctly aligned to my frame, but the calibration would not take effect until I tried doing it aligned to the flight controller instead.

digitalentity, 2016-08-23:
1. I'm not a JavaScript expert, so unless somebody implements a Configurator part of the code we'll have  to stick with what we have.
 1. You calibrate the sensor itself, not the aircraft. To avoid rounding issues calibration is applied prior to rotating the board so it has to be done relative to the board, not to the aircraft. BTW APM works the same way.

mhv-shared, 2016-08-23:
1. Ok - I can look into that and see what would be required. It seems like it wouldn't be difficult to at least log some stuff in the configurator message window indicating the stage of calibration and success/failure to converge.

2. That makes sense, but the wiki instructions (and illustrations) refer to orienting the copter, not the sensor. I realize this is a fair unusual situation, but a note would be helpful. BTW, am I correct in assuming this was the problem? I was also confused because this passage would seem to indicate it should still work:

> There is no need to place copter perfectly aligned, the algorithm does not care about exact positions as long as they are close to 90 degree apart and copter is stationary in every position.

oleost, 2016-08-24:
Wiki mentions that its the flight controller itself, not the aircraft position that matters. 

> Note: If the flightcontroller is mounted in another angle or upside down, do the calibration steps with the flightcontroller pointing as shown in the pictures, not the quad (otherwise calibration won´t work).

Also the qoute you a reffering to is meant that it doesnt matter if the quad is 89 degree or 91 degree instead of exactly 90 degress when performing calibration. Just close to which ever angle is requested.

mhv-shared, 2016-08-24:
Ah ok, I guess that note won't help with people not reading carefully then :)

lkaino, 2016-09-03:
For some reason I'm not able to get the accelerometer calibration working on the latest RC. This is first time I'm doing it on iNav so it might be that I just don't know how.

I stored the calibratedAxisCount to a debug variable and it never gets a value greater than 1. The function performAcclerationCalibration() is called 400 times, after which the variable calibratingA gets to zero and the function is no longer called. By looking at the code I don't understand how it would ever proceed calibrating the other axises as the calibratingA is never set back to 400 after the initial set from MSP.

digitalentity, 2016-11-21:
@DzikuVx do you think we can hack something re. calibration for 1.5?

DzikuVx, 2018-01-21:
Since next version of Configurator will have improved calibration procedure, I'm closing this one. Thanks

closed 2018-01-21 by DzikuVx