iNavFlight/inav #502: Slow Stick reaction in angle mode

MPC561, 2016-08-24:
Target: F3 Evo
Version: 1.2rc2
Frame: Tweaker 210, Racerstar 2205 2300kV

It seems that since "PID: Use full stick in ANGLE/HORIZON (#311)" was introduced the reaction for stick input (pitch/roll) is very slow. As the roll and pitch rates are only used in acro mode, there seems to be a kind of "rate" missing to configure the reaction on stick in angle.

DzikuVx, 2016-08-24:
It might looks like reaction is slower since right now full stick range is used, not around 50% or less of it. You can increase LEVEL_P and/or increase angle inclination but better to learn how it works like. And it works like it should be working.

digitalentity, 2016-08-24:
Now you get `max_angle_inclination_*` at full stick deflection and they are defaulted to 30deg, increase them to your liking, up to 80-85deg is probably fine.
As @DzikuVx suggests you might want to increase LEVEL_P to increase the reation speed but at first make sure you ROLL/PITCH PIDs are properly tuned in ACRO mode, otherwise you might end up with oscillations induced by quad overshooting the desired angle target.

MPC561, 2016-08-24:
max_inclanation roll pitch was already increased from 30° to 45°. Still slow. Level P I will try, although I understood it in that way that it is the strength to get back in level position and not to react faster to get the copter to max inclination.

@digitalentity 
I will try 85°.

But, forgive me. What should I do when I want a max inclination of 45° and a fast stick reaction?

digitalentity, 2016-08-24:
LEVEL P is how fast the copter will attempt to reach the desired angle, not return to center. Rule of thumb here: 

`desired_rate[dps] = p_level / 65.6 * angle_error[deg] * 10;`

So when hoverling level and banging the stick to the edge (45 deg) copter will start with: 20 / 65.6 * 45 * 10 = 137 deg/s rate. If it's slow to you - increase LEVEL P

MPC561, 2016-08-24:
Understood! Will do that!

closed 2016-08-24 by MPC561