iNavFlight/inav #505: Angle mode strange return to horizon

xdigyx, 2016-08-24:
Recently while flying in angle mode I have noticed that while I try to lean my hexacopter 680 to one side then it's moving smoothly which is as it should (pids follows rcCommand strainght away):
[image]
however after releasing pitch/roll stick, at the first stage my copter is rapidly moving back to level and it can be seen that PIDs reaction is delayed approx 100ms of rcCommand and raises very fast. Although I should have done it gently and I did not, I believe is not quite as it supposed to be...
[image]

Anyone got same effect?

digitalentity, 2016-08-31:
Do you have the complete log?

xdigyx, 2016-08-31:
Sent on your email

xdigyx, 2016-08-31:
To be more precised, it's FW 1.1. On 1.2 R4 is much better:
[image]

DzikuVx, 2016-08-31:
@xdigyx there is an LPF filter on ANGLE and HORIZON modes. Its cutoff frequency is set by LEVEL I parameter. If it's low, it can induce delay like you described.

xdigyx, 2016-08-31:
Sorry, but I can't catch you... sorry for my newbie questions:  What's  reducing the delay?  what LPF filter, or level I gain or cut-off freq filter (which one)?

DzikuVx, 2016-08-31:
INAV uses low pass filter in ANGLE and HORIZON modes to "smoothen" difference of target and current inclination. Cutoff frequency of this filter is set by LEVEL_I parameter. Default LEVEL I 15 means that cutoff is set at 15Hz. 
The lower the cutoff, the longer the delay filter introduces and smoother is leveling in those modes.

xdigyx, 2016-08-31:
OK, got it now. So just the question why (or rather if it's normal) there were no delays while increasing inclination, but there were while decreasing/levelling... Was level i gain rescalled 1.1 vs 1.2? Asking because either for 1.1 & 1.2 I had default LVL I gain.

oleost, 2016-08-31:
There is no I gain for level mode. 

The I gain is in reality cutout frequency

DzikuVx, 2016-08-31:
How it behaves depends how fast you move a stick.  Usually, pilots release them faster than they set inclination. 
This value was not rescaled, but 1.2 changed many things, including how sticks behave. Inclination change is smoother now, so delay might not be visible

xdigyx, 2016-08-31:
Thanks for explanation.
So either it could be omitted and issue closed or maybe defaults are not properly set.

digitalentity, 2016-08-31:
@xdigyx, defaults are sane and good enough to get you in the air on most aircrafts, to get your machine flying perfectly you'll have to spend some time fine-tuning it.

Since you issue was related to 1.1 and 1.2 really changed flight dynamics code a lot, I think we can close this.

closed 2016-08-31 by digitalentity