iNavFlight/inav #507: inav 12.RC2 does not log to SD on SP-EVO

stronnag, 2016-08-26:
Initial report on RCG: Originally Posted by chuckstarck 
Using SP Racing EVO, SD board on bottom of controller.
1.2 RC
I've cleared and formatted the card, then after a flight I get a scrambled FAT and windows wants the card formatted, that's over and over.
So decided to put a couple files on the card first, that worked one time, got a BB file, not useful but got a file.
It writes "LOGS" and "FREESPACE" and leaves the .jpg file there but there is nothing in the LOGS folder
Is this working for anyone?
It worked with 1.1 and CF but now with 1.2, just driving me nutz along with having a very hard time getting the PID"s nailed down. I can't even find the nails to do it.
If I could get BB to work, then perhaps I could get some help with the wobble/wobble/wiggle/wiggle.

thanks in advance

stronnag, 2016-08-26:
Confirmed. In my case, no logs are written; however there is no file system corruption.
The same FC / SD card combination works OK with CF 1.13 (valid logs are written), as well as with master iNav using a serial logger.

tested on  2524736

digitalentity, 2016-08-27:
Confirmed. Issue is with `sdcard_poll()` always returning `false` at certain point. Investigating further.

digitalentity, 2016-08-27:
My SDCard is stuck at `SDCARD_STATE_SENDING_WRITE` which happens after logging starts. Dumb code compare between CF and INAV confirms that SDCard driver and FAT implementation are the same so the reason is deeper.

digitalentity, 2016-08-27:
Compiled out NAV, LEDSTRIP, I2C, ADC, EXTI and SDCard is still stuck at `SDCARD_STATE_SENDING_WRITE`. After disabling more features I finally made the board hang on boot. Good sign, as now I can try to debug where the lockup happens.

digitalentity, 2016-08-27:
Dang, nothing useful, a rogue call to `i2cRead()` from mis-enabled hmc5883 driver. Still no clue why SDCard is stuck at `SDCARD_STATE_SENDING_WRITE`. Will try to see if it's working with Cleanflight.

stronnag, 2016-08-27:
I found it working with CF  1.13 and the whole fc hangs on arm with CF master of  2016-08-26.

stronnag, 2016-08-27:
P.s. 5883 is useful for equipped gps

digitalentity, 2016-08-27:
I suspect we are missing the proper GPIO initialisation for clock low mode:
CF does this:
https://github.com/cleanflight/cleanflight/commit/98133e532efb972ac07d7833d0cf38297dc0d034#diff-8345889f754729d3d1311535aacdfa06R79

while we dont do any specific GPIO init when using SPI for SDCard

digitalentity, 2016-08-27:
Nope, didn't help as well.

digitalentity, 2016-08-27:
Ok, we don't enable RCC for SDCard DMA. Enabled manually and now SDCard is not stuck at SDCARD_STATE_SENDING_WRITE anymore, but logs still aren't being saved.

digitalentity, 2016-08-27:
Findings so far:
1. In INAV clock for SD-Card DMA is not get enabled so it actually never worked.
1. For some reason it doesn't work w/o DMA either, possibly a bug in blackbox_io code. Worth trying creating a file manually in the code and checking if it gets written to.

It's too late to continue investigating today. Will continue tomorrow,

digitalentity, 2016-08-28:
Nailed it!

stronnag, 2016-08-28:
Bench tested. Valid BB log is now generated on arm / closed on disarm.

PR #513 (2016-08-29): SDCard & I2C bugfixes
 1. Fix uninitialised I2C usage on F3
 1. Fix incorrect SPI pullup init when using SDCard
 1. Fix DMA clock not being enabled for SDCard
 1. Fix Blackbox deleting the file after finishing the log

Closes #507
closed 2016-08-29 by digitalentity