iNavFlight/inav #508: Stability issues with SP racing Evo

feakesp, 2016-08-26:
Apologies if I have raised this issue incorrectly
I have loaded RC 1.2 on my SOP racing EVO ...
After approx 1 minute of flight the Quad becomes unresponsive to sticks just hovers ... after a few seconds it then looses total control and crashes ..
To try to eliminate possible issue I have completed the following

Re-loaded cleanfight on FC -  flew without issues
Re-loaded iNav 1.2 with very limited functionality ( LED-strip and GPS turned off) - flew without issues
Turned LED-strip back on -  problem returned ( no response, followed by crash)

stronnag, 2016-08-26:
Interesting. I've put perhaps 10 packs through my EVO without issue, but I'm not using LEDSTRIP.

feakesp, 2016-08-26:
I have re-configured the LEDs with a basic set up and will try again .. I had them set for arm-state/indicators/ throttle on one strip and Indicators/colour/warning on the other...

feakesp, 2016-08-26:
I also noticed that telemetry was also rather erratic at the time .. with Vbat continously jumping from 10 - 14 volts ( on a 3s batt ) not sure is this is related ...

xdigyx, 2016-08-30:
you must scale Vbat via CLI

feakesp, 2016-08-30:
Vbat has been set up in the GUI and always reads steady at expected voltage - however when flying the Vbat via telemetry jumps fluctuates between 9.v and 14.5 volts!
Buzzer also beeps so I assume it is the Vbat function, rather than issues just with telemetry - hovere I never see any fluctuations in configurator, always stay steady!
I had been making progress by reducing the LED-strip functions to just arm on one strip and colour + warning on the other!  With just this basic set-up it flew well .. 
I have considered an issue with the 5v from the BET, however I have no issues while flying the same config in Cleanflight ... All very odd.

However my testing has now come to a halt, as the Evo failed during flight and now the Blue light does not light on power up ...   I just wonder if I have been suffering a Vbat issue, causing the problems

stronnag, 2016-08-30:
On my EVO, vbat jumps around more than on other FCs, maybe +/- 0.2v on 3S, so nothing like as extreme as your example, but noticeable after SPRF3, Dodo, Naze.

digitalentity, 2016-09-01:
@feakesp can you check if LEDStrip works with RC4?

feakesp, 2016-09-01:
digitalentity .. Unfortunately My Evo failed on me during the LED testing with RC3 - hope to get a replacement very soon to continue my testing ....

digitalentity, 2016-10-18:
@feakesp any news? Is this issue still present in 1.3-RC1?

closed 2016-11-21 by digitalentity