iNavFlight/inav #512: GPS glitch detection improvement

xdigyx, 2016-08-27:
Let me come back to this topic as the issue #431 has moved a different direction.
As I already mentioned formula
(INAV_GPS_GLITCH_RADIUS + 0.5f * INAV_GPS_GLITCH_ACCEL * dT * dT)
will only work for certain model speed as either INAV_GPS_GLITCH_RADIUS and INAV_GPS_GLITCH_ACCEL are simply contants.

My proposal is to change it  as follow:
1. calc NewDistance between two points (lon,lat), mean previous (good values) and new reading location
2. if NewDistance < INAV_GPS_GLITCH_RADIUS(GPSmeasurementStaticError + 1.0f*sqrtf(sq(lastKnownGoodVelocity.V.X)+sq(lastKnownGoodVelocity.V.Y)) * dT) then accept the GPS readings
else
or skip it or replace it with predictedGpsPosition. Probably could be skipped if velocity is less than 1m/s.

where
GPSmeasurementStaticError - normally GPS error is approx 0.5-10[m] depending of GPS type/model
INAV_GPS_GLITCH_RADIUS - a constant, say 1.20f

digitalentity, 2016-08-29:
This will not always work. If the copter starts rapidly accelerating from zero velocoty then we might get outside of INAV_GPS_GLITCH_RADIUS and never get back in. We need something smart enough to detect this situation.

xdigyx, 2016-08-29:
That would need to be veeeery powerfull and light copter to pass a distance of (say) 3m from start (0m/s) within 200ms (5Hz)... Even though GPS speed is nearly never 0m/s.
And even more: I can not see a chance that it would change its speed itself that much.
Such speed changes are possible only when an operator is actually controlling the copter (by moving a stick), so under this circumstances FC does not control the flight and GPS readings I believe are for indication only.

digitalentity, 2016-08-29:
GPS readings are never for indication only. They are always used to
estimate position and if there's a chance that algorithm will start
permanently ignoring GPS readings without a possibility to recover - it's
not reliable.

xdigyx, 2016-08-29:
Hm...OK, so we also need to take a care about predicted position, but there applies rather aside of PH mode where glitch is most visible. So maybe in PH mode at least INAV_GPS_GLITCH_ACCEL should be decreased?

Or.. also 10m/s/s seems to be quite high for some copters/users.
Maybe a tuning function could be implementer for adaptation of INAV_GPS_GLITCH_RADIUS and INAV_GPS_GLITCH_ACCEL values or make it avaliable in CLI?

Or maybe it is also a need for the recovery algorithm improvement.

xdigyx, 2016-08-29:
I have just calculated the formula:
INAV_GPS_GLITCH_RADIUS + 0.5f * INAV_GPS_GLITCH_ACCEL * dT * dT) and it seems that error distance is **2.7m** (2.5m by INAV_GPS_GLITCH_RADIUS, and 0.2m by 0.5f * INAV_GPS_GLITCH_ACCEL * dT * dT).

As I remember, logs from my copter flight were showing a glitch of at least 4m...

xdigyx, 2016-08-29:
I am looking forward for implementation of glitch changes  based on GPS speed changes. I am willing to test it.

xdigyx, 2016-09-14:
I have made some work on the topic...
after DE referenced #206 I actually had same feeling about GPS reading and tried make same tests and actually for some extends  I have a different results:
as for Pedestrian mode I had very similar position plot, but for A-1G mode besides less fluctuating plot, there were significant location jumps... such as @theArchLadder posted here:
https://cloud.githubusercontent.com/assets/11679497/15257042/bb2141ba-1944-11e6-9a87-a6dd31ef08e4.png
Which actually seems to be much closer to the truth, as in A-1G mode GPS module is less filtering (so most reads will be closer the actual position, but more jumps) the data than for Pedestrian mode.
Therefore I believe that while iNav have not implemented a good GPS data filtering, it is safer to use the Pedestrian mode (at least no sudden "fly-away"s and smoother nav flights).

So at first, a part of glitch detection probably it would be more efficient to find a best GPS configuration. And regarding to this: I found out that although  SBAS is set to NONE (iNav ver 1.2), after connecting it to u-center we can reed it's actually enabled (is the setting actually stored to GPS memory?).

Another point then once GPS filtering is implemented in my opinion the filtering for Position Hold should be stronger than for other modes (which are more dynamic). Probably this could be better solution than detecting a glitch (which as shown in the pic above is not just a single position read, but also few next readings).

I also need to check A-4G mode (actually no GPS module filtering), maybe there most readings would be very close to real position and significant jumps could be just cut off and the rest filtered.

digitalentity, 2016-09-14:
@xdigyx I'm doing some smarter filtering in #543 now. However the code is completely untested. If you have some spare time - please fly it **without** any altitude hold/position hold/RTH modes and record some data for me to analyze.

xdigyx, 2016-09-14:
ok, just share the hex

digitalentity, 2016-09-16:
@xdigyx hex attached. Please be extremely cautious when enabling AltHold or PosHold mode. Changes to position estimation code are large and might be buggy,
[inav_1.2.1_NAZE.hex.gz](https://github.com/iNavFlight/inav/files/476105/inav_1.2.1_NAZE.hex.gz)

Regarding SBAS - GPS autoconfig code in INAV doesn't save settings to GPS EEPROM or battery-backuped memory. Therefore configured settings are only active until power-cycling the module. Once power-cycled, settings from EEPROM will be in effect until INAV autoconfig re-configures the module.

xdigyx, 2016-09-16:
First data batch
[NewPosEst.TXT](https://github.com/iNavFlight/inav/files/477413/NewPosEst.TXT)

digitalentity, 2016-09-16:
Thanks for testing! Looks like I made something very wrong in new position accuracy calculations. More thinking required.

xdigyx, 2016-09-16:
no problem. just to clarify one point (not sure whether this might affect the results), while flashing "full chip erase" was not performed, so settings from 1.2 RC5 were used.
For a second I used Alt hold at the beginning and the model oscillated up and down +-5meters...

digitalentity, 2016-09-16:
Yes, I obviously messed up something.

digitalentity, 2016-09-16:
I think I know what might have happened, btt It will take time to analyse the data with Octave and confirm.

xdigyx, 2016-09-16:
ok
I tried to follow the code changes but for now it's too complicated for me... 
Once new soft is ready, do not forget to provide .hex for further testing ;)

xdigyx, 2016-09-17:
@digitalentity are you sure the eq:
powf(M_Ef, -sq(x) / (2.0f * sq(curveWidth)));
is right?

I am a bit confused, but sq() => square root?
normal distribution would be rather powf( e,-(x*x)/(2*width*width) )
...if I'm not mistaken

martinbudden, 2016-09-17:
@xdigyx `sq` = square, not square root

```#define sq(x) ((x)*(x))```

digitalentity, 2016-09-24:
@xdigyx hi, sorry for long delay. I've made some fixes to the code, would appreciate if you collect some data for me to analyse. Use same protocol as last time - please don't use GPS/BARO-aware modes.
[inav_1.2.1_NAZE.hex.gz](https://github.com/iNavFlight/inav/files/491072/inav_1.2.1_NAZE.hex.gz)

xdigyx, 2016-09-25:
Hi @digitalentity here is the log.
[1.12.1test.TXT](https://github.com/iNavFlight/inav/files/491795/1.12.1test.TXT)
Bad or not, within 2/3 of the flight baro is on. It took me a while to recognize that I flew with this being ON..

oleost, 2016-11-22:
@digitalentity Can this issue be closed due to #543 and the changes there in regard to the changed done there?

digitalentity, 2016-11-22:
Not really, there is still improvements to be made to GPS glitch **detection**. #543 only added better handling of the GPS outage itself.

oleost, 2016-11-23:
Okey.

I think this is an important step regarding support on small frames.

Small frames and small GPS have issue with GPS reception. Using pedestrian mode on GPS would probably be highly beneficial. But I remember having some scary fly aways with pedestrian mode, probably due this.

digitalentity, 2016-11-23:
#543 may partly solve this already. GPS glitch is a condition where GPS is reporting incorrect or inconsistent data. GPS loss should be handled correctly now.

closed 2018-05-27 by stale[bot]