iNavFlight/inav #516: Problem with Modes NAV POSHOLD / NAV RTH

Timon321, 2016-08-28:
Hello, i am on Version: INAV/SPRACINGF3 1.2.0 Aug 18 2016 / 13:48:20 (5b8b978)

when i add a range to NAV POSHOLD or NAV RTH, then i can´t arm the copter.
I can add ranges to other modes without a Problem.
The Modes are not activated throug the aux value and the warning leds blink so i can´t arm when added the ranges.

Maybe i made a configuration-mistake or there is a bug?

stronnag, 2016-08-28:
Please post a CLI dump, then we may be able to tell.
If you have configured nav modes, you need to have a GPS fix to arm unless you set 

````
set nav_extra_arming_safety = OFF
````

Timon321, 2016-08-28:
Thanks for fast reply.
But somehow it didn´t fix the arming problem?
I havent tested it with gps-fix because i am in the cellar:

Here is the dump:

# dump

# version
# INAV/SPRACINGF3 1.2.0 Aug 18 2016 / 13:48:20 (5b8b978)
# pflags
# Persistent config flags: 0x00000001
# dump master

# mixer
mixer QUADX
mmix reset
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -INFLIGHT_ACC_CAL
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -FAILSAFE
feature -SONAR
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DISPLAY
feature -ONESHOT125
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_NRF24
feature -SOFTSPI
feature VBAT
feature RX_SERIAL
feature GPS
feature FAILSAFE
feature LED_STRIP

# beeper
beeper GYRO_CALIBRATED
beeper RX_LOST
beeper RX_LOST_LANDING
beeper DISARMING
beeper ARMING
beeper ARMING_GPS_FIX
beeper BAT_CRIT_LOW
beeper BAT_LOW
beeper GPS_STATUS
beeper RX_SET
beeper ACC_CALIBRATION
beeper ACC_CALIBRATION_FAIL
beeper READY_BEEP
beeper MULTI_BEEPS
beeper DISARM_REPEAT
beeper ARMED
[... 367 lines cut ...]
set max_angle_inclination_pit = 300
set gyro_soft_lpf_hz = 60
set acc_soft_lpf_hz = 15
set dterm_lpf_hz = 40
set yaw_lpf_hz = 30
set yaw_p_limit = 300
set iterm_ignore_threshold = 200
set yaw_iterm_ignore_threshold = 50
set rate_accel_limit_roll_pitch = 0
set rate_accel_limit_yaw = 10000

# dump rates

# rateprofile
rateprofile 0

set rc_expo = 70
set rc_yaw_expo = 20
set thr_mid = 50
set thr_expo = 0
set roll_rate = 20
set pitch_rate = 20
set yaw_rate = 20
set tpa_rate = 0
set tpa_breakpoint = 1500

oleost, 2016-08-28:
Hi..

For starters your acceleremoter is not calibrated. And there can be also other built in safety measures hindering you from arming.

You need to read the wiki. https://github.com/iNavFlight/inav/wiki

Please close this issue, and after you read trough the wiki if you still have issue you can go to rcgroups thread and ask for help.

closed 2016-08-28 by Timon321