iNavFlight/inav #519: HAL: Unify sensor interface

digitalentity, 2016-08-29:
Currently each sensor uses it's own interface for detection, init, configuring and reading. I think we need to unify this. Possibly each sensor driver should expose a method table allowing the following:

```
bool detect(void);
bool init(void * initParams);  // initParams are sensor-dependent
bool setParam(int paramId, int paramValue);  // sensor dependent data
bool read(void * buf);  // sensor-dependent data
void poll(void); // a method to be called periodically by a task to allow driver to do something in the background
```

All current sensors (ACC, GYRO, MAG, BARO, RANGEFINDER, GPS) should be converted to use this framework.

Thoughts?

Linjieqiang, 2016-08-29:
Great for coding .

martinbudden, 2016-08-29:
Currently in `drivers/sensors.h` we have:
```
struct acc_s;
typedef bool (*sensorInitFuncPtr)(void);                    // sensor init prototype
typedef bool (*sensorReadFuncPtr)(int16_t *data);           // sensor read and align prototype
typedef void (*sensorAccInitFuncPtr)(struct acc_s *acc);    // sensor init prototype
typedef void (*sensorGyroInitFuncPtr)(uint8_t lpf);         // gyro sensor init prototype
typedef bool (*sensorInterruptFuncPtr)(void);               // sensor Interrupt Data Ready
```

We can unify these by using some `void *` parameters, but personally I'd prefer to keep the minor differences and retain the type checking.

digitalentity, 2016-08-29:
I don't see a point in having a separate `sensorInitFuncPtr`, `sensorAccInitFuncPtr` and `sensorGyroInitFuncPtr`. 
Also, baro init it handled within `detect()` method which is not very straight-forward.

My idea might be too general, but what I really want to do it to make all sensors respect the detect -> init -> read chain without hiding parts of it.

martinbudden, 2016-08-29:
Yes, I agree about the detect -> init -> read chain.

martinbudden, 2016-08-29:
And I agree we should work on tidying this up.

jaecktec, 2016-08-29:
How about the [Adafruit unified sensor library](https://learn.adafruit.com/using-the-adafruit-unified-sensor-driver/introduction) kind of style?

martinbudden, 2016-08-30:
I had an email github notification from @jaecktec which seems to have been lost by github.

They suggested using the style of the Adafruit unified sensor library -https://learn.adafruit.com/using-the-adafruit-unified-sensor-driver/introduction

This is certainly worth looking into.

closed 2018-05-27 by stale[bot]