iNavFlight/inav #525: Code cleanup

xdigyx, 2016-08-29:
Hi,

I am going through the code ver 1.1, trying to understand some of it and noticed some lines regarding gpsFakeGPSUpdate function, definded only in CC3D, for testing probably, but should be removed from the official release code.

digitalentity, 2016-08-30:
Why? It's compiled out in release binaries.

xdigyx, 2016-08-30:
In case someone could forget to comment this out or put #endif in the wrong place, etc, etc.

digitalentity, 2016-08-30:
We have loads of `#ifdef`'s in the code, there's no point in removing all of the `#ifdef`'ed code even if it is used for one specific case on one specific target. It's either using conditional blocks of code or maintaining zillion of separate source code branches which IMO is an unreasonble waste of time.

digitalentity, 2016-08-30:
BTW, those `#defines` are commented out. If somebody uncomments them he should be knowing what he's doing anyway.

xdigyx, 2016-08-30:
OK. It's your code your decision.
Personally, I'd remove.
Closed.

closed 2016-08-30 by xdigyx