iNavFlight/inav #530: iNAV 1.2.0 RC3 SPRACINGF3 No Magnetometer?

Timon321, 2016-08-29:
Is there no longer Support for the internal Mag on the SP Racing F3?

[image]

As well i can't calibrate it:

[image]

In the RC2 i had a problem with the Magnetometer, too. It always run slowly back to 314 deg when I turned the copter on his z-axis. In cleanflight there was no problem with this.

xdigyx, 2016-08-29:
hmm.. looking at the code, it has not been removed....
F3 processors have enough memory to keep it all.

stronnag, 2016-08-29:
It's also working on SPRF3 firmware board (dodo).

Timon321, 2016-08-29:
Flashed it new with wipe. Same issue. Maybe my mag has a defect, but works with the newest cleanflight.  In the 3D-View everything seems to be okay when i move the copter, while mag is greyed out and cant calibrate.

urlu75, 2016-08-29:
with RC2, same issue as timon321, run slowly and back to some value whatever is the position.
I flash RC3 on 2 SPRF3 board.
On one the mag is detected and working fine, on the other one, no way even with set mag_hardware.

digitalentity, 2016-08-30:
Testes on DODO and two different SPRF3 boards - all working. 
RC3 includes better hardware error detection - if something wrong happens during detection and init magnetometer will be disabled for flight safety.

digitalentity, 2016-08-30:
For those who have SPRF3 and MAG not being detected - please check if I2C error count in the status line of Configurator is zero. If it's not - you have faulty hardware and that's being detected by firmware. Failing sensor is disabled.

digitalentity, 2016-08-30:
I have a strong feeling that it's not a bug, but a safety feature to prevent using marginal hardware in flight.

digitalentity, 2016-08-30:
Specifically, I think this line is responsible for what we are seing:
https://github.com/iNavFlight/inav/pull/515/files#diff-2911d6bba50f9619cf6c96fab7fed31fR740

I don't think it's bad though. We might be hitting the hardware limitation, not the hardware bug here. I have no boards with undetected mags so unfortunately I can't debug this further.

bk79, 2016-08-30:
> 2911d6bba50f9619cf6c96fab7fed31fR740

Mine works good and detect it as well could be probably useful to increase
detection time?

digitalentity, 2016-08-30:
Can you try this build? It increses HMC5883L calibration delay, might fix the issue.
[inav_1.2.0_SPRACINGF3_218bdf9.hex.gz](https://github.com/iNavFlight/inav/files/444161/inav_1.2.0_SPRACINGF3_218bdf9.hex.gz)

urlu75, 2016-08-30:
It might be a hardware/situation issue.
I removed the board from the quad, reflash it and now the mag is detected. The board is stacked on top of a PDB/BEC. When stacked, mag is not detected (PDB/BEC is not powered). Same result with the 2 boards.
I did some tests off the quad, far from the PDB, calibration seems to be ok on both RC2 and 218bdf9. Can't tell if there is any difference. Will run some other tests later.

digitalentity, 2016-08-30:
Switching BEC usually has a ferrite core inductance. Ferrite is a ferromagnetic material which might cause distorsion of magnetic field near the board. This might be enough for HMC5883 calibration procedure to detect saturation and fail.

bk79, 2016-08-30:
Try to put a sandwitch of aluminium foil and double tape between the bec
and the board bottom it should isolate somehow

digitalentity, 2016-08-30:
It's not possible to isolate **magnetic** interference and not affect compass performance. Even if you somehow manage to isolate magnetic field generated by BEC you'll also isolate magnetic field generated by Earth and render your compass useless :)

urlu75, 2016-08-30:
thanks, good to know to run correct tests. The BEC will go to other place, it's a different issue now :)

martinbudden, 2016-08-30:
This is actually a good result. It shows the new compass initialisation code detected a problem and so disabled the compass. Better to detect this at intialisation rather than when flying.

@digitalentity , it does show, however, a need for error logging, so the user knows **why** the compass was not detected.

Timon321, 2016-08-30:
Thanks for that suggestion. My board is also mounted on top of the pdb with a distance of ~4mm. When I´m back home i´ll try to place it new with more distance to the pdb.

digitalentity, 2016-08-30:
@martinbudden yes, logging is a must have feature now, since we can disable sensors on initialisation even on successful detection. It's so much required that I'm considering implementing and merging it into 1.2...

martinbudden, 2016-08-30:
Yes, logging would certainly make the "why is my compass not working?" type of questions easier to answer.

martinbudden, 2016-08-30:
Closing this as iNav is behaving in the correct manner.

closed 2016-08-30 by martinbudden

bk79, 2016-08-30:
It won't isolate the board from the earth mag field  but it will limit the
effect of the bec on the board just one side..it's like placing a board on
a metal frame

oleost, 2016-09-04:
http://www.rcgroups.com/forums/showpost.php?p=35645654&postcount=7968

Issue reported at forum,  something to look into?

schnupperm, 2016-09-04:
fixed it (not really ;-):

changing line 217 in drivers/compass_hmc5883l.c from
     if (-4096 >= MIN(magADC[X], MIN(magADC[Y], magADC[Z]))) {
to
     if (-4096 >= MIN(magADC[X], MIN(magADC[Y], magADC[Z])) && (validSamples >= 5)) {
solved the problem for my setup.
It seems that the first readings from my mag are saturated.

chuckstarck, 2016-09-06:
I had the same issue about a week ago with RC5. I had calibrated the mag, then was doing roll axis check back and forth watching the graphic in Setup, when Calibrate Mag became grayed out. I unplugged/replugged the FC 4-5 times waiting a few seconds between and still it would not come back. I forgot to look above at the sensors detected. I had to shut down and restart iNav to get it to come back on.
SP Racing EVO board and power distribution board only. Motors not running, no battery plugged in, just USB.
I get randomly correct Mag alignments.
I just flew over the weekend numerous flights, same hardware with CF without the issues.
Now that I'm satisfied with the board working well in CF, I will try iNav again and see if this issue pops up.

chuckstarck, 2016-09-06:
Just loaded RC5 again. 
SPRacing evo, no other hardware except distribution board.
Order of events: Setup, ACC alignment, CLI-board orentation, Setup, Mag alignment,  Graphic showed sluggish heading values after setting board down. Configuration, 8.5 degree offset for mag, back to Setup. Mag alignment was grayed out. Had to close inav, unplug/plug board would not fix it.
MAG came back. I2C errors total count went from 1 to now 3. Now board shows fast heading values.

utsxumiao, 2016-10-03:
@chuckstarck 
I am having SPRacing F3 EVO board plus PDB and had exactly same issues several days ago when I was trying to setup a new build quad, iNav 1.2.1. More specifically two issues -- A) sluggish orientation in the graphic, B) Mag disappear randomly (I was thinking it is randomly).

Wasn't lucky enough to head here but after numerous testing I realized that it was the magnet field from my PDB causing Mag disappear, even if the PDB not being powered up. As soon as moving FC away from PDB, restart FC, mag comes back. moving closer, restart, mag disappear, repeatable. Then I was wondering why this board has a magnet field even when not be connected, suddenly my brain start to work -- IT HAS A BUZZER! My plan was simply stack the FC on top of the PDB, change of plan, still stack it but put a thin metal sheet with double sided foam tape on top of the PDB to shield its magnet field and leave the FC as the plan. Tested all good now, at least the mag stopped disappearing. Make sure the metal sheet does magnet field isolation (test with a piece of magnet), I cut it from a gift box.

The sluggish orientation in graphic. Just disconnect by clicking the top right button and connect it again it will be working fine. I guess it is something to do with the configurator or compatibility to the FC not the FC itself as it fixes by just reconnecting it rather than restarting it.

First time posting here, I really wanna give a big thank you to the master and contributors for the awesome iNav project, I cannot express how much I love it.

utsxumiao, 2016-10-03:
By the way, I flipped the PDB as its bottom is flat so the foam does not cover any component. Will try to take some images after work.

bilalasd, 2017-01-14:
@utsxumiao you are a freaking life saver. I was having this problem after I changed my setup and couldn't figure out what the problem was. Turns out to be the buzzer.

Here is what I found in my setup

**Initial Setup:** Buzzer connected to FC with ~4 inch wire.

**Case 1: FC on top of the pdb** 
Magnetometer working fine

**Case 2: FC away from pdb** 
Magnetometer working fine

**Updated Setup:** Buzzer soldered directly to FC.

**Case 1: FC on top of the pdb** 
Magnetometer doesn't show up in the configurator.

**Case 2: FC away from pdb** 
Magnetometer working fine

Its interesting that the buzzer ONLY affects the magnetometer when the FC is on top of the pdb.

digitalentity, 2017-01-14:
@bilalasd it's possible that buzzer plus magnetic fields from PDB mess-up with run-time self-calibration of HMC5883 magnetometer. The logic is - if magnetometer saturates during self-calibration - we can't trust it and thus it's reported as not detected.

Another case of how important is to get magnetometer far far away from **any** magnetic interference.

utsxumiao, 2017-01-14:
@bilalasd the master -- @digitalentity is completely correct, magnet field can be really complex and it make life even harder by invisibility to bare eyes. One more point to you, it might not worth it to try the onboard magnetometer at all, at least I never got any good result from it. I've tried different layouts on my setup, unless you can move FC away from the frame. I ended up cutting a trace to disable the onboard mag and use the one come with GPS.

Anfre, 2017-04-23:
try CLI command "set mag_hardware = AUTO" ?