iNavFlight/inav #531: Implement boot logging facility

digitalentity, 2016-08-30:
I think we need a way to troubleshoot boot process. How about dedicate some RAM to record an event log from powering up to entering main loop?
Log should be plain binary and contain the following:

- Timestamp (ms since boot) [uint32_t]
- Event code [uint16_t]
- Event parameters (dependent on event code) [uint16_t]x4 ?

Each record will be about 14 bytes, I think we can easily dedicate 1KB to store boot events.
The log itself will be displayed in human readable form by a `bootlog` CLI command.

Log should include successful sensor detections, successful/failed sensor initialisation, PWM channel allocation etc.

Having such a log will help us troubleshoot and profile boot process.

Thoughts?

martinbudden, 2016-08-30:
Yes, this is a good idea. Might as well make the record 16 bytes: the extra memory is insignificant and it will make hex dumps much easier to read.

digitalentity, 2016-08-30:
Hm... Actually we might want to make a boot log being dumped into blackbox for easier analysis. 
However, let's implement it first.

digitalentity, 2016-08-30:
This is how bootlog looks. PR to follow
[image]

xdigyx, 2016-08-31:
NAZE32 rev5 does not recognize the command

stronnag, 2016-08-31:
Nor does any other flight controller not build from the correct branch.
````
# bootlog
Time Evt Parameters
   0:  0 
   0:  1 
 500: 18  (0, 0, 0, 0)
 500: 18  (1, 1, 0, 2)
 500: 18  (2, 2, 0, 2)
 500: 18  (3, 3, 0, 2)
 500: 18  (4, 4, 0, 2)
 500: 18  (5, 5, 0, 2)
 500: 18  (6, 6, 0, 2)
 500: 17  (7, 6, 0, 1)
 500: 17  (8, 6, 0, 1)
 500: 17  (9, 6, 0, 1)
 500: 17  (10, 6, 0, 1)
 500:  2 
 636:  9  (2, 0, 0, 0)
 752: 10  (3, 0, 0, 0)
 767: 11  (3, 0, 0, 0)
 767: 12  (2, 0, 0, 0)
2025: 13  (0, 0, 0, 0)
2025:  4 
2526:  8 

# version
# INAV/NAZE 1.2.0 Aug 31 2016 / 20:14:44 (d963186)
````
Are you Timecop in disguise?

xdigyx, 2016-08-31:
lol, no!

I used FW1.2 RC4 for Naze. If this one is not correct then...

stronnag, 2016-08-31:
you need the boot-logging branch

xdigyx, 2016-08-31:
oh.. ok, I thought it's already in 1.2 RC4 which is available on Chrome -> inav cfg tool

closed 2016-09-01 by digitalentity