iNavFlight/inav #537: Motor Mix stopped working

DaleKramer, 2016-08-30:
I have been testing with a hex file I built on July 28 (sorry, don't know commit #)

Today, I built a hex file to 11c42fd. 

For some reason it appears that my motor mix is no longer functioning.

Here is the mix:
Code:
mixer CUSTOM
mmix reset
mmix 0 1.000 -0.334 0.407 1.000
mmix 1 1.000 -0.278 -0.400 1.000
mmix 2 1.000 0.334 0.407 -1.000
mmix 3 1.000 0.278 -0.400 -1.000
mmix 4 1.000 -1.000 -0.007 -1.000
mmix 5 1.000 1.000 -0.007 1.000
smix reset
Now, when I power things up, my motors 4 and 5 just sit there and wiggle.

The motors tab shows 6 motors but motors 4 and 5 (5 and 6 sliders) do not respond to slider movement, the motors just sit there and wiggle at any slider position.

Motors 0-3 work fine with the sliders.

If I put the 7-28 hex file back onto FC, all motors works fine.

Can anyone help with how to debug this?

Thanks,
Dale

DzikuVx, 2016-08-30:
Which target are you using?

DaleKramer, 2016-08-30:
SPracingF3EVO

digitalentity, 2016-08-30:
Worth checking out, thanks for reporting.

digitalentity, 2016-08-30:
@DaleKramer tested the 11c42fd
`INAV/SPRACINGF3EVO 1.2.0 Aug 30 2016 / 15:22:38 (11c42fd)`
Default settings + your custom mixer.

No issue with all 6 motors. Can you provide the full dump?

digitalentity, 2016-08-30:
Issue confirmed and it's OneShot related.

digitalentity, 2016-08-31:
No, it's not OneShot, it's SERVO_TILT that is for some reason mapped to M5 and M6 instead of M7 and M8

digitalentity, 2016-08-31:
@DaleKramer I've figured it out.

BTW, this is exactly the case where bootlog (https://github.com/iNavFlight/inav/pull/536) comes in very handy:
[image]

SERVO_TILT is correctly mapped to M7 and M8.
Actually, your configuration is never supposed to work correctly. It's a hardware limitation - M5-M8 share the same timer so they can only be of same update frequency, also the same timer can't be shared between OneShot and Servo.

The reason it worked previously is that SERVOS were NOT supported on EVO board. Disable SERVO_TILT and you'll have everything functioning properly.

However we need to detect and prevent usage of SERVO_TILT when mixer has more than 4 motors on EVO board.

digitalentity, 2016-08-31:
We can also break compatibility and remap M1-M2 as servos (like on SPARKY board) and keep SERVO_TILT usable up to 6-motor config without any interference with OneShot.

DaleKramer, 2016-08-31:
Thanks for the solution! 

I would not have figured that out.  

At this point I will disable SERVO_TILT as I will currently be testing the vLazair without the automatic pilot seat servo (which was the idea of using Servo_tilt) but I may later ask for help in remapping M1 and M2 so that I can have servo tilt with 6 motors.

I will check out #536, it looks handy.

closed 2016-08-31 by digitalentity