iNavFlight/inav #538: Sparky 2 target?

bazyl78, 2016-08-31:
Any plans for Sparky 2 support in the future? Sparky 1.0 already done.

martinbudden, 2016-08-31:
@bazyl78 , it would certainly be possible to make a Sparky 2 target.

I have no Sparky 2 hardware myself, so would you be able to spend some time testing it?

bazyl78, 2016-09-01:
I'm not very expierienced but can try.

martinbudden, 2016-09-01:
You don't need to be a great pilot to test!

All we need you to do is build an aircraft with the Sparky2 FC and report back any problems so we can fix them.

You just need to do a bench test (no props), and if that works fine a flight test. And report back with any problems you have.

I'll start looking into making a Sparky 2 target.

anbello, 2016-09-01:
@martinbudden 
I have a quad X 250 with sparky 2 and I can do some testing if you want. The problem is that I have spare time only in the week end.

bazyl78, 2016-09-02:
That is great news cause I stoped using Sparky 2 because problems with navigation in dRonin. With inav it works for me out of the box on Naze32. If it starts working with Sparky 2 then perfect :). I will test when the new target is available.

digitalentity, 2016-09-02:
@bazyl78, @anbello 
Please check out the RC5 - it has a very experimental SPARKY2 support (thanks to @martinbudden)

anbello, 2016-09-02:
Thanks I will try as soon as possible, I hope tomorrow.

martinbudden, 2016-09-02:
Let us know if you have any problems.

Note that the motor order is different between iNav and dRonin/LibrePilot. So you will have to either rewire your motors or create a custom mixer.

anbello, 2016-09-02:
A little bit OT question:
is there any way to translate the PID values from dRonin to iNav?

digitalentity, 2016-09-02:
I'm afraid not, the core logic is very different, nobody ever did a mapping between OP/LP/dRonin PID controller and INAV

anbello, 2016-09-03:
I flashed the RC5 on my Sparky 2 based quad, follow what I see:
- in the configurator top bar I have gyro and accel grayed and mag an baro blue
- in the sensor tab the only graph that react in the right way to moving the quad is gyro, accel and baro are fixed to 0 and mag is fixed to 0 with nearly periodic spike (see attached image)
- the receiver tab is OK (I use PPM and I see all bars move in the right way)
- the motors tab is OK (I can spin the motors with the sliders)

Let me know which kind of information I have to post, a dump from the cli or what else?

[image]

oleost, 2016-09-03:
Status command from cli would probably be helpful.

anbello, 2016-09-03:
# status
System Uptime: 108 seconds, Voltage: 0 * 0.1V (3S battery - OK), System load: 0.00
CPU Clock=168MHz, GYRO=MPU9250, BARO=MS5611, MAG=MAG_AK8963
Cycle Time: 2000, I2C Errors: 0, config size: 1996

digitalentity, 2016-09-03:
Looks like accelerometer is not detected for some reason. I wonder if there's a bug in MPU9250 init code that prevents the chip from initializing correctly. What we probably see on compass graph is a result of `mag.read()` returning false.
Baro is probably fine, it's the absence of accelerometer that causes the graph to stay at zero. In Configurator it's not really raw baro altitude, but a result of INAV position estimation filtering which will not work w/o accelerometer.

@anbello can you also post output of "bootlog" command?

anbello, 2016-09-03:
# bootlog
Time Evt            Description  Parameters
   0:  0          CONFIG_LOADED
   0:  1       SYSTEM_INIT_DONE
 500: 18   TIMER_CHANNEL_MAPPED  (0, 0, 0, 0)
 500: 18   TIMER_CHANNEL_MAPPED  (1, 1, 0, 2)
 500: 18   TIMER_CHANNEL_MAPPED  (2, 2, 0, 2)
 500: 18   TIMER_CHANNEL_MAPPED  (3, 3, 0, 2)
 500: 18   TIMER_CHANNEL_MAPPED  (4, 4, 0, 2)
 500: 18   TIMER_CHANNEL_MAPPED  (5, 5, 0, 2)
 500: 18   TIMER_CHANNEL_MAPPED  (6, 6, 0, 2)
 500: 18   TIMER_CHANNEL_MAPPED  (7, 7, 0, 2)
 500: 18   TIMER_CHANNEL_MAPPED  (8, 8, 0, 2)
 500: 18   TIMER_CHANNEL_MAPPED  (9, 9, 0, 2)
 500: 18   TIMER_CHANNEL_MAPPED  (10, 10, 0, 2)
 500:  2          PWM_INIT_DONE
 500:  3       EXTRA_BOOT_DELAY
1185:  9         GYRO_DETECTION  (8, 0, 0, 0)
1237: 10          ACC_DETECTION  (1, 0, 0, 0)
1250: 11         BARO_DETECTION  (3, 0, 0, 0)
1291: 12          MAG_DETECTION  (6, 0, 0, 0)
1374:  4       SENSOR_INIT_DONE
1924:  8           SYSTEM_READY

martinbudden, 2016-09-03:
Yes, looks like the accelerometer is not detected. A look at the code seems to indicate that the detection function is not being called. PR on the way.

martinbudden, 2016-09-03:
@digitalentity PR https://github.com/iNavFlight/inav/pull/554 should fix this.

martinbudden, 2016-09-03:
@anbello , I've attached a build with the fix, can you try it out please?

[SPARKY2.zip](https://github.com/iNavFlight/inav/files/453329/SPARKY2.zip)

anbello, 2016-09-03:
This new build seems to solve all problems in my list but the mag plot is still weird

# status
System Uptime: 349 seconds, Voltage: 0 * 0.1V (3S battery - OK), System load: 0.00
CPU Clock=168MHz, GYRO=MPU9250, ACC=MPU9250, BARO=MS5611, MAG=MAG_AK8963
Cycle Time: 2001, I2C Errors: 0, config size: 1996

# bootlog
Time Evt            Description  Parameters
   0:  0          CONFIG_LOADED 
   0:  1       SYSTEM_INIT_DONE 
 500: 18   TIMER_CHANNEL_MAPPED  (0, 0, 0, 0)
 500: 18   TIMER_CHANNEL_MAPPED  (1, 1, 0, 2)
 500: 18   TIMER_CHANNEL_MAPPED  (2, 2, 0, 2)
 500: 18   TIMER_CHANNEL_MAPPED  (3, 3, 0, 2)
 500: 18   TIMER_CHANNEL_MAPPED  (4, 4, 0, 2)
 500: 18   TIMER_CHANNEL_MAPPED  (5, 5, 0, 2)
 500: 18   TIMER_CHANNEL_MAPPED  (6, 6, 0, 2)
 500: 18   TIMER_CHANNEL_MAPPED  (7, 7, 0, 2)
 500: 18   TIMER_CHANNEL_MAPPED  (8, 8, 0, 2)
 500: 18   TIMER_CHANNEL_MAPPED  (9, 9, 0, 2)
 500: 18   TIMER_CHANNEL_MAPPED  (10, 10, 0, 2)
 500:  2          PWM_INIT_DONE 
 500:  3       EXTRA_BOOT_DELAY 
1185:  9         GYRO_DETECTION  (8, 0, 0, 0)
1237: 10          ACC_DETECTION  (9, 0, 0, 0)
1250: 11         BARO_DETECTION  (3, 0, 0, 0)
1291: 12          MAG_DETECTION  (6, 0, 0, 0)
1374:  4       SENSOR_INIT_DONE 
1924:  8           SYSTEM_READY 

[image]

martinbudden, 2016-09-03:
@anbello , I think the SPARKY2 is the first flight controller that is using the compass in the MPU9250 chip, so this might take a little while to track down.

Can you use an external compass for the moment?

Also if you could test flight modes that don't need a compass, that would be useful as well.

@digitalentity , any thoughts about the compass?

anbello, 2016-09-03:
@martinbudden , I have to disable the Compass in some way (I don't know how) or Can I fly in angle with and without althold and don't care about the malfunction of the Compass?

martinbudden, 2016-09-03:
You can disable the compass by going into the CLI (bottom tab in the configurator) and typing:
```
set mag_hardware = 1
save
```

You can fly in angle mode without the compass.

Remember you are testing a new build, so things could go wrong. Make sure you bench test without props before you fly. Double check all your settings, failsafe etc. And then start off with some very short flights.

anbello, 2016-09-03:
I have not done a real test flight. I only armed the quad on the floor and with throttle a little bit up I moved the roll / pitch stick to see the quad go forward - backward - left - right. At first it was good then I saw strange movements so I disarmed and I connected to the configurator and I saw a big roll pitch drift on the attutude (the setup tab).

https://www.youtube.com/watch?v=2wSw0ahLhfo&feature=youtu.be

digitalentity, 2016-09-04:
@anbello you probably didn't do a 6-point accelerometer calibration.

digitalentity, 2016-09-04:
Hm... PixRacer uses the same MPU9250 gyro/acc/mag combo, I can use it to debug.

anbello, 2016-09-04:
@digitalentity I did the 6-point accelerometer calibration before the test in my precedent post, this are the resulting values for accgain:

# get accgain
accgain_x = 4035
Allowed range: 1 - 8192

accgain_y = 4033
Allowed range: 1 - 8192

accgain_z = 4046
Allowed range: 1 - 8192

martinbudden, 2016-09-04:
Yes, those look like and acc driver/calibration problem. WIthout hardware I can't investigate this further, so I'll leave the reset to @digitalentity .

digitalentity, 2016-09-04:
@anbello what does accelerometer graph read when you enable it on "Sensors" tab? Does Z axis read something near 1.00 ?

[image]

anbello, 2016-09-04:
@digitalentity sometime is near 1.0, sometime is less (0.8 - 0.9)  and sometime is more (1.1 - 1.2), you can also see on the attached screenshot and the linked video on precedent post.

It's strange ... I have the same behavior also with raceflight but if I re-flash dronin the attitude seem stable even if I observe a long period (about few minutes) and small entity (about 1 degree) drift.

digitalentity, 2016-09-04:
@anbello I can't see your video.

On the ground and in stable hover your acceleration should read 1,0 or very close to it. Can you record a blackbox log of your hover test?

jstremmler, 2016-09-04:
Own a Sparky2 as well and I am willing to test as well once I have fixed the problem with the bootloader and have flashed the 1.2.0 RC6 Sparky2 hex file. I have an external Compass which plugs in the master connector with an additional external magnetometer which is connected to the i2C connector of the Sparky2. So I would have to switch the internal magnetometer off and the external on for testing. 

Under TauLabs GCS/firmware I have a working OPLink connection via 433MHz as well. 

Will that work with iNav?

anbello, 2016-09-04:
@digitalentity  now you should see it, i forgot to make it public
https://youtu.be/2wSw0ahLhfo

In this condition I can not hover however I tried to do and this are the log
[blackbox_log.zip](https://github.com/iNavFlight/inav/files/453848/blackbox_log.zip)

digitalentity, 2016-09-04:
@anbello I'd say it's poorly done accelerometer calibration. When sitting still and flat the accelerometer should read 1.00 (0.95 is probably ok)

Can you please re-do the accelerometer calibration?

anbello, 2016-09-04:
I will do it as soon as possibile, now I'am not at home

Anyway I'am pretty sure I did it the right way because I already did it
several time on a naze32 and on a flip32 and all went right

Furthermore with the other flight controller also without calibration I
never saw shuch a big and random drift on the attitude

Maybe my sparky 2 has some problems

digitalentity, 2016-09-04:
@anbello I've seen accelerometers reading 0.8G when uncalibrated. Even slightly bent PCB causes stress that heavily affects accelerometer. Maybe your FC was sligntly bent when attached to the quad. 
Try loosening the mounting screws so FC is slightly loose and re-calibrate.

anbello, 2016-09-04:
@digitalentity Just to clarify: the accelerometer reading not only is
different from 1 but it also change over time from 0.8 to 1 then 1.2 and so
on slowly and in a random way

digitalentity, 2016-09-04:
Yes, I understand that. PixRacer has the same accelerometer chip. I haven't verified yet if it's a software problem. So far it looks like either a faulty accelerometer or excess strain on the board.

digitalentity, 2016-09-04:
#560 fixes compass issue (at least on PixRacer)

anbello, 2016-09-05:
I re-did the accelerometer calibration carefully and I obtained the same results: just after the calibration the attitude and sensors are OK, then I try to take off and I see really strange behaviour, when I connect to the configurator with quad steady on the floor I see what you can see on attached image (slowly changing in time but never good).

[image]

digitalentity, 2016-09-06:
Hm... intereting. @bazyl78 you have a Sparky2, can you confirm the same behavior?

martinbudden, 2016-09-06:
Status: Sparky2 works, but some accelerometer problems. Marking this for 1.3

anbello, 2016-09-06:
@martinbudden @digitalentity is it possible to have a build with also the #560 fix for the compass to try it?

gottmoz, 2016-09-08:
I tested inav on sparky 2, but frsky sbus wont work! I can not change to sbus option, it wont stick after reboot...

digitalentity, 2016-09-10:
@gottmoz let's check the obvious things:

1. Do you use INAV Configurator?
1. Did you select Serial RX function in Ports tab?
1. Did you select Serial RX feature in Configurator tab?

gottmoz, 2016-09-10:
> @gottmoz let's check the obvious things:
>
> Do you use INAV Configurator?
> Did you select Serial RX function in Ports tab?
> Did you select Serial RX feature in Configurator tab?

Yes, all things good! If i flash regular cleanflight with same setup it works!

Skickat från min iPhone

oleost, 2016-09-10:
@gottmoz Unsure if it gotten fixed,  but I heard there are a bug that forces you to use cli and not the GUI  to enable sbus.

oleost, 2016-09-10:
If that's the case we should open a issue as I cannot find any open issues regarding it.

digitalentity, 2016-09-10:
> Yes, all things good! If i flash regular cleanflight with same setup it works!

What board are you testing? Cleanflight **does not** support Sparky 2.

digitalentity, 2016-09-10:
> @gottmoz Unsure if it gotten fixed, but I heard there are a bug that forces you to use cli and not the GUI to enable sbus.

@oleost it has been fied in latest versions of Configurator.

gottmoz, 2016-09-10:
If i start cleanflight configurator with inav flashed on sparky 2 , the reciewer will work! 

Skickat från min iPhone

digitalentity, 2016-09-10:
What version of INAV Configurator are you using? Can you try the latest
from Chrome store?

gottmoz, 2016-09-10:
I use the latest version

Skickat från min iPhone

anbello, 2016-09-12:
@gottmoz apart from the problems related to receiver have you any problems with sparky2 related to accelerometer and attitude like the one described in my other posts?

gottmoz, 2016-09-12:
No, i did not test anymore, because i could not get the reciever working...

Skickat från min iPhone

anbello, 2016-10-04:
The SPRACINGF3EVO has an MPU9250 as gyro, acc, mag sensor, I would like to know if there is someone with this board that report problems with acc and attitude like the ones reported by me in this issue.

I also would like to know why for the SPRACINGF3EVO target in the config.h I see GYRO and ACC defined for MPU6500:

`
#define GYRO
#define USE_GYRO_SPI_MPU6500

#define ACC
#define USE_ACC_SPI_MPU6500

#define ACC_MPU6500_ALIGN       CW180_DEG
#define GYRO_MPU6500_ALIGN      CW180_DEG

#define BARO
#define USE_BARO_BMP280
#define USE_BARO_MS5611

#define MAG
#define USE_MPU9250_MAG // Enables bypass configuration
#define USE_MAG_AK8963
//#define USE_MAG_MAG3110 // External
#define USE_MAG_HMC5883 // External
`

and for the SPARKY2 target I see GYRO and ACC defined for MPU9250:

`
#define GYRO
#define USE_GYRO_SPI_MPU9250
#define GYRO_MPU9250_ALIGN      CW270_DEG

#define ACC
#define USE_ACC_SPI_MPU9250
#define ACC_MPU9250_ALIGN       CW270_DEG

#define MAG
#define USE_MAG_AK8963
#define USE_MAG_AK8975
#define USE_MAG_HMC5883
#define USE_MAG_MAG3110
#define MAG_AK8963_ALIGN CW270_DEG
`

anbello, 2016-11-07:
I have to report the the problems with sparky 2 posted in this issue by me are to be considered related only to my (defective) sparky 2. I reflashed it with dronin and I obtained the same random drift in the attitude and in the accelerometer value with the FC steady on desktop. In precedent test with dronin I did not see the problem only because the values change slowly and after a certain time form the power on.

digitalentity, 2016-11-08:
@anbello so it's a hardware problem after all. good that you were able to figure it out!

anbello, 2016-11-08:
Maybe this is not the right place but I would like to ask which F4 board could be a good (and affordable) solution for iNav?
OpenPilot Revolution or Flip32 F4 with a MS5611 on I2C and (for both) a GPS (serial) with HMC5883l (I2C)?

DzikuVx, 2016-11-08:
@anbello  please join https://gitter.im/iNavFlight/inav . I'm closing thi one since it is hardware issue

closed 2016-11-08 by DzikuVx

anbello, 2016-11-08:
@DzikuVx thanks

[5 more replies]