iNavFlight/inav #539: FW1.2 R4 Channel Forwarding disables some motor outputs

xdigyx, 2016-08-31:
As in the topic:
while Chan FWD is enabled, motors 1 & 2 do not spin for Hexa config.
Board Naze32 rev6.

DzikuVx, 2016-08-31:
I'm affraid this is a problem all MultiWii/Basefligh/Cleanflight derivatives are facing. There are not enough timers for all possible motor and servo configurations. This issue was rised at least few times, there is no solution. 
When Multirotor mixer is used, only 2 outputs can drive servos. Those are output 1 and 2. Then, output 3 acts as output 1 and so no. Motor 5 should be on INPUT 5 and Motor 6 on INPUT 6. But you would have to test it. And it works only with PPM input or SerialRX.

You might want to read #174 #141 cleanflight/cleanflight#1954

DzikuVx, 2016-08-31:
OK, I reopen it. Looks like we've broken something in 1.2 since it was really working like described in 1.1.

DzikuVx, 2016-08-31:
It is also possible that it was working in 1.1 as an actual bug since FEATURE_CHANNEL_FORWARDING was not considered as "servo" output and probably was driven at 490Hz

DzikuVx, 2016-08-31:
@digitalentity we might have a problem here. This stoped working because 2 things happened.

1. in 1.2 FEATURE_CHANNEL_FORWARDING is considered as "servo" output and MOTOR1 and MOTOR2 are remapped to servos
1. 1.2 also introduced TARGET_MOTOR_COUNT and it is set 6 on NAZE target. So, INPUT5 and INPUT6 will not act as outputs even with refresh rate of 490Hz.

So, we either test it with TARGET_MOTOR_COUNT  = 8 and allow 2 first motors to act as servos, or do not remap them to 50Hz and allow INPUT5 and INPUT6 to output AUX channels at 490Hz

Ideas?

digitalentity, 2016-08-31:
@DzikuVx 

1. I wonder if FEATURE_CHANNEL_FORWARDING should remap few channels as servos. I suspect it's good for compatibility reasons and we really had a bug in 1.1
1. It should work anyway since TARGET_MOTOR_COUNT limits number of **motors**, not the total number of outputs - we should still be able to have 2 servos + 6 motors on NAZE.

digitalentity, 2016-09-01:
Ok, looks like on NAZE target `FEATURE_CHANNEL_FORWARDING` only allows 4 motors. INPUT5-8 are mapped to servos as well. This is the way outputs are mapped in Cleanflight as well. 
We might want to fix that.

xdigyx, 2016-09-01:
Do you mean only 4 motors when CH FWD is enabled?
If yes, then I can tell you it's not true. Either CF & BaseF allow 6 motor outputs with this feature.
Is it possible to make 6 motor on Output pins 1-6, and at least 1-2 servos on Inputs 5-8 at 50Hz?

digitalentity, 2016-09-01:
Ok figured it out, Channel forwarding is NOT considered `useServos` in mixer code. We do consider it as true `useServos` flag this causes us to map first two outputs to servos with only `FEATURE_CHANNEL_FORWARDING ` enabled.

digitalentity, 2016-09-01:
@xdigyx can you test #540 and see if it fixes the issue?

xdigyx, 2016-09-01:
OK. Please share .hex

DzikuVx, 2016-09-01:
https://drive.google.com/open?id=0Bx6X245zlrsneWtmTEx3d2Q3Q0U

digitalentity, 2016-09-01:
Check this:
[inav_1.2.0_NAZE.hex.gz](https://github.com/iNavFlight/inav/files/449009/inav_1.2.0_NAZE.hex.gz)

DzikuVx, 2016-09-01:
heh, both within minute :)

digitalentity, 2016-09-01:
:smile:

DzikuVx, 2016-09-01:
@digitalentity I wonder how this really works. If 2 outputs are not remapped as servos, but still used as outputs (INPUT 5 and 6) what is really refresh frequency on them. And if it's really 490 (400?)Hz then why servo he had connected there did not fried after all?

digitalentity, 2016-09-01:
The problem is not with servos, it's with motors. As far as I know @xdigyx is flying a hexacopter - he needs 6 motors while CHANNEL_FORWARDING leaves us with only 4 in current implementation

DzikuVx, 2016-09-01:
@digitalentity I mean in 1.1.

DzikuVx, 2016-09-01:
OK, never mind, I see it now. There is a special routine in `pwm_mapping.c` that allows this to work.
Man... every time I see `pwm_mapping.c` the more and more I do not like that code...

digitalentity, 2016-09-01:
Me to. It's a constant source of bugs

DzikuVx, 2016-09-01:
Plus, looks like, channel forwarding in multirotor is working only with NAZE and SPRACINGF3. Other targets are not configured

digitalentity, 2016-09-01:
the conditional code in pwm_mapping should be made part of target-specific code - no more crazy loads of ifdefs

martinbudden, 2016-09-01:
I believe @blckmn is planning tor rewrite `pwm_mapping` for version 3.1 of betaflight. We can then move it over to iNav.

It truly is the most horrible bit of code!

DzikuVx, 2016-09-01:
That would be really really something!

digitalentity, 2016-09-01:
Several from latest serious bugs originated there :smile: 
Good luck to @blckmn!

blckmn, 2016-09-01:
I'm hopeful I'll have code for at least one target done this weekend, it can then be reviewed before rolling out to all targets. I'll post here once the PR is ready.

digitalentity, 2016-09-01:
Cool, please keep us in the loop! We really need that mess cleaned up!

xdigyx, 2016-09-01:
OK, tested and the results:
1. Enabling CH_FWD does not disables any of 6 motors, which is good!
2. 50Hz PWM is available on selected Input pins (NAZEs pins5-8), which is also good :)
3. in order to enable these channel forwarding, one also MUST enable Servo_tilt which is not as it meant. is it? in CF just CH_FWD was enough. I believe servo tilt should be when mixed levelling with rc input...

digitalentity, 2016-09-02:
@xdigyx 
Can you try this one? CH_FWD feature is an exception from common rules - it's treated by mixer as a servo mixer, but pwm_mapping treats it in another way...

[inav_1.2.0_NAZE_76003f9.hex.gz](https://github.com/iNavFlight/inav/files/451438/inav_1.2.0_NAZE_76003f9.hex.gz)

xdigyx, 2016-09-02:
Now PWMs are enabled just with CH_FWD, but again the same motors are being disabled...
Also when Srovo_tilt enabled there is actually no change to PWM reflecting the copter level changes

digitalentity, 2016-09-02:
Verified on my NAZE, default settings, Hex X frame.

1. `FEATURE_CHANNEL_FORWARDING` - all motors are available, AUX are forwarded to RC5-8.
2. `FEATURE_SERVO_TILT` - servos are enabled on M1-2, motors 5-6 are disabled (this is expected behavior), NAZE target doesn't support SERVO_TILT + 6-motor configuration
3. `FEATURE_CHANNEL_FORWARDING` + `FEATURE_SERVO_TILT` - servos are enabled on M1-2, motors 5-6 are disabled, AUX are forwarded to RC5-8

So far everything works as expected except `SERVO_TILT` not responding to board angle changes.

digitalentity, 2016-09-02:
Ok, forgot `FEATURE_SERVO_TILT` only is driving servos if `CAMSTAB` mode is enabled. 

Everything works as expected on my NAZE. @xdigyx  can you please carefully re-test with attached hex?

[inav_1.2.0_NAZE_1c2a139a.hex.gz](https://github.com/iNavFlight/inav/files/451628/inav_1.2.0_NAZE_1c2a139a.hex.gz)

xdigyx, 2016-09-02:
Yes, with the latest .hex I can confirm the same results. I can also confirm SERVO_TILT+CAMSTAB working on M1-2 for QUAD mixer.
Thanks for sorting this out.

closed 2016-09-02 by digitalentity