iNavFlight/inav #541: Ibus receiver with GPS on Flip32, question

raoul596, 2016-09-01:
Is it possible to connect a receiver via Ibus and a GPS at the same time to a Flip32?

digitalentity, 2016-09-01:
Yes, it's possible. GPS to SoftSerial and IBUS to UART2.
Please also consider using RCG thread or Gitter chat for support.

raoul596, 2016-09-01:
I have found a problem by connecting Ibus to my flip32. All the channels are "glitching" I also tried it with Cleanflight, but there it works perfectly fine. What am I doing wrong or is it a software problem?

digitalentity, 2016-09-01:
I don't have a FlySky TX anymore. 
Can somebody confirm the bug with iBus?

bk79, 2016-09-01:
Serialrx works good on spf3vdeluxe

raoul596, 2016-09-01:
Here a screenshot. My sticks are in the center position.

[image]

digitalentity, 2016-09-01:
Not SerialRX in general. I can confirm that S.Bus is working good as well. It's the IBUS receiver I'm concerned about.

raoul596, 2016-09-01:
Its a FS-IA10b with the flysky FS-I6S transmitter

nibborg, 2016-09-05:
At this moment the  FS-IA10b on my 250 quad with SPRACINGF3 + 1.2-RC5 its working ok. No issues with the Rx.

raoul596, 2016-09-08:
The problem is solved in Inav V1.2, thanks!

DzikuVx, 2016-09-08:
Great to hear it's working. Closing

closed 2016-09-08 by DzikuVx