iNavFlight/inav #546: Reset estimated altitude at the moment of arming

digitalentity, 2016-09-02:
This allows for more precise WP navigation and better OSD display.

martinbudden, 2016-10-12:
@digitalentity , this issue does not seem to have a PR associated with it. Should it not be deferred to 1.4?

digitalentity, 2016-10-13:
Agreed. It's a minor change, but it's even not implemented.

digitalentity, 2016-11-13:
#543 will fix this.

digitalentity, 2016-11-20:
This should be fixed now as #543 is merged

closed 2016-11-22 by digitalentity

carlonb, 2016-12-11:
Hi all,
Just tested the version 1.4 with SP racing F3 and I saw that after arming the altitude (shown on OSD) is not reset.
After a short flight time the baro have a small drift (due to temperature i bet) of about 1 or 2 mt, but after landing, disarm and re-arm, the altitude still +2mt. I'm expect the altitude will reset in order to take off with 0 mt.
Bye
Thanks

digitalentity, 2016-12-11:
Altitude is kept at zero (reset) at first arming. Disarming/re-arming doesn't reset the altitude again.

carlonb, 2016-12-11:
Yes I saw it, but will this feature implemented in future ?
Thanks
Carlo

carlonb, 2016-12-26:
Just tested vers. 1.5, but still no altitude reset after every re-arm. The altitude deviation is caused by a baro drift and may be from 1 to 3 mts drifting during a flight. If I fly at very low altitude (let say 5mt) this is a big erratic indication read on OSD.
Sorry if I'm boring you, but this was a nice and simple improvement in MWii and for me very usefull.
Bye
Carlo

oleost, 2016-12-26:
Can verify this also on Frsky and / or smartport telemetry ( cannot remember which aircraft I saw this on )

digitalentity, 2016-12-26:
Current behavior only ensures that altitude is zero on first ARM. Resetting
altitude on each arming may cause dangerous effects for WP missions.

Maybe we should add an option to reset altitude to zero on each arming?

oleost, 2016-12-26:
I know, I'm pretty sure it was on first arm. But I don't remember 100%. Will recheck after nightshift and some sleep. ( 13 hours )

carlonb, 2016-12-27:
Yes, I can confirm the altitude is reset on very first Arm.
So, if we will  have an altitude drift after first few fly minutes we must switch the copter off and then switch it on again to reset it.
With this future "option" will be enough disarm and re-arm.
Yes. the option may be selectable by CLI.
Bye