iNavFlight/inav #547: Use GCC-5.4 for release builds

digitalentity, 2016-09-02:
GCC-5.4 provide much smaller binaries, we should consider using it for 1.2 release. Thoughs?

martinbudden, 2016-09-02:
How much smaller are the binaries?

digitalentity, 2016-09-02:
1K smaller for current CC3D build. Actually allows to put FrSky telem back in.

martinbudden, 2016-09-02:
Well, I'm normally quite conservative about changing the compiler, but 1K is a lot. And of all the sources of bugs in the next release I think the compiler is the least likely source.

So I think we should use the new compiler.

stronnag, 2016-09-02:
I'd be slightly concerned with the lack of testing on 5.4. I had a torrid time early this year with 5.3 (e.g. http://www.rcgroups.com/forums/showpost.php?p=34761612&postcount=5850), problems which magically disappeared when compiled 4.9 or 6.*. 

As long as you're satisfied that 5.4 is safe (and I think βF is using it), then take the space saving.

digitalentity, 2016-09-02:
I'd really like to use gcc-6.1 for the builds, but I'm not on Linux yet and
debugging inside a virtual machine sounds like a really painful process...

stronnag, 2016-09-02:
I would put out an RC on 5.4 and ask for feedback.

digitalentity, 2016-09-02:
I like the idea. Will do.

martinbudden, 2016-09-02:
You could build everything except CC3D (and possibly NAZE) with 4.9 and just use 5.4 where it makes a beneficial difference.

digitalentity, 2016-10-04:
5.4 provides buggy binaries. Confirmed.

closed 2016-10-04 by digitalentity

bk79, 2016-10-04:
I'm still using 4.9.3 and happy to use it, I think it's better to keep
using older version to stable version unless really sure that it works
without issues