iNavFlight/inav #550: SPRACINGF3OSD target support?

piposa, 2016-09-02:
Is there any plan to support OSD targets like SPRACINGF3OSD in iNAV?
Cleanflight has added support for that in June.

stronnag, 2016-09-02:
Is there any plan for someone to donate some hardware to @digitalentity ??

krzysztofmatula, 2016-09-05:
I wonder what would be the benefits of such thing?
The OSD built out of the CF would just work with iNav controller, wouldn't it?

piposa, 2016-09-05:
This board has a smaller form factor and less wiring compared to external OSD. It works great with Cleanflight for me, so extension with iNav would be great.
http://www.rcgroups.com/forums/showthread.php?t=2633525

krzysztofmatula, 2016-09-05:
Ah, OK... I was thinking the SPRACINGF3OSD is for building OSD firmware for the stackable PDB/OSD board for EVO that Mr.Cleanflight is going to start selling this week.

jflyper, 2016-09-14:
@piposa SPRACINGF3OSD is an OSD, using CF code base to drive OSD chip (thus no separate M328 chip often found in OSDs).

SPRACING branded FC Integrated OSD (without M328) is also planned AFAIK.

There are a couples of this type of FC already in the market; e.g. FuriousFPV, OMNIBUSF3.
These already have support requests in iNav.

If you can't wait, there are bunch of F1/F3 + M328 based OSD FCs around, in addition to the one you linked. These all accepts NAZE or SPRACINGF3 target binaries. Look for AIO keyword in RCG.

digitalentity, 2016-11-21:
Probably not going to happen as nobody seem to be using the board

closed 2016-11-21 by digitalentity