iNavFlight/inav #552: MAG detection on REVO board

urlu75, 2016-09-02:
can't get the mag enable on REVO board with RC5.
with RC2 the mag is detected but i2c error count is running. Calibration KO.
with RC3+ MAG is no more detected even with CLI mag_hardware.

1500: 12 MAG_DETECTION (2, 0, 0, 0)
2755: 15 HMC5883L_READ_FAILED 
2755: 14 MAG_INIT_FAILED 
2755: 4 SENSOR_INIT_DONE 
3305: 8 SYSTEM_READY

MAG detection (add mag_hardware=0 from CLI) & calibration succeed with betaflight 3RC12, no i2c error,manually add mag_hardware=0 with CLI.

digitalentity, 2016-09-03:
HMC5883L init failed for some reason. Unfortunately I don't have a REVO to investigate.

digitalentity, 2016-09-03:
Can you check if https://github.com/iNavFlight/inav/pull/532 fixes the issue for you?

urlu75, 2016-09-03:
#532 is not fixing mag detection

digitalentity, 2016-09-03:
Ok, I definitely need a REVO to investigate.

martinbudden, 2016-09-05:
@urlu75 , can you try PR https://github.com/iNavFlight/inav/pull/563

I don't think it will fix the problem, but it will provide more information in the logs.

urlu75, 2016-09-05:
here is the PR #563 result :
500:  2          PWM_INIT_DONE 
 500:  3       EXTRA_BOOT_DELAY 
1185:  9         GYRO_DETECTION  (6, 0, 0, 0)
1486: 10          ACC_DETECTION  (7, 0, 0, 0)
1500: 11         BARO_DETECTION  (3, 0, 0, 0)
1500: 12          MAG_DETECTION  (2, 0, 0, 0)
2355: 15 HMC5883L_READ_OK_COUNT  (0, 0)
2355: 16   HMC5883L_READ_FAILED  (5, 5)
2455: 14        MAG_INIT_FAILED 
2455:  4       SENSOR_INIT_DONE 
3005:  8           SYSTEM_READY 

and still some I2C error (11 or 12).

martinbudden, 2016-09-06:
So, all the reads to the HMC5883L fail. @digitalentity , any idea what might cause that?

digitalentity, 2016-09-06:
My guess would be the i2c code, not the compass driver. Hard to debug without hardware. Just some $40 more and I can get myself a REVO :wink:

martinbudden, 2016-09-06:
We need some real hardware to make any further progress on this. Marking it for release 1.3.

@urlu75 , can you try using an external compass, to see if that works?

urlu75, 2016-09-07:
did not succeed to detect an external mag. The only external mag I have is the same chip as the one on the board... I understood that can be an issue.

digitalentity, 2016-10-18:
Is this issue still valid in 1.3-RC1?

urlu75, 2016-10-18:
quick feedback with 1.3-RC1 :
-with i2c_overclock=off then i2c error counter is running fast, no mag detected
-with i2c_overclock=on then i2c error stopped at 3 and mag is detected
Mag calibration seems to be ok.
I will try to go through the all sensor's calibration process later.

digitalentity, 2016-11-15:
Can somebody re-test this with latest `development`?

DzikuVx, 2016-11-15:
I can test HMC5883 on Airbot F4 target today.

DzikuVx, 2016-11-15:
@digitalentity I've tested external HMC5883 and AirbotF4. Works fine, mag detected in all tries.
I'd say we close it.

digitalentity, 2016-11-16:
Also tested on REVO (thanks @sambas). No issues.

closed 2016-11-16 by digitalentity