iNavFlight/inav #555: Feature request: Fixed-wing servo trim function.

oleost, 2016-09-03:
New flight mode which is called "Airplane trim toogle".

* Requirres passtrough mode enabled.
* Requirres throttle to be below min_throttle

When it toggles on then off:

* Set current servo output as new servo center.

Procedure to use this:

1. Trim the airplane in air when flying in passtrough mode using your TX.
2. Land the airplane.
3. Toggle the "Airplane trim toogle" while still "passtrough" is active and sticks are centered.
4. Reset your trim to zero on your TX

oleost, 2016-09-03:
Todays solution is either to trim in passtrough mode, and then land and try to replicate current servo position when hooking up to and computer. Or using the method explained here http://www.rcgroups.com/forums/showpost.php?p=35059861&postcount=6741

oleost, 2016-09-03:
Original request http://www.rcgroups.com/forums/showpost.php?p=35636569&postcount=7921

robert-b, 2016-09-03:
the procedure sucks.
level the plane: https://www.eflight.ch/pi/Schwerpunktwaage-MPX693054.html
get the offsets - save them - done.

implementation tooks 20 minutes ...

digitalentity, 2016-09-04:
@oleost sounds like a good idea to me.
@robert-b level plane doesn't mean level flight.

robert-b, 2016-09-04:
yes, i start to understand your view.
it's a two step 'calibration' schema.
and i already have it running.

oleost, 2016-09-04:
As an addition if possible, make use of LED / BEEPER or / and twitching of ailerons as an confimation of trim succesfully applied.

oleost, 2016-09-30:
@digitalentity Postpone to 2.0?

closed 2016-12-12 by digitalentity