iNavFlight/inav #557: Sparkyflight 1.1 not working anymore with new IO

bk79, 2016-09-04:
since 0489eb8b0840719277450a8d15bb6f1d5e6f62be sparky refuse to boot, previous to this commit it works.
I've tested several times flashing the board with other commits and it works, from this commit on it doesn't work not even the latest commit. Probably something is wrong with new IO for sparky

digitalentity, 2016-09-04:
@martinbudden can you have a look?

bk79, 2016-09-04:
probably I've found the issue 

 +#define USABLE_TIMER_CHANNEL_COUNT **11**
 +#define USED_TIMERS             (TIM_N(1) | TIM_N(2) | TIM_N(3) | TIM_N(15) | TIM_N(17))

I'll make a test

bk79, 2016-09-04:
nopo setting it to 17 doesn't work anyway

bk79, 2016-09-04:
@digitalentity @martinbudden  I don't know if it makes sense but also USB connection is not recognized when flashed with that release

martinbudden, 2016-09-04:
@bk79 , yes USB connection issue makes sense. Now I remember they had this problem with betaflight.

@digitalentity here's a possible fix: https://github.com/iNavFlight/inav/pull/558

bk79, 2016-09-04:
In the fix you didn't change the max used timers is it ok to have it to 11
when we use also 17?

martinbudden, 2016-09-04:
Yes, that's OK. `USABLE_TIMER_CHANNEL_COUNT` is for the `timerHardware` array in `target.c`.

bk79, 2016-09-04:
fix works

bk79, 2016-09-04:
@DzikuVx  the fix make the board boot but uart2 is not recognized setting rx-serial from config tab without choosing anything on port config makes it appear, if i set it back to rx-ppm diasppear again

bk79, 2016-09-04:
@martinbudden  sorry the message was for you

martinbudden, 2016-09-04:
@bk79 , yes PPM and UART2 use the same timer, so it is not possible to have both enabled.

bk79, 2016-09-04:
Ok is there anything regarding tx on pwm6 i wasn't able to enable it on
ibus telemetry applying ibus telemetry patch

martinbudden, 2016-09-06:
Fixed by PR https://github.com/iNavFlight/inav/pull/558

closed 2016-09-06 by martinbudden