iNavFlight/inav #559: Unable to arm

Noa16h, 2016-09-04:
Hello, I am on Version: INAV/NAZE 1.1
I changed the Software from Triflight to INAV on my Trikopter with an Naze32 and a neo 6m gps. The triflight software worked but with really bad nav characteristics.
After calibrating the Acc and the Magnetometer as described in the wiki I also used the PID Settings from "the-tricopter-v2-5 on 3S" out of the wiki. Apart of this I only changed the rc/yaw/pitch... rates in the cli as prefered.  Now the problem was that when I tried to arm the copter the arming light was blinking. So I double checked my stick end points and the max_check settings in the cli, I also tried arming via aux. So this can't be the problem and I figured out that the  3 long beeps at powering up mean an "FAILURE_ACC_INIT". So I thought it might be an accelerometer problem and I recalibrated it. After that I was able to fly one time after connecting and disconnecting the battery several times. But this worked only once. So I thought it might be a problem with the gyro and the gyro_sync settings. I also noticed that the CPU was at 300% ?!?
So I set the gyro_lpf to 42HZ and restartet the board. The looptime was stil the same but when I looked at the raw sensor values I noticed that the gyro is not working. This problem happens at every frequence exept the 256HZ. But the interesting thing was that I was able to arm the board with lower frequence than 256HZ. That's why I diabled the gyro_sync and set the gyro_lpf to 42HZ. Now I was able to arm the board after about 5 min after powering up the board. 
When I power the board up it still beeps 5 short times and then 3 times long. Then for about 4 min the arming led is just blinking while I already have a gps 3D fix (I know because of the green blinking led on the gps board) then the led 0 starts to blink faster than the arming led with short on duration. About 20 sec. later I hear a beeping and the arming led stops blinking. After that I can arm the board and fly; also POSHOLD, ALTHOLD, RTH... work. After landing and disarming I am able to rearm. Also after changing the battery the arming led is still not blinking. Only after a longer duration of unpowered board the same problem occurs again.
This seems to me as an GPS problem but does not explain the three long beeps at power up and the problems with the gyro settings. 
After two whole days of testing I absolutely don't know what to do, maybe I have more than one problem. 
Thanks for helping.

DzikuVx, 2016-09-04:
Please share CLI dump

Noa16h, 2016-09-04:
[4.9CLI-DUMP_INAV.txt](https://github.com/iNavFlight/inav/files/453941/4.9CLI-DUMP_INAV.txt)
This is the dump.

digitalentity, 2016-09-04:
@Noa16h three long beeps a while after power-up means "successful gyro autocalibration".
```
set nav_extra_arming_safety = ON
```
is what prevents you from arming (this is precisely why you have to get a GPS fix before INAV will allow you to arm and fly)

Noa16h, 2016-09-04:
oh i missunderstood this from the blinkenlights wiki page
but when the gyro is succesfull autocalibrated, then why do i have the problem that i can't see any gyro output when gyro_sync is enabled gyro_sync_denom is set to 12 as in wiki and Gyro_lpf is set to for example 42Hz; its although not flighing then

DzikuVx, 2016-09-04:
gyro_lpf = 42Hz and gyro_sync_denom = 12 gives in total about 12000 looptime. Not 1200 but 12,000 us. 

What you need is `gyro_lpf=42` and `gyro_sync_denom=2`

Noa16h, 2016-09-04:
Thanks for the fast help. Now I understand the problem, so normaly when i set the gyro_lpf to 42 and sync_denom to 2 it should be able to arm it.
I will try this.

oleost, 2016-09-04:
Also,  even if your gps is blinking green it does not mean that it has 3d lock with 6 sats or more which is required

DzikuVx, 2016-09-04:
@Noa16h remember that with `nav_extra_arming_safety = ON` you really have to have 3D GPS lock

Noa16h, 2016-09-04:
@oleost thank you for this tip I haven't found this information anywhere on the wiki so i thought that it just needs an GPS 3D fix, which I always have. The problem is that i normally just have 5 sats in view so this is why it sometimes works and sometimes not. 
So know i set the inav_gps_min_sats = 5 and yeah it works.

Thanks to everyone helping me with this!

closed 2016-09-04 by Noa16h