iNavFlight/inav #561: Sensor detection on boot to sensitive?

oleost, 2016-09-05:
From user on forum. 

Board:  SPRACINGF3
iNav firmware: 1.2.rc4/5
Configuration: Ublox M8N GPS on UART2, and compas connected to SDA/SLC., MWOSD on UART1, Spektrum Sat. on UART3
Working on earlier iNav:  1.2.rc2/1.1

Issue explained: Since rc4 my Mag is only detected after a soft reset. 

See post and the following replies for more info http://www.rcgroups.com/forums/showthread.php?t=2495732&page=532#post35645654

.. 

Looks like the user have an functional mag but with some interference on cold boot but not warm restart. Something that can be improved/delayed ?

oleost, 2016-09-05:
This is @schnupperm issue. 

Made a new issue instead of using an old closed one #530

lkaino, 2016-09-05:
I'm having similar issues with the HMC5883L compass. I ended up fixing it in my own build so that the compass init is called twice if the first init fails. For some reason after a cold boot the compass gives out "dirty" samples which makes the saturation check fail.

It also depends on where the compass is located, if I move it away from other electronics it sometimes passes on first init call.

bk79, 2016-09-05:
@digitalentity probably is better to have mag init at last with some
delays..

digitalentity, 2016-09-05:
We already have a 500ms cold boot delay. @lkaino, can you increase delay at [this line](https://github.com/iNavFlight/inav/blob/master/src/main/main.c#L410) until the problem goes away?

lkaino, 2016-09-05:
@digitalenty Yes I can test having different delays there.

martinbudden, 2016-09-05:
PR https://github.com/iNavFlight/inav/pull/563 is an attempt to address this problem.

@lkaino can you test this please?

lkaino, 2016-09-05:
@martinbudden Ah I missed that one. Will test it as well.

martinbudden, 2016-09-06:
FIxed by PR https://github.com/iNavFlight/inav/pull/563

oleost, 2016-09-06:
@schnupperm Got a chance to test this fix is working for you aswell?

schnupperm, 2016-09-06:
yes, I hope I have time tonight

schnupperm, 2016-09-06:
switched to master:
mag is detected.

bootlog:
# bootlog
Time Evt            Description  Parameters
   0:  0          CONFIG_LOADED 
   0:  1       SYSTEM_INIT_DONE 
 500: 19   TIMER_CHANNEL_MAPPED  (1, 1, 0, 2)
 500: 19   TIMER_CHANNEL_MAPPED  (2, 2, 0, 2)
 500: 19   TIMER_CHANNEL_MAPPED  (3, 3, 0, 2)
 500: 19   TIMER_CHANNEL_MAPPED  (4, 4, 0, 2)
 501: 19   TIMER_CHANNEL_MAPPED  (5, 5, 0, 2)
 501: 19   TIMER_CHANNEL_MAPPED  (6, 6, 0, 2)
 501: 19   TIMER_CHANNEL_MAPPED  (7, 7, 0, 2)
 501: 19   TIMER_CHANNEL_MAPPED  (8, 8, 0, 2)
 501: 18  TIMER_CHANNEL_SKIPPED  (9, 8, 0, 3)
 501: 18  TIMER_CHANNEL_SKIPPED  (10, 8, 0, 3)
 501: 18  TIMER_CHANNEL_SKIPPED  (11, 8, 0, 3)
 501: 18  TIMER_CHANNEL_SKIPPED  (12, 8, 0, 3)
 501:  2          PWM_INIT_DONE 
 511:  3       EXTRA_BOOT_DELAY 
1047:  9         GYRO_DETECTION  (2, 0, 0, 0)
1163: 10          ACC_DETECTION  (3, 0, 0, 0)
1184: 11         BARO_DETECTION  (2, 0, 0, 0)
1184: 12          MAG_DETECTION  (2, 0, 0, 0)
2812: 17    HMC5883L_SATURATION  (1, 0)
2913: 13  RANGEFINDER_DETECTION  (0, 0, 0, 0)
2913:  4       SENSOR_INIT_DONE 
3464:  5          GPS_INIT_DONE 
3464:  7    TELEMETRY_INIT_DONE 
3514:  8           SYSTEM_READY 

Thank you

oleost, 2016-09-06:
Great, so @digitalentity should it still report saturation even if only the first fails (which seems common?)

martinbudden, 2016-09-06:
Well, we should report what it does. We allow 5 read or saturation failures before failing the initialisation.

The saturation count may vary according to the quad setup or the flight controller, so it is interesting information to have.

oleost, 2016-09-06:
Okey,  guess closing this then.

closed 2016-09-06 by oleost