iNavFlight/inav #564: Fixed wing PID controller improvement

digitalentity, 2016-09-05:
I think we have a significant issue with airplane code that affects takeoff behavior.
At the moment we have full effects of PID integrator right after arming, but this is not desired at takeoff:

At low speeds (initial acceleration) control surfaces don't have full effect on UAV stabilisation, PID control attempts to compensate using I-term. By the moment airplane have gained significant velocity for control surfaces to work reasonably good, I-term has seriously winded up and PID controller will overshoot leading to severe oscillation which might lead to a crash because altitude during takeoff might be too low to allow successful recovery. This is precisely what happened to my Ranger airplane during maiden with stabilisation.

I suggest creating a TAKEOFF mode that will detect a launch (a jerk in acceleration) and disable I-term in PID control until airplane gains some velocity (using GPS, airspeed sensor or a simple timer). During takeoff only P-term will be active - aircraft might not be able to maintain attitude precisely but in general it should be more stable during takeoff.

FW flyers, thougts?

martinbudden, 2016-09-05:
Shouldn't I-term be zero when on the ground for all aircraft, not just fixed wing ones?

So this becomes a general problem: how do you detect takeoff, and at what point after takeoff does the I-term become active.

I seem to remember @borisbstyle, when he added airmode to betaflight, added some code to disable I-term until the throttle had exceeded a certain value (indicating takeoff). I haven't followed this bit of the code, so I don't know what the code does now.

It's very easy for the pilot to tell if the aircraft has taken off, but quite difficult for the flight controller to know. So here's a bit of a mad idea - allow the pilot to turn I-term on and off on a switch?

csurf, 2016-09-05:
sounds cool... how about making it a user-configurable option/feature that auto-enables on ARM and auto-disables after certain height/distance/speed has been reached (based on GPS readings)?

stronnag, 2016-09-05:
A throttle approach as betaflight is OK for MR, let's not over engineer it. For FW, it probably does need to be more complex. I would not favour a switch solution for MR, too easy to forget, and I run out of switches.

digitalentity, 2016-09-05:
@martinbudden airplanes are tricky. We can take launch detection code from APM and make a simple timer that will only allow I-term after certain timer.

@csurf GPS is good, but not always available...

@stronnag the change won't affect MR code at all. I'm happy with it as it is.

oleost, 2016-09-05:
Sad to hear about your crash,  I have not have any issue with mine. But haven't got to fly lately. 

But I term should simple not be allowed to wind up more than than enough to certain percentage up to maximum output or near maximum output.

digitalentity, 2016-09-05:
At extremely low airspeeds I-term is not helping but harming the performance, that's why I'm thinking about cripling it for takeoff.

oleost, 2016-09-05:
Maybe look at project like Open flight stabilizer and APM http://www.rcgroups.com/forums/showthread.php?t=1794672

DzikuVx, 2016-09-05:
I agree that I-term on fixed wing should be "improved". Probably timer would be enough, let's not overcomplicate it. I propopose following algorithm:

1. this applies only if fixed-wing is detected
1. Activated by flight mode or feature. I do not have strong opinion here
1. After ARM I-term is disabled
1. When Throttle crosses predefined threshold (50% let's say) timer start to count down. Maybe something around 10 seconds. That should be enough to throw airplane into air without need to rush it
1. When timer gets to 0, Iterm is enabled and stays this way until disarm

Timer and throttle threshold can be configured  via CLI

xdigyx, 2016-09-06:
Timer could be also activated when an acceleration threshold is reached, which means a model has been thrown for takeoff.

Schalonsus, 2016-09-06:
That would be much better than the throttle threshold. Fox example for bungee start or gliders or pusher planes.

DzikuVx, 2016-09-06:
@Schalonsus idea of Iterm limiting sounds interesting... hmmm.... This is something that is really worth considering.

What I do not like about using accelerator is that tuning it might be quite difficult. How much acceleration is an indication of take-off? What about planes with landing gear? There is no "throwing" so acceleration might be much lower than in case of throwing. For me, throttle idea is simpler and probably safer

xdigyx, 2016-09-06:
Could  be both (detected one OR the other). Also some users first throw, then push THR up...
Tuning can be a setting to be set via CLI, etc

digitalentity, 2016-09-06:
> I think I gain should have always a increasing limit for planes. Max 1/3 of full aileron throw.
I agree, on airplanes I-term is basically automatic trimming, we don't want it to "trim" using full throw. I'll impose additional limit on I-term for airplanes.

oleost, 2016-09-06:
Agreed

oleost, 2016-09-08:
If an "Launched/in air" state is ever implemented it would be handy to use that state for other things like preventing GPS RTH while on ground.

That way we can example arm, enable RTH and just throw the airplane and it would engage and fly up and loiter.

sppnk, 2016-09-28:
How will this detection affect to bungee-cord launching (no throttle)? I amd flying lately a Mini Talon, wich I launch by bungee cord.

digitalentity, 2016-10-18:
I'm closing this as it's no longer relevant

closed 2016-10-18 by digitalentity