iNavFlight/inav #568: Add support for OMNIBUS FC

digitalentity, 2016-09-06:
Support for http://ledge-team.com/Omnibus-Flight-Control

Blocked by #381 (OSD support)

digitalentity, 2016-09-10:
FC on the way, thanks to http://ledge-team.com
It's good to see FC manufacturers starting to support INAV projest :+1:

jflyper, 2016-09-14:
I can work on this one; already have a target files for basic FC functionality; aka OSD-less-BUS :)

There are some configuration decisions we have to make and extensions to support the OMNIBUS F3 per iNav mission.

To save some investigational work for others, these are what I found out so far:

1. I2C and PWM7&8 are exclusive. Must decide which to take; I2C of course.
2. RSSI input is not assigned to the same ADC as VBAT and CURRENT inputs; need a multi-ADC code or PWM based RSSI module in non-PWM configuration, or just stash RSSI input and use the pin for other purposes.
3. PPM and Serial RX (UART3_RX) share input pins. There are 0-ohm jumper resister for switching PPM and Serial RX. They are both installed as they are shipped, and `AVOID_UART3_FOR_PWM_PPM` is used to kill UART3 when PPM is in use. PPM is left untouched when UART3 is in use. We must solder-desolder these jumpers to maximize the resource utilization.
4. UART3 and PWM5&6 are exclusive.
5. An SPI bus is shared among MPU, BARO and AB7456(MAX7456 compat), each initializing the SPI with own preferred speed, but BARO wins.
6. Because the SPI bus is shared, DMA can not be just simply turned on.

3&4 brings us with several configuration options:
(a) PPM and UART3 (install only PPM jumper)
(b) PPM and PWM5&6 (install only PPM jumper)
(c) S.BUS
(d) S.BUS and half-duplex TX application (I thought there were a PR for doing this somewhere) running at S.BUS speed.

digitalentity, 2016-09-14:
@jflyper you are more then welcome to post a PR for initial support (OSD-less).

My thougts on issues you raised:

1. I2C and PWM7&8 are mutually exclusive. **No PWM7 & 8, just I2C. No HEX+SERVO, no OCTA support**. We need navigation features here. This may change later if required.
1. RSSI input is not assigned to the same ADC as VBAT and CURRENT inputs. **Drop RSSI support**.
1. PPM and Serial RX (UART3_RX) share input pins. **Don't see a big problem with this. Just need to add this to docs.**
1. UART3 and PWM5&6 are exclusive. **How this relates to (3) ? Is it different UART used for SerialRX?**
1. An SPI bus is shared among MPU, BARO and AB7456(MAX7456 compat), each initializing the SPI with own preferred speed, but BARO wins. **Don't see a big problem here.**
1. Because the SPI bus is shared, DMA can not be just simply turned on. **This might be a problem if we decide to do fast gyro rate and OSD. Shouldn't matter for initial support though.**

Let's do initial support first and improve things after we get it flyable.

jflyper, 2016-09-19:
@digitalentity 
It's flyable:)

jflyper, 2016-10-28:
@digitalentity 
You can close this.

closed 2016-10-28 by digitalentity