iNavFlight/inav #569: afroflight32 rev.6 flashing / blheli 1wire passthroug issues

pegaz666, 2016-09-07:
Hello all, that's my firstnpost here hopefully i'm writting in the right place. Thank you all the input in that great project.

I'm having few issues: 
1. I wanted to try 1.2 rc5 and after flashing my afroflight32 rev.6 is not connecting to the configurator anymore. Tried all release candidates and they are having the same issue. I've done chip erease etc. after flashing with those i need to do the bootloader trick everytime to bring the board back to live. No issues with clean flight or iNav 1.1. Works like a charm. No matter what i do any of those 1.2 is not booting up properly and i'm unable to connect to the configurator.

2nd issuse i have is blheli suite- with cleanflight it works like a charm, but it does not work while I'm on iNav. I can't connect with the escs. Everytime i need to change escs settings (they all have blheli bootloaders) i need to flash back to cleanflight in order to get blheli suit working. 
Are these issues known? Am i doing something wrong? Any change to get them fixed? 
Thank you

pegaz666, 2016-09-07:
Just found out that the 1.2 (stabe) is out -this time it flashed without any issues but my esc are loosing sync and i'm unable to take off . Only one prop is spinning, calibrating them via configurator/motor tab is not helping either- just one prop is working only. When i play with the sliders in the configurator they all work fine...master slider and they all are working fine...

DzikuVx, 2016-09-07:
Can you provide us a CLI dump ?

pegaz666, 2016-09-07:
Already went back to iNav 1.1 all good now but i have saved cli dump before so here it is: 

# version
# INAV/NAZE 1.2.0 Sep  7 2016 / 11:26:35 (6cc415a)
# pflags
# Persistent config flags: 0x00000001
# dump master

# mixer
mixer QUADX
mmix reset
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -INFLIGHT_ACC_CAL
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -FAILSAFE
feature -SONAR
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DISPLAY
feature -ONESHOT125
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_NRF24
feature -SOFTSPI
feature RX_PPM
feature VBAT
feature INFLIGHT_ACC_CAL
feature MOTOR_STOP
feature SOFTSERIAL
feature GPS
feature FAILSAFE
feature TELEMETRY
feature ONESHOT125

# beeper
beeper GYRO_CALIBRATED
beeper RX_LOST
beeper RX_LOST_LANDING
beeper DISARMING
beeper ARMING
beeper ARMING_GPS_FIX
beeper BAT_CRIT_LOW
beeper BAT_LOW
beeper GPS_STATUS
beeper RX_SET
beeper ACC_CALIBRATION
beeper ACC_CALIBRATION_FAIL
beeper READY_BEEP
beeper MULTI_BEEPS
beeper DISARM_REPEAT
beeper ARMED
beeper SYSTEM_INIT
beeper ON_USB
[... 265 lines cut ...]
set acc_soft_lpf_hz = 15
set dterm_lpf_hz = 40
set yaw_lpf_hz = 30
set yaw_p_limit = 300
set iterm_ignore_threshold = 200
set yaw_iterm_ignore_threshold = 50
set rate_accel_limit_roll_pitch = 0
set rate_accel_limit_yaw = 10000

# dump rates

# rateprofile
rateprofile 0

set rc_expo = 70
set rc_yaw_expo = 20
set thr_mid = 29
set thr_expo = 0
set roll_rate = 6
set pitch_rate = 6
set yaw_rate = 2
set tpa_rate = 0
set tpa_breakpoint = 1500

#

DzikuVx, 2016-09-07:
I think I know what the problem is: PIDs. INAV 1.2 changed a lot in terms of PID scaling and default values. 

Did you just copied PIDs from 1.1 to 1.2? 

Also you rates look rather odd... only 60dps ? 

Please read https://github.com/iNavFlight/inav/wiki/PID-conversion-from-pre-1.2-to-1.2 and https://github.com/iNavFlight/inav/wiki/Rate-conversion-for-INAV-1.2

pegaz666, 2016-09-07:
I haven't touched any PIDs and did not copy them from previous versions . I keept PID's and Rates stock as they came with the firmware. I don't know how to tune them that's why i did not touch them. From rates section i only changed thr_mid from 50 to 29 as my quad hoovers at 29% of throttle. Restoring configurator from restore file does not restore PIDs so that's not the case ;(

DzikuVx, 2016-09-07:
I do not know what you did, but those are not default PID values of inav 1.2. 

> Restoring configurator from restore file does not restore PIDs so that's not the case ;( 

Are you sure of that? Those Pids did not appeared from nowhere

xdigyx, 2016-09-08:
BLHeli interface has been removed for Naze32 target due to insufficient memory of F1. So you need to configure and calibrate ESCs through BLHeli suite.

oleost, 2016-09-08:
@pegaz666 Please do a firmware flash with clean erase and then post cli dump.

pegaz666, 2016-09-08:
Hi - DzikuVx was right those PID values got restored from previous version so I took another chance and re-flashed it now again making sure that PID's are not gonna be changed BUT the problem is still there. Just one prop is spinning... here is a clean CLI dump ( i always got full chip erased marked during every flashing)

# dump

# version
# INAV/NAZE 1.2.0 Sep  7 2016 / 11:26:35 (6cc415a)
# pflags
# Persistent config flags: 0x00000001
# dump master

# mixer
mixer QUADX
mmix reset
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -INFLIGHT_ACC_CAL
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -FAILSAFE
feature -SONAR
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DISPLAY
feature -ONESHOT125
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_NRF24
feature -SOFTSPI
feature RX_PPM
feature VBAT
feature INFLIGHT_ACC_CAL
feature MOTOR_STOP
feature SOFTSERIAL
feature GPS
feature FAILSAFE
feature TELEMETRY
feature ONESHOT125

# beeper
beeper GYRO_CALIBRATED
beeper RX_LOST
beeper RX_LOST_LANDING
beeper DISARMING
beeper ARMING
beeper ARMING_GPS_FIX
beeper BAT_CRIT_LOW
beeper BAT_LOW
beeper GPS_STATUS
beeper RX_SET
beeper ACC_CALIBRATION
beeper ACC_CALIBRATION_FAIL
beeper READY_BEEP
beeper MULTI_BEEPS
beeper DISARM_REPEAT
beeper ARMED
[... 267 lines cut ...]
set acc_soft_lpf_hz = 15
set dterm_lpf_hz = 40
set yaw_lpf_hz = 30
set yaw_p_limit = 300
set iterm_ignore_threshold = 200
set yaw_iterm_ignore_threshold = 50
set rate_accel_limit_roll_pitch = 0
set rate_accel_limit_yaw = 10000

# dump rates

# rateprofile
rateprofile 0

set rc_expo = 70
set rc_yaw_expo = 20
set thr_mid = 29
set thr_expo = 0
set roll_rate = 20
set pitch_rate = 20
set yaw_rate = 20
set tpa_rate = 0
set tpa_breakpoint = 1500

#

pegaz666, 2016-09-08:
Just checked one thing: after DISABLING "one shot esc" all props started spinning (quad was a bit wobbly) , re-enabling tha feature again brought the problem back again. So there is something wrong with one shot escs in 1.2

xdigyx, 2016-09-08:
Regarding ESC setup/calibration: as I already wrote from 1.2+ BLHeli interface will not be supported  (as far as I know). So you either should flash old FW to setup the ESCs or you would need to use external tools (I see it's not an easy task in your case but it's not impossible). You should not re-calibrate ESCs via configurator/motor tab once you setup PPM min/max from BLHeli suite.

Regarding OneShot I can not help you much.

martinbudden, 2016-09-08:
@pegaz666 , one thing you can do is reflash your FC to betaflight, calibrate the ESCs and then reflash back to iNav.

pegaz666, 2016-09-08:
@martinbudden - this is what I'm doing, taking it back to cleanflight, use Blheli suite to change the settings and then back to iNav but this is not my major problem now. The major problem is that my thing is not flying on any 1.2 ( not all the props a spinning) when on 1.1 everything works as it should

xdigyx, 2016-09-08:
if you got an oscilloscope you could compare outputs. If not then maybe an arduino with a simple program to compare times of outputs HIGH/LOW.

pegaz666, 2016-09-08:
That's the thing - i tried 1.2-RC3, 1.2-RC4, 1.2-RC4 before and non of them were booting my FC properly (they were kind of bricking it. To bring it back to live I had to be shortening the bootloader pads . Only official 1.2 fixed the booting problem but now I've got this

digitalentity, 2016-10-18:
@pegaz666 can you please verify that the issue is still present with 1.3-RC1?

closed 2016-12-14 by digitalentity