iNavFlight/inav #573: Extend LTM protocol to include PID, rates, etc

martinbudden, 2016-09-08:
Perhaps by adding a configuration frame which includes the PIDs, rates etc.

This would allow users to see the PIDs on their transmitter while tuning.

The idea has been raised in the deviationTx forum, see http://www.deviationtx.com/forum/protocol-development/6349-new-protocol-for-inav-flight-control-software?start=20#53622

@stronnag , @digitalentity - any thoughts on this?

DzikuVx, 2016-09-08:
Hmmm... Even when X frame was added with HDOP there were some concerns about bandwidth requirements. After all, LTM is designed for low bps, 1200/2400. The fact that we are sending it with 9600 min ATM does not changes that

On the other hand, we still have free 32 bits inside X frame

stronnag, 2016-09-08:
I would prefer not to burden LTM with this as a permanent function. By all means have a configurable option to sent LTM tuning / debug frames, but please do not permanently burden a low data rate position and attitude protocol with tuning data.

martinbudden, 2016-09-08:
@stronnag , @DzikuVx , sorry, I wasn't very clear in my post.

My proposal is just to add the new frame type and its definition to LTM, not for LTM to be changed to transmit those frames.

The reason I ask for this is that I use the LTM frames for the iNav NRF24L01 telemetry (rather than invent my own telemetry frames).

DzikuVx, 2016-09-08:
Ah.... In that case I'm OK with that.
It would be nice tohave it documented somewhere

martinbudden, 2016-09-09:
My proposal is to add a new 'T' (tuning) frame, iNav extension:

```
Tuning Frame (T)

Payload: 12 bytes

P-roll uint8
I-roll uint8
D-roll uint8
P-pitch uint8
P-pitch uint8
I-pitch uint8
D-yaw uint8
I-yaw uint8
rates-roll uint8
rates-pitch uint8
rates-yaw uint8
```

It's an optional frame, so would never actually be transmitted by the LTM protocol.

Ideally I'd like the definition to be added to the LTM page at: https://github.com/stronnag/mwptools/blob/master/docs/ltm-definition.txt

Your thoughts @stronnag

stronnag, 2016-09-09:
I have no objections, and I'll gladly add it to the mwp documentation; actually I'll add it to the iNav wiki (and reference that from mwp), much better idea.

martinbudden, 2016-09-09:
Great. I'm now looking at adding LTM telemetry support to deviationtx.

stronnag, 2016-09-09:
For general review / information 

https://github.com/iNavFlight/inav/wiki/Lightweight-Telemetry-(LTM)

digitalentity, 2016-09-10:
Great! I wonder if we can have native NRF protocol with telemetry in Taranis as well...

martinbudden, 2016-09-10:
The guy who does the multiprotocol hardware module for the Taranis seems to be fairly active in porting over new protocols from deviationtx. So once my code is accepted into deviationtx it should, in due course, become available for Taranis.

martinbudden, 2016-09-10:
@stronnag , thanks for doing that.

stronnag, 2018-04-04:
Resolved 2016-10 with optional 'T' frame. Closing

closed 2018-04-04 by stronnag