iNavFlight/inav #577: Feature Request: Fixed Wing Auto take off.

oleost, 2016-09-08:
Feature request, getting a proper auto takeoff for iNav for fixed wing.

Suggestion, as per #564 it was talking about getting a launch detection based on accelerometer data.

With and launch detection / state we could do:

Prevent RTH to engage if launch is not detected. And do RTH as soon as launch is detected.

So to use this would be to:

Get home lock.
Arm and enable RTH.
Throw airplane and watch it rise to the sky and start loiter :+1: 

As and sidenote, it could be and idea that when RTH is engage but not yet gotten launched state, it could spin prop at min_throttle to let the user know its ready to fly, also this way you already have some rpm on the motor.

digitalentity, 2016-09-10:
I like the idea!

I think we may also make takeoff mode initially limit maximum climb angle to allow reaching cruise speed faster and prevent stalling on underpowered airplanes. I.e. if minimum preconfigured speed is not reached - cap climb angle to some small value (i.e. 5 deg)

bk79, 2016-09-10:
Auto take off is something i would like also for multirotor something like
arm and stay up 1m on poshold

Schalonsus, 2016-09-11:
The new Parrot Disco has cool Auto Takeoff feature. Long press button, motor starts spinning low RPM, after few seconds motor spins high RPM, ready to throw.
https://www.youtube.com/watch?v=PQK9OPCAhyk

oleost, 2016-10-15:
Closing this one.  It's being implemented #664

closed 2016-10-15 by oleost