iNavFlight/inav #578: Skyline32+OSD Advanced - no mag detected

safurniss, 2016-09-08:
Just got the above flight controller and connected it to Clean Flight Configurator. It came with 1.10.0 firmware and only showed Gyro, Accel & Baro as green... the mag is not detected.

From CLI I ran status and got the following output
`System Uptime: 953 seconds, Voltage: 0 * 0.1V (1S battery - NOT PRESENT)
CPU Clock=72MHz, GYRO=MPU6050, ACC=MPU6050.n, BARO=MS5611
Cycle Time: 3498, I2C Errors: 0, config size: 2016`

So either I have a faulty controller or its just not being detected.

DzikuVx, 2016-09-09:
@safurniss "1.13.0" is Cleanflight, can you confirm the same is happening with INAV?

safurniss, 2016-09-09:
Hi yes just tried it in both iNAV and BaseFlight and I get the same result... mag is not detected.

DzikuVx, 2016-09-09:
could you confirm target you used in INAV (SPRF3?) and is this thing really using HMC5883l magneytometer?

safurniss, 2016-09-09:
You will have to forgive me here... being new to all this I wonder if I have flashed it with the wrong firmware... I was under the impression that the Skyline32 is a NAZE clone and used the NAZE firmware...

Should I have used the SPRF3?

DzikuVx, 2016-09-09:
Ah, sorry, I was under impression it was STM32F3 not NAZE clone, my bad....
Still, if none of software can detect MAG then something is wrong:
1 - it is not HMC5883l after all
2 - board damaged
3 - there is something special with this board

Can you confirm HMC5883L is present (small chip labeled 883L)

raoul596, 2016-09-10:
I have a NAZE clone. I am getting the same problem. In iNav 1.1 the MAG did work.

anbello, 2016-09-11:
Similar problem with naze clone, all OK with inav 1.1, with 1.2 if I do several reboot (both hw an sw) sometimes mag is detected and sometimes is not detected.

anbello, 2016-09-11:
when naze32 detects the mag:
# status
System Uptime: 11 seconds, Voltage: 0 * 0.1V (3S battery - OK), System load: 0.02
CPU Clock=72MHz, GYRO=MPU6500, ACC=MPU6500, BARO=BMP280, MAG=HMC5883
Cycle Time: 2009, I2C Errors: 1, config size: 1668

when naze32 don't detects the mag:
# status
System Uptime: 7 seconds, Voltage: 0 * 0.1V (3S battery - OK), System load: 0.01
CPU Clock=72MHz, GYRO=MPU6500, ACC=MPU6500, BARO=BMP280
Cycle Time: 2001, I2C Errors: 2, config size: 1668

oleost, 2016-09-11:
@anbello cli command bootlog would be useful. 

Probably you got some issues on your setup, iNav detects this and refuse to use the sensor. 

Either this check must be soften up more (more delayed or accept more fails on boot)  or you got an issue you need to fix. 

Is it internal or external mag?

anbello, 2016-09-11:
My FC is naze32 rev6 10DOF from banggood and has internal mag and baro

#bootlog when naze32 detects the mag:
Time Evt Parameters
   0:  0 
   0:  1 
 500: 19  (0, 0, 0, 0)
 500: 19  (1, 1, 0, 2)
 500: 19  (2, 2, 0, 2)
 500: 19  (3, 3, 0, 2)
 500: 19  (4, 4, 0, 2)
 500: 19  (5, 5, 0, 2)
 500: 19  (6, 6, 0, 2)
 500: 18  (7, 6, 0, 1)
 500: 18  (8, 6, 0, 1)
 500: 18  (9, 6, 0, 1)
 500: 18  (10, 6, 0, 1)
 500:  2 
 551:  3 
1087:  9  (7, 0, 0, 0)
1548: 10  (8, 0, 0, 0)
1600: 11  (4, 0, 0, 0)
1600: 12  (2, 0, 0, 0)
3258: 13  (0, 0, 0, 0)
3258:  4 
3759:  5 
3810:  8

#bootlog when naze32 don't detects the mag:
Time Evt Parameters
   0:  0 
   0:  1 
 500: 19  (0, 0, 0, 0)
 500: 19  (1, 1, 0, 2)
 500: 19  (2, 2, 0, 2)
 500: 19  (3, 3, 0, 2)
 500: 19  (4, 4, 0, 2)
 500: 19  (5, 5, 0, 2)
 500: 19  (6, 6, 0, 2)
 500: 18  (7, 6, 0, 1)
 500: 18  (8, 6, 0, 1)
 500: 18  (9, 6, 0, 1)
 500: 18  (10, 6, 0, 1)
 500:  2 
 586:  9  (7, 0, 0, 0)
1048: 10  (8, 0, 0, 0)
1100: 11  (4, 0, 0, 0)
1100: 12  (1, 0, 0, 0)
1100: 13  (0, 0, 0, 0)
1100:  4 
1600:  5 
1651:  8

digitalentity, 2016-09-29:
@anbello according to the log MAG chip was not detected entirely without any initialisation errors. Do you observe any correlation between mag detection and soft (from Configurator) vs hard (power cycle) reboots?

anbello, 2016-09-29:
@digitalentity I did not observe any correlation, I have sometimes mag detected and sometimes not detected both after soft and hard reboots.

digitalentity, 2016-09-29:
Now this is really interesting. Hard boot adds a 500ms extra delay so in your case it's something else. Probably faulty chip as post-1.1 versions will check for I2C read errors. If your compass chips fails to respond or doesn't finish the i2c transaction correctly, it won't be detected.

anbello, 2016-09-29:
Other information on my problems with internal compass on naze32 in #594 in particular it could be of interest to know that with external pullups I have no problems.

jstremmler, 2016-10-04:
Have a look at #594! 

I would recommend downloading the actual development code and compile it again with cygwin64!

http://cygwin.com

It looks like the compiler by which the *.hex files have been produced  on github has an issue.....

digitalentity, 2016-10-18:
Is this issue still valid in 1.3-RC1?

closed 2016-12-14 by digitalentity