iNavFlight/inav #586: Occasional, random invalid mag readings post f262154

stronnag, 2016-09-11:
Since f262154  I've observed 2 instances of bizarre mag behaviour over perhaps 20 nav flight, on two normally reliable craft (the usual suspects).  When this condition happens, the mag reports readings with an offset from the true reading that varies between 30°and 100°. A reboot will fix the problem. 

When this happens at boot, prior to takeoff, it's benign, as  I will catch it from the CGS reporting the wrong heading prior to arm (or takeoff). On one occasion, it happened mid-flight (which was noticeable from the CGS as I'm familiar with orientation of 'my' field); fortunately no nav function was engaged at the time.

I have BBLogs (no bootlog) of the occurrences. I guess my next move will be to revert 0f63bc6, but given the relative rarity of occurrence ( once on each of 2016-09-07 (6cc415a) and 2016-09-11 (6cc415a)) reaching any conclusion will not be easy, and I've never seen any saturation warning in logs.

f262154 was the last firmware flown prior to seeing this issue, but given the nature of the problem that doesn't say much, as I've flown nav on 3 other days on  6cc415a or 8918e6f without any problem.  Bizzare,  if it hadn't happened twice I'd have just dismissed a single occurance as a glitch in the matrix.

digitalentity, 2016-09-13:
Interesting. Definitely worth looking but unfortunately I'm too busy to investigate now and for two more weeks.

stronnag, 2016-09-13:
No problem, I've reverted 0f63bc6  to see if that helps. Difficult to prove a negative.

stronnag, 2016-09-14:
So far seems completely stable again without  0f63bc6 ... testing continues

martinbudden, 2016-09-14:
That's interesting, especially since that PR was designed specifically to improve the compass initialisation. I wonder which part of it is problematic.

stronnag, 2016-09-18:
#612 looking hopeful as a solution. More testing to follow.

stronnag, 2016-09-20:
Another long set of flights on a different BMP280 machine. No problems. I'm guessing #612 fixes this one, closing.

closed 2016-09-20 by stronnag

martinbudden, 2016-09-20:
Great.