iNavFlight/inav #590: SPracing F3evo nav_gps_hold drifts in position and altitude after 10 seconds

elepete, 2016-09-11:
this SW is really great - on a SP F3evo board the GPS_HOLD mode starts after some time to drift
drift is in position and altitude
I have tried this 2 times - and 12sec

oleost, 2016-09-11:
If it continues to drift and drift my first bet are your stick are not centeres at 1500us.

If not that have a look at this post to get a hint what you should include when asking for help or submitting a bug,  since without any infomation we are only guessing here. 

http://www.rcgroups.com/forums/showpost.php?p=35637535&postcount=7930

This is also mentioned in the wiki https://github.com/iNavFlight/inav/wiki/1.-Getting-started-with-iNav

elepete, 2016-10-01:
As the issue looked like a misaligned sensor I finally found what needs to be done mag y- or x- axis needs to be mirrored.
I have not yet been flying - but watching the model on the status screen - I expect that fixes the compass issue. Is the MPU9250 - or the AK8963 potentially not correctly factory trimmed or simply inverting 1 axis?

I did that in the compass_ak8963.c file:

    magData[X] = -(int16_t)(buf[1] << 8 | buf[0]) * magGain[X];
//line reinserted with *1 / *-1 below:    magData[Y] = -(int16_t)(buf[3] << 8 | buf[2]) * magGain[Y];
    magData[Z] = -(int16_t)(buf[5] << 8 | buf[4]) * magGain[Z];

#if defined(USE_SPI) && defined(MPU9250_SPI_INSTANCE)
//inserted line not inverted   
    magData[Y] = (int16_t)(buf[3] << 8 | buf[2]) * magGain[Y];  
    // cache mag data for reuse
    memcpy(cachedMagData, magData, sizeof(cachedMagData));
    state = CHECK_STATUS;
    lastReadResult = true;
#else
//inserted line *-1 as original
    magData[Y] = -(int16_t)(buf[3] << 8 | buf[2]) * magGain[Y];
    lastReadResult = ak8963SensorWrite(AK8963_MAG_I2C_ADDRESS, AK8963_MAG_REG_CNTL, CNTL_MODE_ONCE); // start reading again
#endif

digitalentity, 2016-10-02:
@elepete what was the result of your investigations?

elepete, 2016-10-03:
Rain stopped - and I flew with mag_y = -mag_y
Stable position hold in 3D poshold mode - no drift - no toilet bowl effect visible.

the initial issue was (start of issue see top), once the frame had a pitch or roll angle >20 degree the mag heading (because of the swap z-axis) changed. this caused drift / toilet bowl. as long as the frame is at level this is no issue, once roll / pitch angles go near or over magnetic inclination (in my location 64 degrees) the z-axis changes sign and the issue observed starts.

Reading through spec of AK8963 and - my preliminary conclusion is:
- either the AK8963 is broken
could that be? the z-axis (+/-0.2 in sensors tab) is having only ~1/2 output compared to x- and y-axis (+/-0.35)
- configuring all orientations per datasheet of MPU9250 (and AK8963) in my board the z-axis is inverted (and ~1/2 lower than x- / y-axis; i should have inverted z-axis which just results in different orientations)
- the compass reading is not exactly pointing N / E / W / S, but seems a little elliptic (I can test that in compiling with an extra factor coded on z-axis if it helps)
- before concluding it would be good to get more feedback on other boards with an MPU9250 or AK8963; questions: 1) are all 3 mag axis going to have the exact same full scale reading 2) z-axis inverted like in my board or normal?
- or there is something other strange - but I would not guess so...

elepete, 2016-10-03:
New compile with hard coded x1.5 multiplier on z-axis.
-> mag z-axis now on sensor tab scales to ~same value as x- and y-axis
-> the compass reading is now approx. a circle
so i would guess this is a board issue -> send a mail to SP racing on that

hydra, 2016-10-03:
@elepete It's highly doubtful there's an issue with the hardware.  Perhaps @digitalentity has time to look into this a little further.  I don't currently have time to investigate this further right now as I'm busy working on other issues.  Maybe provide some screenshots and blackbox logs highlighting the issue.

@digitalentity is there anything you can think of that he can try?  I don't have brain space right now.

hydra, 2016-10-03:
@elepete If there's a specific test you'd like me someone else (me or another person) to replicate please provide a config dump, link to the exact firmware you are running and steps to repeat the test.  Only this will allow someone to replicate your issue.

elepete, 2016-10-07:
thank you for looking to look into that - and sorry for being late responding

I have atached - the CLI with settings INAV 1.2.1 no modifictaions - except calibration of Accelerometer and Magnetormeter (CLI dump attached)
I have checked the axis with ACC and GYRO (F3 EVO board with USB on top side front, X-ax pointing fwd, Y-ax pointing left, Z-ax pointing up)

on the picture you see (calibrated sensors):
left circle (inside left to right): x (blue) pointing down (~65°) to max field (mid little circle 90° mid = ~0) x pointing up (~65°) min field
middle circle: y (green) the same as with x-axis
right circle: z (red) the same as with x-axis
while x- and y-axis have their sign in same direction, z-axis is reverse polarity.

You can also see that as at 65° angle, accelerometer above behaves the same for all 3-axis starting low (neg) and ending positive. Comparing this at x- and y-mag-axis all fine - z-mag-axis is inverted.

my conclusion is, the mag sensor has an issue.
Also for calibrated mag from AKM / Invensense I am pretty sure all three axis should scale to the same end values.

Let e know if any other data is needed.

[image]

# status
System Uptime: 1698 seconds, Voltage: 0 * 0.1V (1S battery - NOT PRESENT), System load: 0.01
CPU Clock=72MHz, GYRO=MPU6500, ACC=MPU6500, BARO=BMP280, MAG=MAG_AK8963
Cycle Time: 2008, I2C Errors: 3, config size: 1996

# dump

# version
# INAV/SPRACINGF3EVO 1.2.1 Sep 24 2016 / 10:47:22 (36954a8)
# pflags
# Persistent config flags: 0x00000001
# dump master

# mixer
mixer QUADX
mmix reset
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -INFLIGHT_ACC_CAL
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -FAILSAFE
feature -SONAR
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DISPLAY
feature -ONESHOT125
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_NRF24
feature -SOFTSPI
feature RX_PPM
feature VBAT
feature FAILSAFE
feature TELEMETRY
[... 390 lines cut ...]
set max_angle_inclination_pit = 300
set gyro_soft_lpf_hz = 60
set acc_soft_lpf_hz = 15
set dterm_lpf_hz = 40
set yaw_lpf_hz = 30
set yaw_p_limit = 300
set iterm_ignore_threshold = 200
set yaw_iterm_ignore_threshold = 50
set rate_accel_limit_roll_pitch = 0
set rate_accel_limit_yaw = 10000

# dump rates

# rateprofile
rateprofile 0

set rc_expo = 70
set rc_yaw_expo = 20
set thr_mid = 50
set thr_expo = 0
set roll_rate = 20
set pitch_rate = 20
set yaw_rate = 20
set tpa_rate = 0
set tpa_breakpoint = 1500

oleost, 2016-10-09:
@elepete Hi, I'm having som issue with SPRACINGF3mini target. Spesific the magnetometer, got issue with drift and not locked in movement.

I want to try your fix. But I don't understand what you have done. You change this line only? https://github.com/iNavFlight/inav/blob/master/src/main/drivers/compass_ak8963.c#L345

And what did you replace it with?

elepete, 2016-10-09:
@oleost  - yes - basically you can change line
https://github.com/iNavFlight/inav/blob/master/src/main/drivers/compass_ak8963.c#L345
as I needed to invert the axis I have just removed the "-"
New line 345 if you want to invert mag x would look like:
 magData[X] = (int16_t)(buf[1] << 8 | buf[0]) * magGain[X];
in my case i just had to invert one axis.

However - I am still convinced the cleanflight / inav code is matching the datasheet of AK8963 (MPU9050).
In the end in my case I did modify the
https://github.com/iNavFlight/inav/blob/master/src/main/drivers/compass_ak8963.c#L242
This as I am still pretty sure the issue is with the z-axis that is also having too less output compared to x- and y-axis.
 magGain[Z] = (((((float)(int8_t)calibration[Z] - 128) / 256) + 1) * -45);
So I have inverted with the -45 in the end of the equation; this was before a factor of 30.

I my view the question is - is the AK8963 output:
a) same strength in all directions?
b) are all 3 axis when pointing to north max field strength (in my case at 65° into the earth) having the same sign?

oleost, 2016-10-09:
@elepete Okey, thank you. Tried it but that didnt help me.

elepete, 2016-10-09:
@oleost - did you check the output of the AK8963 in the sensors tab vs. expectation and theory?
i researched until I found the issue - and with "my fix" that compensates a likely wrong factory programmed or "broken" AK8963 the quad flies

oleost, 2016-10-10:
@elepete This is my magnetometer when board is laying flat pointing north.

[image]

Laying flat and pointing west:
[image]

elepete, 2016-10-10:
@oleost - thank you:
north laying flat is for my board properly calibrated:
[image]
west and laying flat:
[image]
If you are somewhere in Europe (I am in Germany) the z-axis for your and my position should have the same sign.

I do think negative z-axis is correct.
If magnetic flow is direction North (and for Europe downward into the earth) plus board properly oriented:
North:
x-ax (pointing fwd): + magnetic flow, y-ax (pointing left): ~0, z-ax (pointing up): - magnetic flow
West:
x-ax (pointing left): ~0, y-ax (pointing back): - magnetic flow, z-ax (pointing up): - magnetic flow

So if my thinking is correct. You would need to align all properly that the SP F3 mini is looking like mine.

In case all is configured correct - there may really be a systematic issue. As in my board case the z-axis is inverted... looks like the same for you.

oleost, 2016-10-10:
I live in Norway.

elepete, 2016-10-10:
@oleost - did you try inverting z-axis?
(sorry to be precise z-axis should be for all north of equator the same polarity)

oleost, 2016-10-11:
@elepete Thanks for pointing it out, will try again tonight and report back.

digitalentity, 2016-10-18:
Ok, I compared axis directions returned by current driver and compared with datasheet for MPU9250. It appears that **all three axis** are inverted by current driver. As there is no rotation to invert all 3 axis we have invalid vector reporting regardless of compass alignment setting.

hydra, 2016-10-19:
good work everyone!  I'm so glad you guys are looking at all the navigation stuff!  @digitalentity @stronnag you guys rock.   thanks for persevering and providing all the info and testing time @elepete and @oleost

digitalentity, 2016-10-19:
@hydra more improvements (especially airplane) are coming in soon. I'm not very happy with airplane performance and that's what I intend to change next.

elepete, 2016-10-20:
@all 
Great - thank you all for looking into this - of curiosity I looked again in the datasheet of MPU9250.
Theory now matches outputs. I did overlook that all 3 axis have been wrong...
On my EVO board the fix #707 works as well.
(sorry for being silent - was 4 days traveling for my company)

However - for me one item remains. My MPU9250 / AK8963 inside has x- and y-axis 1.5x more output than z-axis. That results in an elliptic compass reading. Resulting in some some compass drift when tilting.
Is that only my AK8963?
In case this is "normal" the compass calibration could or should replace the factory compensation values
https://github.com/iNavFlight/inav/blob/master/src/main/drivers/compass_ak8963.c#L240-L242

digitalentity, 2016-10-21:
@elepete, on my EVO all 3 axis scale equally. It's possibly flawed hardware or magnetic interference.
Calibration logic can calculate scales as well, but I'm not sure it's necessary. 

In worst case (in your case 90 deg tilt which is not a very realistic scenario) x1.5 error in axis scale will only result in 11 deg error which is still flyable.

I had an error of 24 deg once - forgot the sign of mag declination - and RTH still worked. Copter came home in a weird spiral path, but it still got back and landed perfectly.

digitalentity, 2016-10-21:
Fix merged, closing.

closed 2016-10-21 by digitalentity

maRCnh33, 2018-11-05:
> As the issue looked like a misaligned sensor I finally found what needs to be done mag y- or x- axis needs to be mirrored.
> I have not yet been flying - but watching the model on the status screen - [...]

how can i put this data in?? i have download the inav zip file with everything in it, but how to change the compass.h file and add it to my current configuration???  https://github.com/iNavFlight/inav/issues/3987

please help me!!

commit 26f2affd (2015-05-20): Ignore SBus end byte checking, this should improve compatibility with
some Futaba SBUS RX's that send telemetry data after SBus frames.  Fixes
#590
PR #707 (2016-10-21): Fix axis directions for AK8963 compass (MPU9250 compass chip)
Fixes #590