iNavFlight/inav #599: Led strip on NAZE 32

langph, 2016-09-12:
I have a Skyline 32 advanced 1.1 and would really like to add a LED for status purposes. Is it possible to enable led strip for  release 1.2 as it was on release 1.1?

DzikuVx, 2016-09-13:
@langph you might want to read #457 
Bottom line is you would have to compile your own version of INAV

closed 2016-09-13 by DzikuVx

martinbudden, 2016-09-13:
Perhaps we should put something in the wiki about this.

stronnag, 2016-09-13:
Like the existing "Getting Started " page that no one appears to read?

martinbudden, 2016-09-13:
You're right. It's all there on the "Getting Started" page. I must confess I haven't read that for ages.

stronnag, 2016-09-13:
Really, we should perhaps
1. Fix the configurator so it points to iNav builds (not CF builds)
2. When the user selects a F1 build, show some text that warns them about missing features / points them go the wiki

DzikuVx, 2016-09-13:
I think 1 was done
about 2, I have a plan to work on configurator a little, so maybe it will be done

langph, 2016-09-13:
I have overlooked that, looked in the release notes and saw that led strip was re-enabled in 1.1, hence my question.

The Features safe to remove are incomplete though.

I will compile my own. Would it be possible to read out a board and deselect support for sensors etc. that it does not have, compile on the spot and then upload it in the board? Might save some space for new features.

oleost, 2016-09-13:
@langph Feel free to help out making that list more complete. 

Also updated wiki with a small video to help out.