iNavFlight/inav #600: Servo

csarka, 2016-09-13:
I just updated my multirotor from inav 1.1 to 1.2. I am using the RMDO flight controller. I'm having trouble now controlling any servo via AUX channel. 

Previously (inav 1.1), the servo worked (AUX 3) with FEATURE SERVO_TILT enabled. Strangely CHANNEL_FORWARDING did not need to be enabled.

Now (inav 1.2), no combination of SERVO_TILT enabled/disabled and/or CHANNEL_FORWARDING enabled/disabled will allow any servo action on any AUX channel.

Are servos on AUX channel still working?

Dumps before and after are attached.
[dump_9-10-16_inav1.1.txt](https://github.com/iNavFlight/inav/files/468613/dump_9-10-16_inav1.1.txt)
[dump_9-12-16_inav1.2.txt](https://github.com/iNavFlight/inav/files/468614/dump_9-12-16_inav1.2.txt)

Thanks for your help,

Teja28, 2016-09-14:
Same problem...

xdigyx, 2016-09-14:
Not missing, just reconfigured. Read last posts on #539

csarka, 2016-09-15:
I tried enabling CAMSTAB:
aux 6 7 2 900 2100
with various combinations of SERVO_TILT and CHANNEL_FORWARDING
still no success...
The first 4 channels are for the motors, that means any of the last 4 should be available for servo, right ?
@Teja28 Can you able to enable CAMSTAB mode and try as @xdigyx  susggests in #539 ?

xdigyx, 2016-09-15:
no.. you did not read the issue I posted carefully...
    FEATURE_CHANNEL_FORWARDING - all motors are available, AUX are forwarded to RC5-8.
    FEATURE_SERVO_TILT - servos are enabled on M1-2, motors 5-6 are disabled (this is expected behavior), NAZE target doesn't support SERVO_TILT + 6-motor configuration
 FEATURE_SERVO_TILT only is driving servos if CAMSTAB mode is enabled. 
    FEATURE_CHANNEL_FORWARDING + FEATURE_SERVO_TILT - servos are enabled on M1-2, motors 5-6 are disabled, AUX are forwarded to RC5-8

JPRivet, 2016-09-27:
Could you elaborate on what RC5-8 is referring to please? Does this mean the outputs are available on M5-8 on a F3 FC?

Thanks

DzikuVx, 2016-09-27:
@JPRivet RC5-8 stands for RC input pins 5 to 8. Some inputs can be remapped to outputs

JPRivet, 2016-09-27:
Ah ok - thanks!

On a SPRacingF3 EVO, it doesn't have these pins (RC1-8) since it's designed for PPM/SBUS instead - is this correct?

csarka, 2016-09-28:
Thank you @DzikuVx DzikuVx  That explains a lot!  I am trying to enable at least one servo output on M5-8.  (M1-4 are connected to the ESCs). RC-5-8 are already occupied for other I/O.  Using the RMRC Dodo Rev 3a.
I have tried using M6, M7, and M8 with every combination of features, still no success.

DzikuVx, 2016-09-29:
@csarka unfortunately we are quite limited here and without complete rework of code that drives output mappings there is not much we can do.

krzysztofmatula, 2016-09-30:
So, does it mean that CHANNEL_FORWARDING is not supposed to work for SPF3-EVO?

krzysztofmatula, 2016-10-17:
I'm trying to use the CHANNEL_FORWARDING on SPF3-EVO without success...
I have 4 motors on outputs 1-4 which provide PWM at ~400Hz. Outputs 5-8 are constantly silent (no PWM).
Can someone confirm if the channel forwarding should be working on this board or not?
I'm on INAV-1.2.1 currently.

digitalentity, 2016-10-17:
Please provide the dump and confirm that you are trying the latest firmware
- 1.3-RC1

krzysztofmatula, 2016-10-18:
Here's the dump (latest firmware):
```
# dump

# version
# INAV/SPRACINGF3EVO 1.3.0 Oct 15 2016 / 08:42:45 (d54a6c5)
# pflags
# Persistent config flags: 0x00000000
# dump master

# mixer
mixer QUADX
mmix reset
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -UNUSED_1
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -FAILSAFE
feature -SONAR
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DISPLAY
feature -UNUSED_2
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -SOFTSPI
feature -PWM_SERVO_DRIVER
feature RX_PPM
feature VBAT
feature GPS
feature FAILSAFE
feature TELEMETRY
feature RSSI_ADC
feature LED_STRIP
feature BLACKBOX
feature CHANNEL_FORWARDING

# beeper
beeper GYRO_CALIBRATED
beeper RX_LOST
beeper RX_LOST_LANDING
beeper DISARMING
beeper ARMING
beeper ARMING_GPS_FIX
beeper BAT_CRIT_LOW
beeper BAT_LOW
beeper GPS_STATUS
beeper RX_SET
beeper ACC_CALIBRATION
beeper ACC_CALIBRATION_FAIL
beeper READY_BEEP
beeper MULTI_BEEPS
beeper DISARM_REPEAT
[... 375 lines cut ...]
set dterm_lpf_hz = 40
set yaw_lpf_hz = 30
set yaw_p_limit = 300
set iterm_ignore_threshold = 200
set yaw_iterm_ignore_threshold = 50
set rate_accel_limit_roll_pitch = 0
set rate_accel_limit_yaw = 10000

# dump rates

# rateprofile
rateprofile 0

set rc_expo = 70
set rc_yaw_expo = 20
set thr_mid = 50
set thr_expo = 0
set roll_rate = 40
set pitch_rate = 40
set yaw_rate = 40
set tpa_rate = 0
set tpa_breakpoint = 1500

# 
```

digitalentity, 2016-10-19:
Actually, I've looked at the code and channel forwarding is only supported on NAZE and origianl SPRacingF3. So far EVO is not supporting the feature.

digitalentity, 2016-10-19:
We can enable channel forwarding on S7 asnd S8 outputs, but this will effectively limit multirotor setups with servos to 4 motors due to hardware limitations.

krzysztofmatula, 2016-10-19:
If it's hardware limitation we must live with it...
4+2 or 8+0 switchable by CHANNEL_FORWARDING feature is still much better than no servo support at all.

digitalentity, 2016-10-19:
@hydra, as a designer of SPRacingF3 family maybe you can give us a clue how to get channel forwarding working on EVO?

mifan1, 2016-10-19:
I found that in older version CF/BTFL is diference between F1 and F3 with SERVO_TILT :
- for F1 targets the ESC outputs move up two pins to 3-6 and AUX outputs on pins 1 & 2
- whereas on F3 targets the AUX outputs on ESC pins 7 & 8 and the ESC outputs stay on pins 1-4.

hydra, 2016-10-19:
@mifan1 it's not specifically ALL F3 targets, it's dependent on the timer configuration and PWM mapping code.  Servos need a different time to motor outputs.

EVO:
```
    { TIM8,  GPIOA, Pin_15, TIM_Channel_1, TIM8_CC_IRQn,            0, Mode_AF_PP_PD, GPIO_PinSource15, GPIO_AF_2},  // PPM

    { TIM2,  GPIOA, Pin_0,  TIM_Channel_1, TIM2_IRQn,               1, Mode_AF_PP, GPIO_PinSource0,  GPIO_AF_1},  // PWM1
    { TIM2,  GPIOA, Pin_1,  TIM_Channel_2, TIM2_IRQn,               1, Mode_AF_PP, GPIO_PinSource1,  GPIO_AF_1},  // PWM2
    { TIM15, GPIOA, Pin_2,  TIM_Channel_1, TIM1_BRK_TIM15_IRQn,     1, Mode_AF_PP, GPIO_PinSource2,  GPIO_AF_9},  // PWM3
    { TIM15, GPIOA, Pin_3,  TIM_Channel_2, TIM1_BRK_TIM15_IRQn,     1, Mode_AF_PP, GPIO_PinSource3,  GPIO_AF_9},  // PWM4
    { TIM3,  GPIOA, Pin_6,  TIM_Channel_1, TIM3_IRQn,               1, Mode_AF_PP, GPIO_PinSource6,  GPIO_AF_2},  // PWM5
    { TIM3,  GPIOA, Pin_7,  TIM_Channel_2, TIM3_IRQn,               1, Mode_AF_PP, GPIO_PinSource7,  GPIO_AF_2},  // PWM6
    { TIM3,  GPIOB, Pin_0,  TIM_Channel_3, TIM3_IRQn,               1, Mode_AF_PP, GPIO_PinSource0,  GPIO_AF_2},  // PWM7
    { TIM3,  GPIOB, Pin_1,  TIM_Channel_4, TIM3_IRQn,               1, Mode_AF_PP, GPIO_PinSource1,  GPIO_AF_2},  // PWM8
```

```
#if defined(SPRACINGF3EVO)
            // remap PWM8+9 as servos
            if ((timerIndex == PWM8 || timerIndex == PWM9) && timerHardwarePtr->tim == TIM3)
                type = MAP_TO_SERVO_OUTPUT;
#endif
```

you have to ignore the comments on the timer.c code. but basically ESC outputs 7+8 should be used as servos.

TIM3 is also used for softserial 1/2 IIRC.

The pwm mapping code could be updated to use ESC outputs 5,6,7 & 8 to drive 4 servos or updated so that ESC 1/2 are servo outputs and 3 onwards are for motor/softserial.

csarka, 2016-10-19:
So, the RMDO would be the same configuration?

hydra, 2016-10-19:
Ask RMRC.

hydra, 2016-10-19:
@digitalentity the channel forwarding is here:

```

        if (init->useChannelForwarding && !init->airplane) {
#if defined(NAZE) && defined(LED_STRIP_TIMER)
            // if LED strip is active, PWM5-8 are unavailable, so map AUX1+AUX2 to PWM13+PWM14
            if (init->useLEDStrip) { 
                if (timerIndex >= PWM13 && timerIndex <= PWM14) {
                  type = MAP_TO_SERVO_OUTPUT;
                }
            } else
#endif

#if defined(SPRACINGF3) || defined(NAZE)
                // remap PWM5..8 as servos when used in extended servo mode
                if (timerIndex >= PWM5 && timerIndex <= PWM8)
                    type = MAP_TO_SERVO_OUTPUT;
#endif
        }
```

gosh I hate pwm_mapping.c it's such a hack.

hydra, 2016-10-19:
you also have to look at the pwm map that is used, the amount of servos used by the mixer and then you might just be able to figure out on which pin a forwarded channel might end up on.  good luck! 😄

digitalentity, 2016-10-19:
So basically, we have channel forwarding only for SPRF3 and NAZE, that's a problem. We need more targets supporting this.

On EVO TIM3 can probably be used for channel forwarding (4 channel) - outputs 4-8 will be servo channels.

hydra, 2016-10-19:
yeah, i'm fixing it here, 1 sec.

hydra, 2016-10-19:
really, *all* unused motor outputs should just be remapped to servos i think.

hydra, 2016-10-19:
oh gosh.  I'm not touching that code right now.  Gonna wait for the BF changes to fix it. @blkmn was working on it.

hydra, 2016-10-19:
channel forwarding will be used on the first available servo pin, IIRC.  The code above just assigns extra pins on the NAZE and SP Racing F3 - both of those boards have 8 channel PWM input.  When the mapping is a PPM input the PWM inputs are free to be used as outputs.  The EVO and other boards do not need that code to run because they don't have any used PWM input pins.

digitalentity, 2016-10-19:
I'm not sure any code change can fix it. We are limited by timers - they have different base frequency for pwm/oneshot/etc. Basically, if it wasn't considered during HW design phase - our choices are very limited.

hydra, 2016-10-19:
agreed.  it's why I specifically grouped the pins in pairs and 4's on the the SP Racing boards.

Unlike the Sparky and CC3D which have pins and timers all over the place.

digitalentity, 2016-10-19:
I totally agree - pwm_mapping.c is a huge messy pile of junk. Can't wait for somebody to clean it up. If @blakmn is on it - I'll wait for his change as well. We don't need 3 implementations of the same thing.

digitalentity, 2016-10-19:
CC3D is one messiest board I've hacked, but it's dirt cheap and pretty functional.

hydra, 2016-10-19:
i think we might be able to do this:

```
        if ((init->useServos || init->useChannelForwarding) && !init->airplane) {
```

as a quick workaround.  gonna have a little look, but if it starts taking too long I'm backing out.

hydra, 2016-10-19:
yeah, that seems to work.  e.g.

SPRACINGF3EVO, Tricopter + channel forwarding.
ESC1-3 = motors,
ESC5 = servo
ESC6/7/8 = AUX1/2/3

digitalentity, 2016-10-19:
This might affect other targets. I'll test, thanks for suggestion.

hydra, 2016-10-19:
Yeah, I don't see the harm in the change though.  I'll merge that in CF now and people can do some testing in the next RC.

hydra, 2016-10-19:
In looking at this it appears there's more bugs in the code.  No point me duplicating @blckmn 's code and effort right now.

hydra, 2016-10-19:
The pwm mapping code is not aware of how many actual servos and actual motors it needs.  And that is the fundamental problem.  Nor is it aware of the order in which timers and channels should be allocated for motors/servos.
It should also know about which other pins the user has allocated - e.g. for sonar, ppm, pwm, softserial, etc.

If you had that information, on a per-target basis, then the code could probably be a LOT simpler.  any references to airplane and dependencies on other subsystems would the be able to be removed entirely.

blckmn, 2016-10-19:
We've just merged the resource io code in BF3.1 (development branch). You can now assign any timer mapping (for any target) to servo output. I've yet to test if channel forwarding works.

blckmn, 2016-10-19:
https://github.com/betaflight/betaflight/pull/1189 First comment has the instructions on use.

blckmn, 2016-10-19:
@hydra the mapping to actual use is now done by the user and stored in configuration. The pwm mapping code is effectively removed. The target has the opportunity to map its default working configuration on reset in a target specific config, or it goes by the default output order in timer mapping (based on function).

It will require effort in the Configurator at some point in the future for potential quick config scenarios per target, and warnings on resource conflicts.

blckmn, 2016-10-19:
By quick config scenarios I mean Quad Racer, Airplane, Tricopter etc

hydra, 2016-10-19:
@blckmn yeah, I'm looking forward to seeing that.

@digitalentity and others - see these two commits

94256215f19c577f8a86d4650b4a5933ae6f0eca
696d5d919310d8ac9c443654aafbeda06dca56d3

I figured I'd take the safest approach and only risk breaking two targets 😄 

Channel forwarding/servo allocation takes precedence over motor outputs so if your mixer uses more than four motors then enabling channel forwarding when using more than 4 motors is not supported right now.  This is probably OK for most people.  Users of Hex/Y6/etc just can't use channel forwarding so not a deal breaker considering the usage of those craft.

krzysztofmatula, 2016-10-20:
I checked those 2 commits and I can only say they work great :)
I have 400Hz PWM on M1-M4 and 50Hz on M5-M8 with AUX 1-4 values forwarded.

closed 2016-12-14 by digitalentity